Koenkk/zigbee2mqtt #21580: I just can't get it to work

Lashandan, 2024-02-26:
### What happened?

I tried to install Zigbee2MQTT following the instructions of this colleague from Youtube (https://www.youtube.com/watch?v=4y_dDgo0i2g&t=203s) but it just doesn't work.
Mosquitto broker is installed.
User for mqtt is created.
ZHA is not configured.
In the configuration of Zigbee2Mqtt user and password are stored and the core-mosquitto:1883 etc.
The correct port of the SkyConnect stick is also specified, here I keep finding different statements.
But I have finally decided on port: /dev/ttyUSB0.
Testing whether I can write here was returned with success.
Countless restarts included.

I keep getting the error:

Zigbee2MQTT:info  2024-02-26 13:02:20: Starting Zigbee2MQTT version 1.35.3 (commit #unknown)
Zigbee2MQTT:info  2024-02-26 13:02:20: Starting zigbee-herdsman (0.33.8)
Zigbee2MQTT:error 2024-02-26 13:03:30: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2024-02-26 13:03:30: Failed to start zigbee
Zigbee2MQTT:error 2024-02-26 13:03:30: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-02-26 13:03:30: Exiting...
Zigbee2MQTT:error 2024-02-26 13:03:30: Error: Failure to connect
    at Ezsp.connect (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:364:19)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:158:13)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:60:27)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)

I urgently need help.
Can you provide me with all the information I need to solve my problem?
Many thanks in advance.

### Zigbee2MQTT version

1.35.3-1

### Adapter firmware version

dont know

### Adapter

SkyConnect_v1.0

### Setup

Addon

### Debug log

[13:02:17] INFO: Preparing to start...
[13:02:17] INFO: Socat not enabled
[13:02:18] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2024-02-26 13:02:19: Logging to console and directory: '/config/zigbee2mqtt/log/2024-02-26.13-02-19' filename: log.txt
Zigbee2MQTT:info  2024-02-26 13:02:20: Starting Zigbee2MQTT version 1.35.3 (commit #unknown)
Zigbee2MQTT:info  2024-02-26 13:02:20: Starting zigbee-herdsman (0.33.8)
Zigbee2MQTT:error 2024-02-26 13:03:30: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2024-02-26 13:03:30: Failed to start zigbee
Zigbee2MQTT:error 2024-02-26 13:03:30: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-02-26 13:03:30: Exiting...
Zigbee2MQTT:error 2024-02-26 13:03:30: Error: Failure to connect
    at Ezsp.connect (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:364:19)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:158:13)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:60:27)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)

Katyara22, 2024-03-14:
I have the same problem

```powershell
Zigbee2MQTT:info  2024-03-13 15:52:07: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2024-03-13.15-52-07' filename: log.txt
Zigbee2MQTT:info  2024-03-13 15:52:07: Starting Zigbee2MQTT version 1.36.0 (commit #2a53b8e)
Zigbee2MQTT:info  2024-03-13 15:52:07: Starting zigbee-herdsman (0.35.1)
Zigbee2MQTT:error 2024-03-13 15:53:18: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2024-03-13 15:53:18: Failed to start zigbee
Zigbee2MQTT:error 2024-03-13 15:53:18: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-03-13 15:53:18: Exiting...
Zigbee2MQTT:error 2024-03-13 15:53:18: Error: Failure to connect
    at Ezsp.connect (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:373:19)
    at Driver.startup (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:163:13)
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:62:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:109:27)
    at start (/opt/zigbee2mqtt/index.js:107:5)
    ```

Katyara22, 2024-03-14:
hi again i just fixed it i just rebooted my OragePi you can try rebooting your OrangePi or RaspberryPI

closed 2024-10-11 by github-actions[bot]