Koenkk/zigbee2mqtt #21583: [New device support]: Detect _TZE200_ft523twt as Woox R7049 (aka V2)

amaciuc, 2024-02-26:
### Link

https://wooxhome.com/products-c10/security-c6/r7049-smoke-alarm-single-unit-discontinued-p70

### Database entry

{"id":37,"type":"EndDevice","ieeeAddr":"0xa4c1387b1635811d","nwkAddr":43796,"manufId":4417,"manufName":"_TZE200_ft523twt","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"GRj-fHRj-\u0012","65506":51,"65508":1,"appVersion":66,"modelId":"TS0601","manufacturerName":"_TZE200_ft523twt","powerSource":3,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1708762038163}

### Comments

I bought some `Woox R7049` smoke sensors and these came as `V2`. They look same as original R7049 but with label `R7049-V2`.

[image]

I loaded as external converter the same definition as older one from [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/8fce3e89c05316110b0598f75d767a3b45a82035/src/devices/woox.ts#L51) and it works with no problems.

@Koenkk I think you can just append the new model `_TZE200_ft523twt` in the actual list with other fingerprints of this smoke sensor.

How it is exposed in Z2M:

[image]

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_ft523twt'}],
    model: 'R7049',
    vendor: 'Woox',
    description: 'Smart smoke alarm',
    meta: {timeout: 30000, disableDefaultResponse: true},
    fromZigbee: [legacy.fromZigbee.R7049_status, fz.ignore_tuya_set_time, fz.ignore_time_read],
    toZigbee: [legacy.toZigbee.R7049_silenceSiren, legacy.toZigbee.R7049_testAlarm, legacy.toZigbee.R7049_alarm],
    exposes: [e.battery_low(),
        e.binary('smoke', ea.STATE, true, false).withDescription('Smoke alarm status'),
        e.binary('test_alarm', ea.STATE_SET, true, false).withDescription('Test alarm'),
        e.enum('test_alarm_result', ea.STATE, ['checking', 'check_success', 'check_failure', 'others'])
            .withDescription('Test alarm result'),
        e.enum('battery_level', ea.STATE, ['low', 'middle', 'high']).withDescription('Battery level state'),
        e.binary('alarm', ea.STATE_SET, true, false).withDescription('Alarm enable'),
        e.binary('fault_alarm', ea.STATE, true, false).withDescription('Fault alarm status'),
        e.binary('silence_siren', ea.STATE_SET, true, false).withDescription('Silence siren')],
};

module.exports = definition;
```

ljakab, 2024-05-31:
I just got one of these V2 devices, same issue. It would be nice to have support out-of-the-box, and maybe document on the device support page that there are two versions...

Koenkk, 2024-06-04:
`_TZE200_ft523twt` is supported out-of-the-box in Z2M 1.38.0, @ljakab could you provide the data/database.db entry of yours?

ljakab, 2024-06-12:
I just upgraded to 1.38.0, and indeed, after removing it, it got recognized when re-discovered. Thank you!

closed 2024-06-12 by Koenkk