Koenkk/zigbee2mqtt #21586: Device does toggle light, but throws errors and more - Device ID 3004482/3137308/3137309

exetico, 2024-02-26:
### What happened?

If I send a command, it normally works after a second or similar, but hereafter a error is thrown in the log.

I'm able to reproduce it over and over again, no matter if it's a brightness change, power-toggle or similar.

Device: https://www.zigbee2mqtt.io/devices/3004482_3137308_3137309.html#light%2520solutions-3004482%252F3137308%252F3137309

Manual in details for the hardware: https://ledpaerer.dk/media/catalog/product/attachment/Manual-engelsk.pdf

In all scenarios, the device does react fact the first time. Hereafter a error is thrown. If I try and change the state again, the light does first react to the command (actually turning off), after 10-20 secounds or similar, like it's shown in the logs.

Let me know if I can help with other details. I've also repaired the device, to another router (hue bulb instead directly to coordinator), resetted the device, and more.

[image]

I'll also try dev-branch later, and update my firmware, too.

### What did you expect to happen?

That the light would work properly for commands like toggle, brightness and more.

### How to reproduce it (minimal and precise)

I can reproduce it by toggle the light or change the brightness.

### Zigbee2MQTT version

1.35.3 commit: fe0742a

### Adapter firmware version

20220219

### Adapter

LAUNCHXL-CC26X2R1

### Setup

Docker with Zigbee2MQTT, in the same system, HA and more. The system normally works perfect with 30+ devices. No other devices have problems.

### Debug log

Here's one example, where I change the power state:
```
Debug 2024-02-26 19:08:40Received MQTT message on 'zigbee2mqtt/lightsolutions_dim_1/set' with data '{"state":"OFF"}'
Debug 2024-02-26 19:08:40Publishing 'set' 'state' to 'lightsolutions_dim_1'
Debug 2024-02-26 19:08:40Received Zigbee message from 'lightsolutions_dim_1', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Debug 2024-02-26 19:08:40Received Zigbee message from 'lightsolutions_dim_1', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":1}' from endpoint 1 with groupID 0
Info 2024-02-26 19:08:40MQTT publish: topic 'zigbee2mqtt/lightsolutions_dim_1', payload '{"brightness":254,"elapsed":21385,"linkquality":72,"power_on_behavior":null,"state":"OFF"}'
Info 2024-02-26 19:08:40MQTT publish: topic 'zigbee2mqtt/lightsolutions_dim_1', payload '{"brightness":1,"elapsed":4,"linkquality":72,"power_on_behavior":null,"state":"OFF"}'
Debug 2024-02-26 19:08:44Received MQTT message on 'zigbee2mqtt/lightsolutions_dim_1/set' with data '{"state":"ON"}'
Debug 2024-02-26 19:08:44Publishing 'set' 'state' to 'lightsolutions_dim_1'
Error 2024-02-26 19:09:03Publish 'set' 'state' to 'lightsolutions_dim_1' failed: 'Error: Command 0x385b44fffe1b3d45/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 29765 - 1 - 73 - 6 - 11 after 10000ms)'
Debug 2024-02-26 19:09:03Error: Command 0x385b44fffe1b3d45/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 29765 - 1 - 73 - 6 - 11 after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:502:47) at processTicksAndRejections (node:internal/process/task_queues:95:5) at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.command (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:746:28) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:46:17) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:1116:32) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:249:36) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)
Debug 2024-02-26 19:09:03Received Zigbee message from 'lightsolutions_dim_1', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
```

Here another example, going from "On" to "Off":
```
2024-02-26 19:17:34Publish 'set' 'state' to 'lightsolutions_dim_1' failed: 'Error: Command 0x385b44fffe1b3d45/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 29765 - 1 - 200 - 6 - 11 after 10000ms)'
Debug 2024-02-26 19:17:34Error: Command 0x385b44fffe1b3d45/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 29765 - 1 - 200 - 6 - 11 after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:502:47) at processTicksAndRejections (node:internal/process/task_queues:95:5) at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.command (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:746:28) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:46:17) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:1116:32) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:249:36) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)
Debug 2024-02-26 19:17:34Received Zigbee message from 'lightsolutions_dim_1', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2024-02-26 19:17:34MQTT publish: topic 'zigbee2mqtt/lightsolutions_dim_1', payload '{"brightness":1,"elapsed":23058,"linkquality":43,"power_on_behavior":null,"state":"ON"}'
Debug 2024-02-26 19:17:34Received Zigbee message from 'lightsolutions_dim_1', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":231}' from endpoint 1 with groupID 0
Info 2024-02-26 19:17:34MQTT publish: topic 'zigbee2mqtt/lightsolutions_dim_1', payload '{"brightness":231,"elapsed":18,"linkquality":43,"power_on_behavior":null,"state":"ON"}'on","linkquality":43,"power":9,"power_outage_memory":"restore","state":"ON","voltage":219}'
Error 2024-02-26 19:17:57Publish 'set' 'state' to 'lightsolutions_dim_1' failed: 'Error: Command 0x385b44fffe1b3d45/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 29765 - 1 - 203 - 6 - 11 after 10000ms)'
Debug 2024-02-26 19:17:57Error: Command 0x385b44fffe1b3d45/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 29765 - 1 - 203 - 6 - 11 after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:502:47) at processTicksAndRejections (node:internal/process/task_queues:95:5) at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.command (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:746:28) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:46:17) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/convert [... cut ...]
Debug 2024-02-26 19:17:57Received Zigbee message from 'lightsolutions_dim_1', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":1}' from endpoint 1 with groupID 0
Info 2024-02-26 19:17:57MQTT publish: topic 'zigbee2mqtt/lightsolutions_dim_1', payload '{"brightness":231,"elapsed":23068,"linkquality":171,"power_on_behavior":null,"state":"OFF"}'
Info 2024-02-26 19:17:57MQTT publish: topic 'zigbee2mqtt/lightsolutions_dim_1', payload '{"brightness":1,"elapsed":4,"linkquality":171,"power_on_behavior":null,"state":"OFF"}'
````

codedesperate, 2024-03-02:
I would like to add that this exact thing is happening to me as well, and I have 6 of these devices all having this behaviour. 

I'm on latest zigbee2mqtt, having it installed as a home assistant add-on. 
I'm using Sonoff ZBDongle-E on firmware 7.4.1.0

exetico, 2024-03-02:
Yes, same error as me 👍. There must be a way to enforce the custom definition for the device. To be honest, I've not found the proper amount of time to fiddle with it, after I submitted the issue.

Could you share your work in a gist, and add the reference here, too?

_If you join the official Discord, you'll find my nickname by searching after the issue (link). Fell free to add me 👍, and send a PM._

codedesperate, 2024-03-02:
I was just playing around a bit actually, as I haven't tried adding an external converter before. I wanted to get it to load the initial converter, and then expand from there. 

I took this as a reference. (The "Zigbee smart in-wall switch") https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/livingwise.ts

And then did what the guide zigbee2mqtt guide states: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

This is what I was trying to make it load:
https://gist.github.com/codedesperate/2347e499ecbb86c02d623f14d78e9fba

But even after including it in zigbee2mqtt, it would still match it with the "wrong" device after re-pairing.

skurki112, 2024-03-10:
Same problem, behaviour and error code

exetico, 2024-03-26:
So, I'm trying to update the firmware.

But I keep ending in a situation like this:

```
Debug 2024-03-26 19:01:08Received Zigbee message from 'lightsolutions_dim_1', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":2,"imageType":1000,"manufacturerCode":4107}' from endpoint 1 with groupID 0
Debug 2024-03-26 19:01:08Device 'lightsolutions_dim_1' requested OTA
Debug 2024-03-26 19:01:08Responded to OTA request of 'lightsolutions_dim_1' with 'NO_IMAGE_AVAILABLE'
```
...

```
Debug 2024-03-26 20:11:17Received Zigbee message from 'lightsolutions_dim_1', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":2,"imageType":1000,"manufacturerCode":4107}' from endpoint 1 with groupID 0
Debug 2024-03-26 20:11:17Device 'lightsolutions_dim_1' requested OTA
Debug 2024-03-26 20:11:17Responded to OTA request of 'lightsolutions_dim_1' with 'NO_IMAGE_AVAILABLE'
```

But, why can that be?

I've added this to my local ota index file:

```
[
    {
        "fileVersion": 4,
        "fileSize": 286646,
        "manufacturerCode": 4107,
        "imageType": 1000,
        "sha512": "2bc2678b5b990a5996ac55a58c81aa9510ecfdc1725d7bed00b6f84d20cd99f2ce5a143910ab6416a340a9326df61241289c834308e7025488fd380acbb87363",
        "url": "local_ota_firmwares/z3_dimmer_in_1_level_out_1_level_hzc_FW4.ota",
        "force": true
    }
]
```

Both `manufacturerCode` and `imageType` matches, and the fileVersion are higher, too. The file are located in the folder called `local_ota_firmwares`, with the name `z3_dimmer_in_1_level_out_1_level_hzc_FW4.ota`.

And if other devices asks, Zigbee2MQTT does check the index file:

```
Debug 2024-03-26 18:51:57Device 'gledopto_light1' requested OTA
Debug 2024-03-26 18:51:57OTA: Checking if an update is available for '0x84fd27fffe97fab4' (GL-B-008P)
Debug 2024-03-26 18:51:57OTA: Is new image available for '0x84fd27fffe97fab4' (GL-B-008P), current '{"fieldControl":1,"manufacturerCode":4687,"imageType":0,"fileVersion":17}'
Debug 2024-03-26 18:51:57ZigbeeOTA: Getting image metadata for 'GL-B-008P'
Debug 2024-03-26 18:51:57ZigbeeOTA: Downloaded main index
Debug 2024-03-26 18:51:57ZigbeeOTA: Loading override index '/app/data/local_ota_index_file.json'
Debug 2024-03-26 18:51:57OTA: Is new image available for '0x84fd27fffe97fab4' (GL-B-008P), latest meta '{"fileVersion":7,"fileSize":291726,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Gledopto/GL-B-008P_V17A1_OTAV7.ota","sha512":"788341b48089076a6d1c6f020f34699428adc5103ba18d54fe3b0b00bbee2212bef13ba853fe3f1c1fb247f6d17d29608a937ed897fdd9e5b5294b4ceee13ebe"}'
Debug 2024-03-26 18:51:57OTA: Update available for '0x84fd27fffe97fab4' (GL-B-008P): NO
Warning 2024-03-26 18:51:57OTA: Firmware on '0x84fd27fffe97fab4' (GL-B-008P) is newer than latest firmware online.
```

Also, check:
https://www.alibaba.com/product-detail/DK-220V-240V-250W-Fuga-Standard_1600183409775.html

Didn't notice that, last time.

codedesperate, 2024-04-24:
@exetico I tried making a converter, but it kept recognizing it as the Light Solutions wire dimmer, and I couldn't get it to work. 

Anyways..
I have no idea idea if my issue is due to this specific device, but I just wanted to report the following:

Updating to zigbee2mqtt 1.36.1 makes the devices unresponsive. Going back to 1.36.0 fixes this issues. I did read there was an issue with external converters, but this isn't a external converter right?
https://github.com/Koenkk/zigbee2mqtt/discussions/22023#discussioncomment-8983175

@exetico do you have any issues on 1.36.1?

exetico, 2024-04-26:
@codedesperate Fair. It's annoying that it doesn't work that well... I didn't manage to get it to firmware update. Even though I spend hours on that part. Personally I'm thinking about trying to see if it's possible to update it with a Hue Gateway, just to see if it acts better here, on the newer firmware.

What firmware are reported from your device?

<hr>

Regarding Zigbee2MQTT, and the question about 1.36.x.

Before update:

```
Zigbee2MQTT version
[1.36.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.36.0) commit: [86ed71c](https://github.com/Koenkk/zigbee2mqtt/commit/86ed71c)
Coordinator type
zStack3x0
Coordinator revision
20230507
Coordinator IEEE Address
0x00124b001caa682e
Frontend version
0.6.159
zigbee_herdsman_converters_version
18.42.0
zigbee_herdsman_version
0.35.1
```

After update:

```
Zigbee2MQTT version
[1.36.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.36.1) commit: [ffc2ff1](https://github.com/Koenkk/zigbee2mqtt/commit/ffc2ff1)
Coordinator type
zStack3x0
Coordinator revision
20230507
Coordinator IEEE Address
0x00124b001caa682e
Frontend version
0.6.161
zigbee_herdsman_converters_version
19.11.2
zigbee_herdsman_version
0.40.3
```

Ater I updated, I turned off the light in Home Assistant. It responded as it would normally do. Same thing with dimming and so on. Just very delayed, with the same issues as normally in zigbee2mqtt:
[image]

I'd really like to get this product supported, as it's installed in my wall... But I guess someone, or I, need to find the proper amount of time, to get this working as expected.

codedesperate, 2024-04-26:
@exetico 
Yeah i know what you mean, i have 6 installed in my walls. Would love it to be fully supported.

I have firmware version 1.00
Builddate: 20220523

I don't see OTA ability for these devices in Zigbee2mqtt. Did you find newer firmware versions somewhere?

exetico, 2024-04-27:
Thats identical to mine, @codedesperate .

I asked the company where I ordered it from. You're more than welcome to DM me. I'm not gonna publish it here, in a public place.

Regarding 1.36.1, did you get some new issues in the logs for zigbee2mqtt, or?

codedesperate, 2024-04-27:
@exetico Good News!

You know when you turn the light on, you can't turn it off straight away, but have to wait like 20 seconds?

Well today I migrated from my Sonoff ZBdongle-E to my old Conbee 2. Now the lights reacts instantly to all my commands, also when I execute them quickly. So, it behaves like it should. However, the log still gets riddled with the error messages, but at least they "work" now. This is the specific error:

`Publish 'set' 'state' to 'Stuelyset' failed: 'Error: Command 0x385b44fffe1b3ab2/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (108))'`

Also, I flashed the latest stable firmware as be the supported adapters page, but my Aqara Roller shade drivers E1 required a new firmware to function. So I flashed latest firmware for Conbee 2, and so far it haven't given me issues, though I've only ran it for a few hours. 

I'm gonna buy a Zigstar Stick of Sonoff ZBdongle-P or whatever my googeling tells me will perform the best. 

I'm on Zigbee2mqtt 1.36.0. 1.36.1 breaks conbee 2 completely as well, every device unresponsive. But I can see on the issues page many conbee 2 users have problems with 1.36.1.

codedesperate, 2024-05-03:
Okay, I switched to smlight slzb-07 cc2652p7 today, and it's back to being buggy :( I can control the lights but only with 30 second break between each command, and the log is riddled with errors.

exetico, 2024-05-09:
Thanks for the update. I'd really like to see support for this in the future, but I'm not really sure how I can continue on my own. 

If you end up finding a Hue Bridge, just to test it after the OTA, it would be great to hear if thinks works as expected after that.

codedesperate, 2024-05-09:
I ended up buying a hue bridge, and running only these devices on a hue zigbee network, and then used hue integration in homeassistant. This works without any issues, but I hope only temporarily until someone clever enough can get it to work with zigbee2mqtt.

The hue bridge is reporting firmware 1.00 | 100b-3e8 and showing up-to-date, no update available. So it doesn't seem like a a newer firmware has been officially launched.

garret, 2024-05-12:
@codedesperate could you recap your current (hopefully temporary) solution? So I have a rpi3b+ with a zigbee sonoff adapter and zigbe22mqtt + home assistant. In order to have the light dimmers working as they are supposed to be (fast and reliable) it seems you need to connect them to a philips hue bridge. If I get one, could I connect the hue bridge to the same instance of zigbee2mqtt or I need another one or I would have to set it up directly in home assistant without passing from zigbee2mqtt?

exetico, 2024-05-12:
@garret I'm sure he's just adding the Hue Bridge to Home Assistant. Nothing form the Hue Bridge it looped through Zigbee2MQTT. That's my guess.

codedesperate, 2024-05-12:
@garret and @exetico Yeah pretty much. 
I'm setting it up on the hue bridge on it's own zigbee network with the philips Hue app. Exactely as a regular Hue customer would. 

Then in Homeassistant I add the Philips Hue integration, and the hue devices are added. It dosn't touch zigbee2mqtt at all. 

It does result in two different zigbee networks which is really annoying. Though I run my WiFi on Channel 6, Philisp Hue channel 11, zigbee2mqtt channel 25. And everything works stable. 

If these devices ever get added to zigbee2mqtt though, I would retire the Hue hub, and repair them with zigbee2mqtt.

garret, 2024-08-28:
If one does not have a physical Hue bridge available, could setting up https://github.com/chrivers/bifrost be a solution? As I understand, this project tries to emulate a Philips Hue Bridge from Zigbee2mqtt

exetico, 2024-08-28:
@garret If I understand the product scope the right way, its a solution to use the Hue app, and things in that eco-system. I don't ser how it should fix hos z2m talks with the device.

With that said; I'll take a look at the suggestion next week.

exetico, 2024-08-28:
@codedesperate Did you spend some time trying and reverse the original commands sent from the Hue Bridge? I mean; It should be possible to get proper support in z2m by doing so, right?

Lyng308, 2024-10-08:
Hi
I had the same problem, but a friend of mine had an older version of this dimmer. So I copied settings in Zigbee2MQTT from his dimmer and now my dimmer works perfectly. 😊

the settings i copyed
[image]
[image]

exetico, 2024-10-08:
@Lyng308 Thank you for sharing this! 

I adjusted Reporting, but that ended in the same error:

```
z2m: Publish 'set' 'state' to 'lightsolutions_dim_1' failed: 'Error: ZCL command 0x385b44fffe1b3d45/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 48863 - 1 - 63 - 6 - 11 after 10000ms)'
```

However, adjusting the Settings (specific) did the trick!

[image]

That's awesome!

garret, 2024-10-08:
I want to confirm that I just tried to **ONLY** change the "**Settings (specific)**" as @Lyng308 wrote above and now I can control the dimmer with home assistant without any delay. I will test in the coming days as I have many of these dimmers around my house.

Lyng308, 2024-10-08:
nice to hear it work for others to  :)

the settings in reportning was only becuse HA was slow to display the changes in the dashboard when i adjusted on the dimmer :)
@exetico  @garret

garret, 2024-10-08:
@Lyng308 do you know what changing the "**Reporting**" section does? 

What I mean is if it improves for the zigbee2mqtt/home assistant working of these dimmers to also set the **Reporting** section like yours?

garret, 2024-10-08:
I can't manage to change the reporting section. I set the parameters showed in @Lyng308 image but when I go out and back again in the Reporting page, the default settings are there. I also tried by clicking "apply" to each setting but without success. Is there any special way of doing it?

Lyng308, 2024-10-08:
@garret i just typed it and use the arrows in text filed and the click apply

exetico, 2024-10-30:
I still get these a lot, and I think it actually a making noise in my network.

```
error 2024-10-30 21:25:58z2m: EventBus error 'OTAUpdate/deviceMessage': CommandResponse 0x385b44fffe1b3d45/1 genOta.queryNextImageResponse({"status":152}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225))
error 2024-10-30 21:39:15z2m: Failed to read state of 'lightsolutions_dim_1' after reconnect (ZCL command 0x385b44fffe1b3d45/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225)))
error 2024-10-30 22:09:49z2m: Failed to read state of 'lightsolutions_dim_1' after reconnect (ZCL command 0x385b44fffe1b3d45/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 48863 - 1 - 144 - 6 - 1 after 10000ms))
```

Could some of you guys confirm if you're facing the same thing? It's hard to flag for Light Solutions, if it's just a "3rdparty problem", and if it works smoothly with a Hue Gateway (I'm unable to test that).

[image]

The "Light Solution" device are the only one making this kind of problem on my network with **many** devices. A lot have been paired since, and no other device are facing any issues...

Lyng308, 2024-11-01:
@exetico have you copyed all the settings or ?

I hade the samme issue when i pair a new dimmer

exetico, 2024-11-01:
@Lyng308 Yes, same as suggested. But I'll try and repair it later today.

closed 2025-01-09 by github-actions[bot]