Koenkk/zigbee2mqtt #21592: Slow/Delayed outbound control of Zigbee devices

jwGeorge, 2024-02-27:
### What happened?

After upgrading from 1.35.1-1 to 1.35.3-1, Zigbee device control became extremely laggy and slow. 

- Messages from HA -> Zigbee devices would take between 3 and 8 seconds to be delivered. 
- Messages from Zigbee devices -> HA were fast. 

When HA is first booted, it performs as expected but within minutes the issue happens.

I tried updating adapter to ZStack 20230507 but then devices began to drop off the network, so reverted back to 20221226.
Devices only became responsive again after reverting Z2M back to version 1.35.1-1 from backup.

An example that I used to test was a two-way light switch automation.
1. Switch A is toggled and triggers automation
2. Automation toggles light (Shelly Dimmer 2 - happens instantly)
3. Automation toggles Switch B (Nue/3A Zigbee switch - delayed many seconds)

I didn't enable debug logging, but I can do so if required. 
Thanks!

### How to reproduce it (minimal and precise)

Upgrade from 1.35.1-1 to 1.35.3-1

### Zigbee2MQTT version

1.35.3-1

### Adapter firmware version

20221226

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus CC2652P

### Setup

Add-on to Home Assistant OS 10.5 running on Intel NUC

FloCodeRian, 2024-03-02:
Due to missing downgrade skills, I've instantly updated to 1.36.0 

It's getting better but most of the time all devices respond with a heavy delay- like 30s+ or the didn't respond at all. 
The map could be loaded again, which hasn't been working with 1.35.3 

I'm using the sonoff USB3.0 adapter.

Would love if someone could help me by downgrading to 1.35.1 or similar. 

I'm able to use the terminal addon and installed the HA - zigbee2mqtt just like the docs wished.

Greets 
Flo

wizardofozzie, 2024-03-03:
>I'm using the sonoff USB3.0 adapter.

P or E?

FloCodeRian, 2024-03-03:
I'm using the CC2652P ...

But after several restarts of the whole system and installing a new ikea socket instead of an old hue bulb, for range extension purposes, seem like working perfect again. 

Sry for the confusion!

jwGeorge, 2024-03-04:
**Update**: Installed 1.36.0 and seems to be working well since this morning. Messages between HA and devices are quick and no timeouts. Will monitor for the rest of the day and report back.

jwGeorge, 2024-03-25:
Working fine with no further problems. Closing Issue.

closed 2024-03-25 by jwGeorge