Koenkk/zigbee2mqtt #21604: No converter available for ‘state_L1’(“ON”)

hanzala-123, 2024-02-27:
### What happened?

Newly supported device
https://github.com/Koenkk/zigbee2mqtt/issues/21225#issuecomment-1962427705
Only the last switch works

### What did you expect to happen?

For all 6 switches to work 

### How to reproduce it (minimal and precise)

Use the zigbee2mqtt interface to try and  switch the light on and off

### Zigbee2MQTT version

1.35.3-dev

### Adapter firmware version

20210708

### Adapter

Sonoff zigbee dongle

### Setup

Homeassistant addon

hanzala-123, 2024-02-27:
> Have you tried either a lower case L ?

It is with a lower case L my bad, but this is following on from the previous issue where they added support for the device. One switch works but the others throw up the error message

sjorge, 2024-02-27:
@Koenkk I'm not familier with  TuYA device aside that they are pretty non standard zigbee... but I assume this

https://github.com/Koenkk/zigbee-herdsman-converters/blob/cb4d26d9a6fbc6816ecf95e547c1567d66448c27/src/devices/tuya.ts#L1834

Should at least map to different endpoint IDs ?

Koenkk, 2024-02-29:
@sjorge that's expected for TuYa devices, all switches are controlled through the same endpoint.

This should have been fixed in https://github.com/Koenkk/zigbee2mqtt/pull/21576 , please update to the latest dev branch.

hanzala-123, 2024-02-29:
I've updated, get it to work occasionally but I get this message 80% of the time [image]

Koenkk, 2024-03-02:
`No network route` indicates a connectivity issues, routers may not be relaying messages correctly, sometimes it also helps to re-pair the device.

closed 2024-09-29 by github-actions[bot]