Koenkk/zigbee2mqtt #21611: Battery only updates after alter friendly name

Ev15, 2024-02-28:
### What happened?

Battery value of some (for example TS0203) shows 1%. Battery is new. After changing the name of the device (friendly name) the battery shows 100%. So when i change for example ''door'' into ''door1'' the value change from 1% to 100%.

### What did you expect to happen?

Getting the right battery values after changing the name of a device.

### How to reproduce it (minimal and precise)

rename device. After it battery shows 1% change the name to anything.

### Zigbee2MQTT version

1.35.3

### Adapter firmware version

20220219

### Adapter

Sonoff dongle ZBDongle-P

### Setup

Pi4

krair, 2024-03-06:
I have exactly the same issue, however I haven't found a way to even get it to even temporarily show the updated percentage after a replacement.

I replaced one battery about a month ago which is reporting correctly (started reporting correctly just after replacement), but two others I replaced about 5 days ago and they are not updated. 

I am seeing the `battery_low` field as reporting false, so it's just the percentage that's not reading correctly.

krair, 2024-03-06:
So I set herdsmen into debug mode, and clicked `reconfigure` for the "Window Lounge" TS0203 device. The only thing I can find here is that it seems to fail to find the genPowerCfg endpoint:

```
2024-03-06T10:07:30.477Z zigbee-herdsman:controller:endpoint Bind 0xa4c1381753519a62/1 genPowerCfg from '0x00124b002a2e8f19/1'
2024-03-06T10:07:30.478Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - bindReq - {"dstaddr":53990,"srcaddr":"0xa4c1381753519a62","srcendpoint":1,"clusterid":1,"dstaddrmode":3,"dstaddress":"0x00124b002a2e8f19","dstendpoint":1}
2024-03-06T10:07:30.478Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,23,37,33,230,210,98,154,81,83,23,56,193,164,1,1,0,3,25,143,46,42,0,75,18,0,1,94]
2024-03-06T10:07:30.489Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,33,0,69]
2024-03-06T10:07:30.489Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,33,0,69]
2024-03-06T10:07:30.489Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 33 - [0] - 69
2024-03-06T10:07:30.489Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - bindReq - {"status":0}
2024-03-06T10:07:30.489Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2024-03-06T10:07:31.737Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,29,68,129,0,0,2,7,164,95,2,1,0,21,0,176,179,35,0,0,9,8,90,10,0,4,42,1,0,0,164,95,29,136]
2024-03-06T10:07:31.737Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,29,68,129,0,0,2,7,164,95,2,1,0,21,0,176,179,35,0,0,9,8,90,10,0,4,42,1,0,0,164,95,29,136]
2024-03-06T10:07:31.737Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 29 - 2 - 4 - 129 - [0,0,2,7,164,95,2,1,0,21,0,176,179,35,0,0,9,8,90,10,0,4,42,1,0,0,164,95,29] - 136
2024-03-06T10:07:31.738Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1794,"srcaddr":24484,"srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"linkquality":21,"securityuse":0,"timestamp":2339760,"transseqnumber":0,"len":9,"data":{"type":"Buffer","data":[8,90,10,0,4,42,1,0,0]}}
2024-03-06T10:07:31.740Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":90,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":1024,"dataType":42,"attrData":1}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":24484,"endpoint":2,"linkquality":21,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2024-03-06T10:07:31.744Z zigbee-herdsman:controller:endpoint DefaultResponse 0xe0798dfffe35f821/2 1794(10, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2024-03-06T10:07:31.745Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0xe0798dfffe35f821:24484/2 (0,0,2)
2024-03-06T10:07:31.745Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":24484,"destendpoint":2,"srcendpoint":1,"clusterid":1794,"transid":113,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,90,11,10,0]}}
2024-03-06T10:07:31.745Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,164,95,2,1,2,7,113,0,30,5,24,90,11,10,0,254]
2024-03-06T10:07:31.746Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:debug 2024-03-06 11:07:31: Received Zigbee message from 'Heater-Ext-Hallway', type 'attributeReport', cluster 'seMetering', data '{"instantaneousDemand":1}' from endpoint 2 with groupID 0
Zigbee2MQTT:info  2024-03-06 11:07:31: MQTT publish: topic 'zigbee2mqtt/Heater-Ext-Hallway', payload '{"energy":null,"linkquality":21,"local_temperature":15.75,"occupied_heating_setpoint":19.5,"pi_heating_demand":0,"power":1,"schneider_pilot_mode":"pilot","system_mode":"off"}'
2024-03-06T10:07:31.756Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2024-03-06T10:07:31.756Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2024-03-06T10:07:31.756Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2024-03-06T10:07:31.757Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2024-03-06T10:07:31.757Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2024-03-06T10:07:31.759Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,113,183]
2024-03-06T10:07:31.759Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,113,183]
2024-03-06T10:07:31.759Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,113] - 183
2024-03-06T10:07:31.759Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":113}
2024-03-06T10:07:31.759Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2024-03-06T10:07:40.493Z zigbee-herdsman:controller:endpoint Bind 0xa4c1381753519a62/1 genPowerCfg from '0x00124b002a2e8f19/1' failed (AREQ - ZDO - bindRsp after 10000ms)
Zigbee2MQTT:info  2024-03-06 11:07:40: Successfully configured 'Window Lounge'
```

krair, 2024-03-06:
Perhaps related to: https://github.com/Koenkk/zigbee2mqtt/issues/18447

lee-costa, 2024-03-20:
Any update on this? I also have a couple devices that refuse to update. I wanted a few batteries before realizing it simply wasn't updating.

krair, 2024-06-13:
Still an issue:
```
[2024-06-13 10:07:28] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Window-Kitchen', payload '{"battery":1,"battery_low":false,"contact":false,"last_seen":"2024-06-13T09:05:59+02:00","linkquality":167,"tamper":false,"voltage":2500}'
```
I tried following the guide [here](https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html) to see if I could at least see other data, but on creating the new external converter, it appears to use the new external converter to start, then revert to the old definitions:
```
[2024-06-13 11:41:18] debug: 	z2m:mqtt: Received MQTT message on 'homeassistant/sensor/0xa4c138013eb8652f/linkquality/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Window-Kitchen/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138013eb8652f"],"manufacturer":"Tuya","model":"Tuya Door/Window sensor (TS0203_new)","name":"Window-Kitchen","via_device":"zigbee2mqtt_bridge_0x00124b002a2e8f19"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Linkquality","object_id":"window-kitchen_linkquality","origin":{"name":"Zigbee2MQTT","sw":"1.38.0","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/Window-Kitchen","unique_id":"0xa4c138013eb8652f_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
[2024-06-13 11:41:18] debug: 	z2m:mqtt: Received MQTT message on 'homeassistant/sensor/0xa4c138013eb8652f/last_seen/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Window-Kitchen/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138013eb8652f"],"manufacturer":"Tuya","model":"Tuya Door/Window sensor (TS0203_new)","name":"Window-Kitchen","via_device":"zigbee2mqtt_bridge_0x00124b002a2e8f19"},"device_class":"timestamp","enabled_by_default":true,"entity_category":"diagnostic","icon":"mdi:clock","name":"Last seen","object_id":"window-kitchen_last_seen","origin":{"name":"Zigbee2MQTT","sw":"1.38.0","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/Window-Kitchen","unique_id":"0xa4c138013eb8652f_last_seen_zigbee2mqtt","value_template":"{{ value_json.last_seen }}"}'
[2024-06-13 11:41:23] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/0xa4c138013eb8652f/contact/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Window-Kitchen/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138013eb8652f"],"manufacturer":"TuYa","model":"Door sensor (TS0203)","name":"Window-Kitchen","via_device":"zigbee2mqtt_bridge_0x00124b002a2e8f19"},"device_class":"door","object_id":"window-kitchen_contact","origin":{"name":"Zigbee2MQTT","sw":"1.38.0","url":"https://www.zigbee2mqtt.io"},"payload_off":true,"payload_on":false,"state_topic":"zigbee2mqtt/Window-Kitchen","unique_id":"0xa4c138013eb8652f_contact_zigbee2mqtt","value_template":"{{ value_json.contact }}"}'
[2024-06-13 11:41:23] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/0xa4c138013eb8652f/battery_low/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Window-Kitchen/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138013eb8652f"],"manufacturer":"TuYa","model":"Door sensor (TS0203)","name":"Window-Kitchen","via_device":"zigbee2mqtt_bridge_0x00124b002a2e8f19"},"device_class":"battery","entity_category":"diagnostic","object_id":"window-kitchen_battery_low","origin":{"name":"Zigbee2MQTT","sw":"1.38.0","url":"https://www.zigbee2mqtt.io"},"payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/Window-Kitchen","unique_id":"0xa4c138013eb8652f_battery_low_zigbee2mqtt","value_template":"{{ value_json.battery_low }}"}'
[2024-06-13 11:41:23] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0xa4c138013eb8652f/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Window-Kitchen/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138013eb8652f"],"manufacturer":"TuYa","model":"Door sensor (TS0203)","name":"Window-Kitchen","via_device":"zigbee2mqtt_bridge_0x00124b002a2e8f19"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","object_id":"window-kitchen_battery","origin":{"name":"Zigbee2MQTT","sw":"1.38.0","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/Window-Kitchen","unique_id":"0xa4c138013eb8652f_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
[2024-06-13 11:41:23] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0xa4c138013eb8652f/voltage/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Window-Kitchen/availability","value_template":"{{ value_ [... cut ...]
[2024-06-13 11:47:22] error: 	z2m: Publish 'set' 'read' to 'Window-Kitchen' failed: 'Error: ZCL command 0xa4c138013eb8652f/1 genPowerCfg.read(["batteryPercentageRemaining"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 38875 - 1 - 19 - 1 - 1 after 10000ms)'
```

Am I missing something? Is there another way I could debug this to dig into why we aren't seeing the battery endpoint update anymore?

java-devil, 2024-06-14:
For whatever it's worth (spoiler: not much) I seem to have the same issue.

Possibly noteworthy: my window sensors seemed to permanently expose the battery levels that were true as of the moment of initial pairing... and simply never refreshed said values.

After bumping into this issue I decided to attempt short-pressing the power button and... lo and behold - "manually" refreshed battery levels (I will report back if they revert to their previous "corrupted" values as they did for @Ev15).

**So it appears there is a mechanism through which Z2M is able to infer this information. However [contrary to the documantation](https://www.zigbee2mqtt.io/devices/TS0203.html) - it does so only during "power up", and **not** recurrently.**

java-devil, 2024-06-18:
[image]

So as you can see - it's relatively safe to assume that:
- the battery level "refreshes" only once per short-button press
- tit doesn't "revert" to the previous battery level either

Soukyuu, 2024-08-06:
I just noticed that some of my TS0203 window contacts have a different zigbee manufacturer.

Those that update the battery level fine are `_TZ3000_oxslv1c9`
Those that do not update the battery level after initial battery runs out are `_TZ3000_26fmupbb`

PaulHam211, 2024-08-09:
> I just noticed that some of my TS0203 window contacts have a different zigbee manufacturer.
> 
> Those that update the battery level fine are `_TZ3000_oxslv1c9` Those that do not update the battery level after initial battery runs out are `_TZ3000_26fmupbb`

I have one with manufacturer _TZ3000_oxslv1c9 with the 1% battery issue

[image]

Ev15, 2024-08-10:
After the latest update (1.39.1), most sensors indicate either 100, 87 or 22 percent. All my sensors are of this type: TZ3000_oxslv1c9.
For example, I have 3 sensors that indicate 87 percent but I am sure the battery level is different.

java-devil, 2024-08-16:
[image]

Potential lead.

As you can see while trying to query the `batteryPercentageRemaining` attribute in the "reporting" tab I recieve a timeout that as far as I am able to tell is indicative of Z2M not being possible to query the device as long as it is in sleep mode. And since the device is a Zigbee battery powered window sensor it is pretty much always in sleep mode unless you are opening or closing the window.

However now short-pressing the "RESET" button doesn't help either :P I assume due to some recent Z2M version bump.

Now if we could only somehow query the `batteryPercentageRemaining` attribute whenever the door/window opens/closes... Any ideas anyone?

My sensor is of type: `_TZ3000_6zvw8ham`

hardwareadictos, 2024-08-18:
Same issue here, battery percentages not updating rear replacing batteries.

leonardolisboa, 2024-09-26:
I'm facing the same issue. Is there any solution or workaround? Thanks.

fernandolcardoso, 2024-11-12:
Same here:

[image]

TOPIC: zigbee2mqtt/sensor_porta_entrada

```
{
  "battery": 100,
  "battery_low": false,
  "contact": true,
  "linkquality": 65,
  "tamper": false,
  "voltage": 3000
}
```

TOPIC: homeassistant/sensor/0xa4c1389373a0bb59/battery/config

```
{
  "availability": [
    {
      "topic": "zigbee2mqtt/bridge/state",
      "value_template": "{{ value_json.state }}"
    }
  ],
  "device": {
    "identifiers": [
      "zigbee2mqtt_0xa4c1389373a0bb59"
    ],
    "manufacturer": "Tuya",
    "model": "Door sensor (TS0203)",
    "name": "Sensor Porta Entrada",
    "via_device": "zigbee2mqtt_bridge_0x00124b00258cd2d8"
  },
  "device_class": "battery",
  "enabled_by_default": true,
  "entity_category": "diagnostic",
  "object_id": "sensor_porta_entrada_battery",
  "origin": {
    "name": "Zigbee2MQTT",
    "sw": "1.41.0",
    "url": "https://www.zigbee2mqtt.io"
  },
  "state_class": "measurement",
  "state_topic": "zigbee2mqtt/sensor_porta_entrada",
  "unique_id": "0xa4c1389373a0bb59_battery_zigbee2mqtt",
  "unit_of_measurement": "%",
  "value_template": "{{ value_json.battery }}"
}
```

Some device tweaks in configuration.yaml file

```
devices:
  "0xa4c1389373a0bb59":
    friendly_name: sensor_porta_entrada
    homeassistant:
      name: Sensor Porta Entrada
      contact:
        device_class: door
```

barmazu, 2025-01-12:
I noticed the same issue. Battery died while still being reported at 30%, replaced it yesterday - after >24hrs it is still reported at 30% like nothing happened.  I double checked the battery itself and no problem there +3,2V.  Strange. (This is SNZB-04)

kaischmidt1, 2025-01-14:
I have the same problem. New battery but it still only shows 14%, for 7 days.

raffoul, 2025-03-10:
Same here. 1% with new battery.
Battery low is marked as false.

[image]

florian-asche, 2025-03-10:
Hi,

i have a lot of  _TZ3000_26fmupbb and they don't report the latest battery percentage. With latest i mean, when reconfiguring the device, and you need to power on the device while pressing the button in Z2M then the percentage gets a updated value.

I hope that helps.

florian-asche, 2025-03-10:
> Hi,
> 
> i have a lot of _TZ3000_26fmupbb and they don't report the latest battery percentage. With latest i mean, when reconfiguring the device, and you need to power on the device while pressing the button in Z2M then the percentage gets a updated value.
> 
> I hope that helps.

By the way, i have the latest version running (2.1.3) and its a fresh installation.

florian-asche, 2025-03-11:
> I just noticed that some of my TS0203 window contacts have a different zigbee manufacturer.
> 
> Those that update the battery level fine are `_TZ3000_oxslv1c9` Those that do not update the battery level after initial battery runs out are `_TZ3000_26fmupbb`

can you check the device version and configuration, is there any difference?

florian-asche, 2025-05-30:
The Tuya ZD08, TS0203, _TZ3000_7d8yme6f doesnt have that problem. There reporting battery after short amount of time.

cloudbr34k84, 2025-06-02:
> I just noticed that some of my TS0203 window contacts have a different zigbee manufacturer.
> 
> Those that update the battery level fine are `_TZ3000_oxslv1c9` Those that do not update the battery level after initial battery runs out are `_TZ3000_26fmupbb`

Add this _TZ3000_8yhypbo7 to the list of zigbee manufacturer that dont update

Soukyuu, 2025-06-10:
Actually, it seems the fix is already applied in https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts

If you search for `_TZ3000_26fmupbb`, you will see that it has the `voltageToPercentage: "3V_1500_2800"` quirk applied. It is not working, though, because the sensor still shows 1% / 2500mV in the GUI.

Soukyuu, 2025-06-10:
Sorry for the spam, but I might have a solution. Removing and pairing one of the `_TZ3000_26fmupbb` sensors again resulted in this:

[image]

Note that the voltage is not reported as 2500mV but 2800mV and the battery is also at 100% now. I will re-pair all the other sensors and keep an eye on the battery level, but it seems to be working as intended now. Just not for the already paired sensors, as the setup only happens on joining the network.

florian-asche, 2025-06-10:
On repairing or when you click reconfigure and press the button one time. Then the device reports or z2m sets the new battery state.

florian-asche, 2025-06-10:
> @florian-asche: sorry, I did not understand that. Do you mean it only updates the value on pressing the switch once or when reconfiguring, not by itself?
> 
> The ZD-08 is part of the same device definition as the other TS0203 sensors, so if you say it's working correctly, all of the variants should work the same. See below for an excerpt from the current `tuya.ts`, especially the `whiteLabel` and `battery` sections.
> 
> ```
> {
>         zigbeeModel: ["TS0203"],
>         model: "TS0203",
>         vendor: "Tuya",
>         description: "Door/window sensor",
>         fromZigbee: [fz.ias_contact_alarm_1, fz.battery, fz.ignore_basic_report, fz.ias_contact_alarm_1_report],
>         toZigbee: [],
>         whiteLabel: [
>             {vendor: "CR Smart Home", model: "TS0203"},
>             {vendor: "Tuya", model: "iH-F001"},
>             {vendor: "Tesla Smart", model: "TSL-SEN-DOOR"},
>             {vendor: "Cleverio", model: "SS100"},
>             tuya.whitelabel("Niceboy", "ORBIS Windows & Door Sensor", "Door sensor", ["_TZ3000_qrldbmfn"]),
>             tuya.whitelabel("Tuya", "ZD06", "Door window sensor", ["_TZ3000_rcuyhwe3"]),
>             tuya.whitelabel("Tuya", "ZD08", "Door sensor", ["_TZ3000_7d8yme6f"]),
>             tuya.whitelabel("Tuya", "MC500A", "Door sensor", ["_TZ3000_2mbfxlzr"]),
>             tuya.whitelabel("Tuya", "19DZT", "Door sensor", ["_TZ3000_n2egfsli"]),
>             tuya.whitelabel("Tuya", "DS04", "Door sensor", ["_TZ3000_yfekcy3n"]),
>             tuya.whitelabel("Moes", "ZSS-JM-GWM-C-MS", "Smart door and window sensor", ["_TZ3000_decxrtwa"]),
>             tuya.whitelabel("Moes", "ZSS-S01-GWM-C-MS", "Door/window alarm sensor", ["_TZ3000_8yhypbo7"]),
>             tuya.whitelabel("Moes", "ZSS-X-GWM-C", "Door/window magnetic sensor", ["_TZ3000_gntwytxo"]),
>             tuya.whitelabel("Luminea", "ZX-5232", "Smart door and window sensor", ["_TZ3000_4ugnzsli"]),
>             tuya.whitelabel("QA", "QASD1", "Door sensor", ["_TZ3000_udyjylt7"]),
>             tuya.whitelabel("Nous", "E3", "Door sensor", ["_TZ3000_v7chgqso"]),
>             tuya.whitelabel("Woox", "R7047", "Smart Door & Window Sensor", ["_TZ3000_timx9ivq"]),
>         ],
>         exposes: (device, options) => {
>             const exps: Expose[] = [e.contact(), e.battery(), e.battery_voltage()];
>             const noTamperModels = [
>                 // manufacturerName for models without a tamper sensor
>                 "_TZ3000_rcuyhwe3", // Tuya ZD06
>                 "_TZ3000_2mbfxlzr", // Tuya MC500A
>                 "_TZ3000_n2egfsli", // Tuya 19DZT
>                 "_TZ3000_yfekcy3n", // Tuya DS04
>                 "_TZ3000_bpkijo14",
>                 "_TZ3000_gntwytxo", // Moes ZSS-X-GWM-C
>                 "_TZ3000_4ugnzsli", // Luminea ZX-5232
>                 "_TZ3000_timx9ivq", //Woox R7047
>             ];
>             if (!device || !noTamperModels.includes(device.manufacturerName)) {
>                 exps.push(e.tamper());
>             }
>             const noBatteryLowModels = ["_TZ3000_26fmupbb", "_TZ3000_oxslv1c9", "_TZ3000_osu834un", "_TZ3000_timx9ivq"];
>             if (!device || !noBatteryLowModels.includes(device.manufacturerName)) {
>                 exps.push(e.battery_low());
>             }
> 
>             return exps;
>         },
>         meta: {
>             battery: {
>                 // These sensors do send a Battery Percentage Remaining (0x0021)
>                 // value, but is usually incorrect. For example, a coin battery tested
>                 // with a load tester may show 80%, but report 2.5V / 1%. This voltage
>                 // calculation matches what ZHA does by default.
>                 // https://github.com/Koenkk/zigbee2mqtt/discussions/17337
>                 // https://github.com/zigpy/zha-device-handlers/blob/c6ed94a52a469e72b32ece2a92d528060c7fd034/zhaquirks/__init__.py#L195-L228
>                 voltageToPercentage: "3V_1500_2800",
>             },
>         },
>         configure: async (device, coordinatorEndpoint) => {
>             try {
>                 const endpoint = device.getEndpoint(1);
>                 await reporting.bind(endpoint, coordinatorEndpoint, ["genPowerCfg"]);
>                 await reporting.batteryPercentageRemaining(endpoint);
>                 await reporting.batteryVoltage(endpoint);
>             } catch {
>                 /* Fails for some*/
>             }
> 
>             const endpoint = device.getEndpoint(1);
>             if (endpoint.binds.some((b) => b.cluster.name === "genPollCtrl")) {
>                 await endpoint.unbind("genPollCtrl", coordinatorEndpoint);
>             }
>         },
> ```

Normally it should update itself. But for some reason that is not working.

florian-asche, 2025-06-10:
Hi,

The Tuya ZD08, TS0203, _TZ3000_7d8yme6f updated the battery while changing the adapter from a usb one to the network based. But it only reported the same battery percentage. Now you mentioned it, i am not sure if it really pushes the new battery state or only a existing one. 
I cannot confirm this because the devices are being used at a location far away from me. 

I use the _TZ3000_26fmupbb at home and they don't report a battery change. I tested this by switching the battery. When i click reconfigure and then shortly after that press the button at the device, the battery state gets updated.

Soukyuu, 2025-06-12:
Yes indeed, switched two batteries (87% and 100% reported) and they still report the same a few days later. Sensors are operated regularly (balcony doors), so it's not an update interval issue.

I am pretty sure it used to work, however, because I saw the sensors drop from 100% to 1% over a span of 1.5 years. And then, after replacing the battery, it was stuck at 1% because of the bug.

florian-asche, 2025-06-12:
> Yes indeed, switched two batteries (87% and 100% reported) and they still report the same a few days later. Sensors are operated regularly (balcony doors), so it's not an update interval issue.
>
> I [...]

and i think, i can confirm that. I also saw it working or i can remember it working.

Today my door sensor died, no battery update since 87% till it was empty. I switched the battery and its still 87%. 
The sensor is Tuya TS0203 _TZ3000_oxslv1c9.

DominikWrobel, 2025-08-21:
same problem, the thing is it did work for a while and now it is gone in the newest version of z2m

tshan1229, 2025-10-02:
Not working for me either :(

Rosen01, 2026-01-03:
I've the same problem with 5 door sensors : 1x _TZ3000_26fmupbb and 4x _TZ3000_oxslv1c9.
I think that the issue is open in the wrong repo, should be in the [zigbee-herdsman-converters](https://github.com/Koenkk/zigbee-herdsman-converters/issues).

What I've found on the subject : 
Discussion: https://github.com/Koenkk/zigbee2mqtt/discussions/17337
PR: https://github.com/Koenkk/zigbee-herdsman-converters/pull/7681
PR: https://github.com/Koenkk/zigbee-herdsman-converters/pull/7725
ZHA handling the battery calculation: https://github.com/zigpy/zha-device-handlers/blob/c6ed94a52a469e72b32ece2a92d528060c7fd034/zhaquirks/__init__.py#L195-L228
Integration of the sensor in deCONZ REST plugin : https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5865

For me, this two models (_TZ3000_26fmupbb, _TZ3000_oxslv1c9) should use the percentage (divided by 2, see deCONZ integration) sent by the sensor and not the one calculated on the battery power. Or there's a problem with the calculation and it is not working as expected.

maxifly, 2026-02-15:
Same problem

[image]

Soukyuu, 2026-04-17:
@florian-asche: someone who knows which repo to look in would have to do it

Soukyuu, 2026-08-18:
Shush, stale-bot.

florian-asche, 2026-10-24:
Still an issue.

java-devil, 2026-10-28:
To anyone who must know, a workaround that worked on 8 of my 8x TS0203  devices:
1) open/close the window/door to wake the device
2) immediately send `{"read":{"cluster":"genPowerCfg","attributes":["batteryVoltage"]}}` to `zigbee2mqtt/<name>/set`
3) done - your battery level should immediately refresh and keep refreshing from here on

@Koenkk I assume that this should be handled  in the TS0203 definition in [zigbee-herdsman-converters](https://github.com/koenkk/zigbee-herdsman-converters)? If so I can draft up a quick PR.