Koenkk/zigbee2mqtt #21620: Aqara E1 thermostat (SRTS-A01) opens valve on occupied heating setpoint change even in system_mode == off

ladzar, 2024-02-29:
### What happened?

Thermostat turns on (opens valve) on occupied heating setpoint change even when system_mode == off but doesn't change the system_mode. So the valve is open, heater is heating, but in HA it looks like the heating is off and valve is closed.

### What did you expect to happen?

I expect either the valve to stay closed (off) or the system_mode to change to heat.

### How to reproduce it (minimal and precise)

Set the valve preset to manual, system_mode to off and change the occupied heating setpoint to temp higher than local temperature.

### Zigbee2MQTT version

1.35.3

### Adapter firmware version

20210120

### Adapter

Electrolama

### Setup

Home assistant OS + Addon

lukas-kraus, 2024-05-12:
Same problem here. Could you find any solution?

ladzar, 2024-05-13:
I did not :/. I would gladly try to fix it even, but don't know, where to begin.

lukas-kraus, 2024-05-19:
I replaced the batteries and started a new calibration - seems fine now :)

notownblues, 2024-10-20:
+1 and putting new batteries/re-calibrating didn't help

arpaterson, 2024-12-08:
I am seeing this issue as well.

1) In HA/lovelace, I set the system mode to off, the valve shows off, the system mode shows off, and the valve is closed (no flow, radiator is cold after some time).

2) I call climate.set_temperature to a new value, higher than previous, and higher than the current room temperature (i am using the external temp measurement)

3) the UI shows the system mode is OFF, but there is flow (initial large valve movement and the radiator becomes hot, and small valve positions adjustments can be heard after some time).

So at least, the system mode 'heat is not being reflected correctly in HA.

Mainly I want to understand the behaviour so I can effectively work around it (I want window sensors to set system mode 'off'/valve fully closed, and geolocation to set preset 'away'/lower setpoint if windows are closed.

It would be good if the UI tracked what the valve was actually doing correctly for a start, but better still would be if the controls were not coupled like this. But maybe thats an Aqara firmware issue.

arpaterson, 2024-12-08:
Firmware version on my three valves is Feb 2023, 0.0.0_1030

notownblues, 2025-01-08:
Same issue here. All my radiators were on tonight despite their state being off.

ladzar, 2025-01-09:
As a workaround, until this hopefuly get's fixed, I created an automation, which sets the state of each of the TRVs to heat, when it's heating setpoint changes. Not ideal, but seems to work for me now.

closed 2025-03-21 by github-actions[bot]

ku1ik, 2025-03-23:
I'm experiencing the same issue.