Koenkk/zigbee2mqtt #21625: Tuya TS0505B_1 trouble turning light on, and feature partially works

whc2001, 2024-03-01:
### What happened?

I have purchased one of [this](https://www.amazon.ca/dp/B0B2JR1DXZ?psc=1&ref=ppx_yo2ov_dt_b_product_details) Zigbee light strip controller. It pairs without problem and got recognized as a Tuya [TS0505B_1](https://www.zigbee2mqtt.io/devices/TS0505B_1.html#tuya-ts0505b_1). However when actually using it I found out several problems where one of them is serious.

After powering up, it works fine. However a few gliches:
- The `Effect` exposure does not work, select anything will just cause the light to breath once
- The `Transition` setting does not work, no matter what value I set (0, 10, etc), the turn on ramp up time is always around 1 second.

These are not critical for usage. However, the following is: When I let it idle for several hours (no exact time here, more than 6 hours maybe?), I will no longer able to turn it on. When turning on, the light will ramp up to full brightness, **then immediately turns off**. If I click the refresh button on the WebUI, it's still showing ON, but there is no light. Now, if I adjust the brightness afterwards, the light will come on and it will keep working for several hours, after which the problem will appear again.

I have tried to turn on first then adjust the brightness in the automation, however this is not working:
```yaml
  - service: light.turn_on
    metadata: {}
    data:
      brightness_pct: 99
      color_temp: 500
      transition: 0
    target:
      entity_id: light.bedside_light_strip
  - delay:
      hours: 0
      minutes: 0
      seconds: 0
      milliseconds: 500
  - service: light.turn_on
    metadata: {}
    data:
      transition: 0
      brightness_pct: 100
      color_temp: 500
    target:
      entity_id: light.bedside_light_strip
```

At this stage I'm really not sure what to do. This problem is basically rendering the controller useless.

### What did you expect to happen?

The light should be on steadily when it's turned on no matter when.

### How to reproduce it (minimal and precise)

1. Pair the controller with coordinator
2. Set color temperature and brightness
3. Turn on the switch, it lights up fine
4. Turn off and wait for ~6 hours
5. Turn on again, the LED strip ramps up to the set brightness, but then immediately goes out
6. The above state can be triggered multiple times
7. Click anywhere on the brightness slider when turned on, the strip lights up to the set brightness
8. Problem is temporarily solved until idle for another several hours

### Zigbee2MQTT version

1.35.1-dev commit: 769bc14

### Adapter firmware version

zStack3x0 20230507

### Adapter

Sonoff ZBDongle-P

### Setup

Add-on (Edge) on supervised HASS installation inside Proxmox VE, dongle passed through

### Debug log

[Attach later when the problem appears again]

whc2001, 2024-03-03:
After some further testing, I can get the light to ramp up faster by setting the `Transition` in specific settings to `0.1`. However the auto off problem is still there randomly, and each time it appears, the transition time goes back to around 1s. Tried a lot of things and it just won't work correctly...

whc2001, 2024-03-10:
I think I have found a way to get it work with Z2M (temporarily). 

You will have to set up HomeAssistant ZHA with another coordinator. Unpair the LED driver with Z2M and pair it with ZHA, turn on and off once and pair back with Z2M again. After that, it should work for simple turn on/off without automatically turning off.

However, **NEVER EVER touch anything** in the `Settings (specific)` tab in the dashboard device page, especially the `Transition` setting. If you did so, the problem will appear and you need to do the above steps again to recover. To be on the safe side, don't use the transition time argument when calling from HA either and just use the default transition time.

Maybe there is something wrong in the settings implementation?

rasika-jay, 2024-07-27:
I too have the same issue with the light strip turning off immediately after turning on. I will try removing the transition option from z2m to see if that works.

whc2001, 2024-12-30:
1

wucke13, 2025-01-08:
I have a similar problem, when once I touched the Settings (specific), I always get a transition, even if I set the setting to 0 (aka no transition).

prinze77, 2025-01-18:
+

spiran66, 2025-02-21:
Same fore me

nevembalint, 2025-04-04:
Were you able to find a proper solution?

jemmiegod, 2025-05-29:
I too am having this issue. During pairing I manually set the transition time to 2 for about 30 of these devices. I'll try the ZHA workaround and see if that resolves the issue for now.