Koenkk/zigbee2mqtt #21626: [New device support]: Aqara Ceiling Light T1 lumi.light.acn031 HCXDD12LM

banister4728, 2024-03-01:
### Link

https://homesmart.sg/product/aqara-ceiling-light-l1/

### Database entry

{"id":13,"type":"Router","ieeeAddr":"0x54ef4410008f1023","nwkAddr":52696,"manufId":4447,"manufName":"Aqara","powerSource":"Mains (single phase)","modelId":"lumi.light.acn031","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[5,4,3,0,768,8,6,64704],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"61440":1,"modelId":"lumi.light.acn031","manufacturerName":"Aqara","powerSource":4,"zclVersion":3,"appVersion":23,"stackVersion":27,"hwVersion":1,"dateCode":"20230525"}},"aqaraOpple":{"attributes":{"223":{"type":"Buffer","data":[22,33,0,0,23,33,0,0,12,33,0,0,24,33,0,0]},"229":{"type":"Buffer","data":[12,0,12,0,0,4,0,59,21,37,37,221,22,37,0,218,86,37,130,72,91,37,83,137,95,37,52,19,125,37,0,217,150,37,97,216,159,37,0,41,218,37,117,88,221,37,188,142,223,37,48]},"247":{"type":"Buffer","data":[117,32,254,115,35,153,153,204,76]}}},"lightingColorCtrl":{"attributes":{"245":13490258,"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370,"colorTemperature":217,"colorMode":2}},"genOnOff":{"attributes":{"245":13490262,"onOff":0}},"genLevelCtrl":{"attributes":{"245":13490259,"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":258,"inClusterList":[5,4,3,768,8,6,64704],"outClusterList":[],"clusters":{"lightingColorCtrl":{"attributes":{"245":47044608,"currentX":19660,"currentY":39321,"colorCapabilities":24,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":23,"stackVersion":27,"hwVersion":1,"dateCode":"20230525","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1309948793},"lastSeen":1698378541893,"defaultSendRequestWhen":"immediate"}

### Comments

Seems to be working mostly.

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;

module.exports = [
  {
    zigbeeModel: ['lumi.light.acn031'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'T1', // Vendor model number, look on the device for a model number
    vendor: 'Aqara', // Vendor of the device (only used for documentation and startup logging)
    description: 'Symphony Ceiling Light T1', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    meta: { multiEndpoint: true },
    endpoint: () => ({ white: 1, rgb: 2 }), // Endpoints need to follow defined list of names.
    fromZigbee: [
      // fromZigbee seems to be generic across all preset profiles.
      fz.color_colortemp,
      fz.on_off,
      fz.brightness,
      fz.level_config,
      fz.ignore_basic_report,
      fz.power_on_behavior,
      fz.aqara_opple,
    ],
    toZigbee: [
      // [White light with color temperature]
      tz.light_colortemp,
      tz.light_colortemp_move,
      tz.light_colortemp_step,
      // tz.light_colortemp_startup, // Disabled as it's not supported by the device
      // [Notification light]
      tz.light_color,
      tz.light_hue_saturation_move,
      tz.light_hue_saturation_step,
      tz.light_color_options,
      tz.light_color_mode,
      // [Common]
      tz.light_onoff_brightness,
      tz.ignore_transition,
      tz.ignore_rate,
      tz.light_brightness_move,
      tz.light_brightness_step,
      tz.level_config,
      tz.effect,
      tz.power_on_behavior,
      tz.xiaomi_switch_power_outage_memory,
    ],
    exposes: [
      // [Generic aqara device attributes]
      e.power_outage_memory(),
      e.device_temperature(),
      e.power_outage_count(),

      // [White light with color temperature]
      e.light_brightness_colortemp([153, 370]).removeFeature('color_temp_startup').withEndpoint('white'), // White light: On/Off, brightness, color temperature, no color_temp_startup
      // e.effect().withEndpoint('white'), // Disabled as it's not supported by the device
      // e.power_on_behavior(['off', 'on', 'toggle', 'previous']).withEndpoint('white'), // Power on behavior: Off, On, Toggle, Previous (unknown if working)

      // [Notification Light]
      e.light_brightness_colorxy().withEndpoint('rgb'), // On/Off, brightness, color
      // e.effect().withEndpoint('rgb'), // Disabled as it's not supported by the device
      // e.power_on_behavior(['off', 'on', 'toggle', 'previous']).withEndpoint('rgb'), // Power on behavior: Off, On, Toggle, Previous (unknown if working)
    ],
    configure: extend.light_onoff_brightness_colortemp_color().configure, // Use common configure method from light_onoff_brightness_colortemp_color
    //  ota: ota.zigbeeOTA, // Disabled as we are not keeping track of firmware updates on Aqara side
  },
];
```

mattipt, 2024-03-01:
I can confirm this appears to work for [Aqara Ceiling Light T1M](https://www.aqara-shop.co.uk/product/aqara-ceiling-light-t1m/) (`lumi.light.acn032`) as well.

TooBri, 2024-03-02:
> I can confirm this appears to work for [Aqara Ceiling Light T1M](https://www.aqara-shop.co.uk/product/aqara-ceiling-light-t1m/) (`lumi.light.acn032`) as well.

@mattipt How did you do it? I am unable to get it running.

mattipt, 2024-03-02:
> @mattipt How did you do it? I am unable to get it running.

@TooBri this is probably the wrong venue to try debugging specific issues; I will say though that I had to change `zigbeeModel` and comment out a few lines which I assume have to do with specific software versions (running on Home Assistant): `fz.aqara_opple`, `tz.xiaomi_switch_power_outage_memory`, `e.power_outage_memory()`, `e.device_temperature()`, `e.power_outage_count()`.

TooBri, 2024-03-02:
> @TooBri this is probably the wrong venue to try debugging specific issues; I will say though that I had to change `zigbeeModel` and comment out a few lines which I assume have to do with specific [...]

@mattipt Thanks! one step closer. Is yours respoding on the color setting? Mine isn't. Are you able to share the content of your external converter file?

Pel1can111, 2024-03-03:
> > > @mattipt How did you do it? I am unable to get it running.
> > 
> > 
> > @TooBri this is probably the wrong venue to try debugging specific issues; I will say though that I had to change `zigbeeModel` and comment out a few lines which I assume have to do with specific software versions (running on Home Assistant): `fz.aqara_opple`, `tz.xiaomi_switch_power_outage_memory`, `e.power_outage_memory()`, `e.device_temperature()`, `e.power_outage_count()`.
> 
> @mattipt Thanks! one step closer. the only strange thing now is that i can't change the color from Zigbee2MQTT out, but it is posible within the HA entity.

Thanks for this info, can also confirm this worked with the T1M. I am also seeing the issue where the colour can't be changed within Zigbee2MQTT but works fine in HA.

tomknocke, 2024-03-06:
I can confirm same for T1M acn032. Below configuration (based on 031 with disabling some lines) allows control of both lights in Home Assistance.
But no RGB control in Zigbee2Mqtt Dashboard. Besides additional options missing and also no effects.
But anyhow its a first draft to use the Light.
In case somebody has good idea to improve the converter, let me know. I can also test with my device.

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;

module.exports = [
  {
    zigbeeModel: ['lumi.light.acn032'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'T1M', // Vendor model number, look on the device for a model number
    vendor: 'Aqara', // Vendor of the device (only used for documentation and startup logging)
    description: 'Ceiling Light T1M', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    meta: { multiEndpoint: true },
    endpoint: () => ({ white: 1, rgb: 2 }), // Endpoints need to follow defined list of names.
    fromZigbee: [
      // fromZigbee seems to be generic across all preset profiles.
      fz.color_colortemp,
      fz.on_off,
      fz.brightness,
      fz.level_config,
      fz.ignore_basic_report,
      fz.power_on_behavior,
      // fz.aqara_opple,
    ],
    toZigbee: [
      // [White light with color temperature]
      tz.light_colortemp,
      tz.light_colortemp_move,
      tz.light_colortemp_step,
      // tz.light_colortemp_startup, // Disabled as it's not supported by the device
      // [Notification light]
      tz.light_color,
      tz.light_hue_saturation_move,
      tz.light_hue_saturation_step,
      tz.light_color_options,
      tz.light_color_mode,
      // [Common]
      tz.light_onoff_brightness,
      tz.ignore_transition,
      tz.ignore_rate,
      tz.light_brightness_move,
      tz.light_brightness_step,
      tz.level_config,
      tz.effect,
      tz.power_on_behavior,
      // tz.xiaomi_switch_power_outage_memory,
    ],
    exposes: [
      // [Generic aqara device attributes]
      // e.power_outage_memory(),
      // e.device_temperature(),
      // e.power_outage_count(),

      // [White light with color temperature]
      e.light_brightness_colortemp([153, 370]).removeFeature('color_temp_startup').withEndpoint('white'), // White light: On/Off, brightness, color temperature, no color_temp_startup
      // e.effect().withEndpoint('white'), // Disabled as it's not supported by the device
      // e.power_on_behavior(['off', 'on', 'toggle', 'previous']).withEndpoint('white'), // Power on behavior: Off, On, Toggle, Previous (unknown if working)

      // [Notification Light]
      e.light_brightness_colorxy().withEndpoint('rgb'), // On/Off, brightness, color
      // e.effect().withEndpoint('rgb'), // Disabled as it's not supported by the device
      // e.power_on_behavior(['off', 'on', 'toggle', 'previous']).withEndpoint('rgb'), // Power on behavior: Off, On, Toggle, Previous (unknown if working)
    ],
    configure: extend.light_onoff_brightness_colortemp_color().configure, // Use common configure method from light_onoff_brightness_colortemp_color
    //  ota: ota.zigbeeOTA, // Disabled as we are not keeping track of firmware updates on Aqara side
  },
];`

to0b, 2024-03-07:
@tomknocke Thanks!
Seems that On/Off is also not working with a hue wall switch.

banister4728, 2024-03-08:
Does anyone know what else needs to be done to get it updated into the official release?

coniferous1, 2024-03-08:
@tomknocke Added this to my z2m config and it worked properly. Thanks a bunch. 

@banister4728 a pull request has to be made to /zigbee-herdsman-converters/. The code for this should be inserted into the lumi.ts file here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/lumi.ts

You can see the instructions at the bottom here: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-2-adding-converter-s-for-your-device

banister4728, 2024-03-09:
@coniferous1 Thanks. Let me try. I've not done that before, and I won't be able to test it so hopefully I copy and paste it in correctly.

burgoyne, 2024-03-09:
> @coniferous1 Thanks. Let me try. I've not done that before, and I won't be able to test it so hopefully I copy and paste it in correctly.

Are you planning to create a PR for lumi.light.acn031 or lumi.light.acn032? The lumi.light.acn032 does not seem to be fully fleshed out as RGB control is not working within Zigbee2MQTT. I've been attempting to resolve that issue, but haven't had any luck. I will update here if I solve it.

coniferous1, 2024-03-09:
@burgoyne acn032 seems to work for me, and while not *fully* featured it can be improved upon. I'd say for now partial support is better then no support.

burgoyne, 2024-03-09:
> @burgoyne acn032 seems to work for me, and while not _fully_ featured it can be improved upon. I'd say for now partial support is better then no support.

Aside from not having the ability to set Aqara's scenes, this does allow full control of both lights through HA. If this repo allows PR's for partial support, then may as well push it through!

coniferous1, 2024-03-09:
@banister4728 If you have trouble please let me know. I didn't want to contribute your code as I wanted to be sure that you got credit.

banister4728, 2024-03-09:
@coniferous1 Actually it's fine if you do it. I don't have time to do the PR. Also, i realised my external converter for acn031 doesn't fully work with latest version. Seems like the RGB lights is controlled the same as the main light.

veryboringname, 2024-03-11:
> @coniferous1 Actually it's fine if you do it. I don't have time to do the PR. Also, i realised my external converter for acn031 doesn't fully work with latest version (1.36.0-1), I'm still on 1.33.1-1. Seems like the RGB lights is controlled the same as the main light.

I have acn02 (T1M) and 1.36.0-1 and can't control the RGB lights too. Is it meant to appear as a separate entity in HA or do I control it from the one light entity? 

The one light entity in HA has power, brightness, RGB, and color temp controls. Power turns the main light on/off, brightness adjusts the brightness of the main light, RGB changes the colour of the brightness slider (but the notification light doesn't change colour), colour temperature changes the colour of the main light. The notification light is stuck green at the moment.

banister4728, 2024-03-12:
> @coniferous1 Actually it's fine if you do it. I don't have time to do the PR. Also, i realised my external converter for acn031 doesn't fully work with latest version (1.36.0-1), I'm still on 1.33.1-1. Seems like the RGB lights is controlled the same as the main light.

@coniferous1 Seems like your file works for my T1 acn031 on 1.36.0-1 too (after changing the zigbeeModel obviously). Can I trouble you to do the PR?

banister4728, 2024-03-15:
Merged in https://github.com/Koenkk/zigbee-herdsman-converters/pull/7216

closed 2024-03-15 by banister4728

geduxas, 2024-04-22:
For RGB led stripe, i have some idea, has anyone tried to mix support from T1 LED strip to this lamp? Don't aquare use same control logic for led strip effect's ?
It seams https://github.com/Koenkk/zigbee2mqtt/issues/18471#issuecomment-1873575860 allready have effect's enabled, maybe same control logic should be and for this lamp?

pikausp, 2024-04-30:
Hey guys, was anyone able to control the color of the secondary light from z2m?

cmsimike, 2024-05-05:
@coniferous1 Sorry what do you mean by "Right now it works through HA but not z2m" ?
I have Home Assistant setup with z2m.

geduxas, 2024-05-05:
> @coniferous1 Sorry what do you mean by "Right now it works through HA but not z2m" ? I have Home Assistant setup with z2m.

RGB ring and  main light control works from HA interface, but not from z2m web interface..

zen2, 2024-05-08:
I join this thread as I'm really interested to acquire this ceiling light and use the secondary RGB light as indicator and alerts.
I wonder if is it possible to find the commands to control the secondary light by studying the "Cluster" and send commands on the "Dev Console" on the the zigbee2mqtt front end.