Koenkk/zigbee2mqtt #21643: Aqara Pet Feeder sometimes loses its schedule when Z2M is restarted

rccoleman, 2024-03-02:
### What happened?

I have 3 separate daily feeding times configured in the Aqara Pet Feeder ([ZNCWWSQ01LM](https://www.zigbee2mqtt.io/devices/ZNCWWSQ01LM.html#aqara-zncwwsq01lm)) through Z2M, and it seems like it sometimes loses its local schedule when I restart Z2M.  This was also reported here: https://community.home-assistant.io/t/aqara-pet-feeder-in-z2mqtt-wrong-schedule-after-reboot/689354

As I wrote there, sometimes the schedule disappears in the Z2M UI, and then the schedule reappears if I switch between tabs and reload the page.  I suspect that there may be a race condition that causes the initial configuration that Z2M performs at startup to sometimes send an empty schedule even when Z2M does appear to have the proper schedule based on the UI.

It's now in a state where I can't get it to feed automatically based on a schedule that I set regardless of whether I change or add a time and hit apply or force a reconfiguration.  It just seems completely unreliable.

### What did you expect to happen?

I expect the device to feed based on the schedule that I set in Z2M.

### How to reproduce it (minimal and precise)

It's hard to say exactly what causes it, but it definitely seems related to restarting Z2M.

### Zigbee2MQTT version

1.36.0-dev commit: 38ed953

### Adapter firmware version

20221226

### Adapter

Tubeszb USB ZStack adapter

### Setup

Container installation

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14467425/log.txt)

Klikini, 2024-03-02:
This also happens to me, but I’m using a ConBee II.

rccoleman, 2024-03-02:
I wonder if it's related to https://github.com/nurikk/zigbee2mqtt-frontend/issues/1939, which calls out a mismatch between the frontend and backend.  I found that while trying to debug this issue and I noticed that I couldn't find the `manuSpecificLumi` cluster in the Dev Console for the Pet Feeder, but it was clearly listed in the cluster info tab.

j0n17, 2024-05-31:
Having similar issues of timeouts when trying to setup the schedule or the LED state.
(I'm using a Sonoff ZBDongle-E Dongle Plus)

dannytranlx, 2024-06-14:
FWIW I have the same clock offset issue using the native Aqara G3 camera hub and the feeder. It often happens when I turn on and off the camera on a schedule. But if I leave it on, I had better experience. This could be hardware-related!

closed 2025-01-07 by github-actions[bot]