Koenkk/zigbee2mqtt #21646: Covers () stopped working since v1.36.0

elad-bar, 2024-03-02:
### What happened?

Since v1.36.0 all my covers stopped working - position / open / close / stop actions are not working,
converters were not change for the devices i use so i susspect something in the logic behind mapping or general in Z2M caused it

Devices (Tuya):
- _TZE200_p6vz3wzt
- _TZE200_jhkttplm
- _TZE200_5nldle7w

### What did you expect to happen?

actions to work

### How to reproduce it (minimal and precise)

click on one of the actions leads to error as described above

### Zigbee2MQTT version

1.36.0 commit: 86ed71c

### Adapter firmware version

7.3.1.0 build 176

### Adapter

SLZB-06M

### Setup

Docker container

### Debug log

03/02/2024 13:42:32 Zigbee2MQTT:info  2024-03-02 13:42:32: MQTT publish: topic 'zigbee2mqtt/**REDACTED**', payload '{"backlight_mode":"OFF","backlight_mode_l1":"OFF","last_seen":1709379752597,"linkquality":124,"power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","state_l1":"OFF","state_l2":"OFF"}'
03/02/2024 13:42:32 Zigbee2MQTT:info  2024-03-02 13:42:32: MQTT publish: topic 'zigbee2mqtt/**REDACTED**', payload '{"backlight_mode":"OFF","backlight_mode_l1":"OFF","last_seen":1709379752597,"linkquality":124,"power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","state_l1":"OFF","state_l2":"OFF"}'
03/02/2024 13:42:32 Zigbee2MQTT:error 2024-03-02 13:42:32: Publish 'set' 'state' to 'Living Room Window and Lights' failed: 'Error: Value: 'STOP' not found in: [ON, OFF]'
03/02/2024 13:42:32 Zigbee2MQTT:info  2024-03-02 13:42:32: MQTT publish: topic 'zigbee2mqtt/**REDACTED**', payload '{"backlight_mode":"OFF","backlight_mode_l1":"OFF","last_seen":1709379752813,"linkquality":124,"power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","state_l1":"OFF","state_l2":"OFF"}'
03/02/2024 13:42:32 Zigbee2MQTT:info  2024-03-02 13:42:32: MQTT publish: topic 'zigbee2mqtt/**REDACTED**', payload '{"backlight_mode":"OFF","last_seen":1709379752859,"linkquality":188,"power_on_behavior":"previous","state":"OFF"}'
03/02/2024 13:42:33 Zigbee2MQTT:error 2024-03-02 13:42:33: Publish 'set' 'position' to 'Living Room Window and Lights' failed: 'Error: No datapoint defined for 'position''

Koenkk, 2024-03-02:
Could you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

elad-bar, 2024-03-02:
Hi, sure, sorry for not providing it originally,
if you need the entire log can zip and upload, extracted the messages related to the actions

Backlight failure on startup:
```
03/02/2024 19:27:57 Zigbee2MQTT:debug 2024-03-02 19:27:57: Error: No datapoint defined for 'backlight_mode'
03/02/2024 19:27:57 at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:1149:27)
03/02/2024 19:27:57 at Publish.onMQTTMessage (/app/lib/extension/publish.ts:259:52)
03/02/2024 19:27:57 at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:23)
03/02/2024 19:27:57 at EventEmitter.emit (node:events:529:35)
03/02/2024 19:27:57 at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:115:22)
03/02/2024 19:27:57 at MQTT.onMessage (/app/lib/mqtt.ts:141:27)
03/02/2024 19:27:57 at MqttClient.emit (node:events:517:28)
03/02/2024 19:27:57 at handlePublish (/app/node_modules/mqtt/src/lib/handlers/publish.ts:172:11)
03/02/2024 19:27:57 at handle (/app/node_modules/mqtt/src/lib/handlers/index.ts:29:17)
03/02/2024 19:27:57 at work (/app/node_modules/mqtt/src/lib/client.ts:726:17)
```

Click on OPEN:
```
03/02/2024 19:29:29 Zigbee2MQTT:debug 2024-03-02 19:29:29: Received MQTT message on 'zigbee2mqtt/Home Office Window and Lights/set' with data '{"state":"OPEN"}'
03/02/2024 19:29:29 Zigbee2MQTT:debug 2024-03-02 19:29:29: Publishing 'set' 'state' to 'Home Office Window and Lights'
03/02/2024 19:29:29 Zigbee2MQTT:error 2024-03-02 19:29:29: Publish 'set' 'state' to 'Home Office Window and Lights' failed: 'Error: Value: 'OPEN' not found in: [ON, OFF]'
03/02/2024 19:29:29 Zigbee2MQTT:debug 2024-03-02 19:29:29: Error: Value: 'OPEN' not found in: [ON, OFF]
03/02/2024 19:29:29 at Object.getFromLookup (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:580:15)
03/02/2024 19:29:29 at Object.to (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:434:53)
03/02/2024 19:29:29 at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:1153:57)
03/02/2024 19:29:29 at Publish.onMQTTMessage (/app/lib/extension/publish.ts:259:52)
03/02/2024 19:29:29 at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:23)
03/02/2024 19:29:29 at EventEmitter.emit (node:events:529:35)
03/02/2024 19:29:29 at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:115:22)
03/02/2024 19:29:29 at MQTT.onMessage (/app/lib/mqtt.ts:141:27)
03/02/2024 19:29:29 at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:130:27)
03/02/2024 19:29:29 at WebSocket.emit (node:events:517:28)
```

Click on CLOSE:
```
03/02/2024 19:29:30 Zigbee2MQTT:debug 2024-03-02 19:29:30: Received MQTT message on 'zigbee2mqtt/Home Office Window and Lights/set' with data '{"state":"CLOSE"}'
03/02/2024 19:29:30 Zigbee2MQTT:debug 2024-03-02 19:29:30: Publishing 'set' 'state' to 'Home Office Window and Lights'
03/02/2024 19:29:30 Zigbee2MQTT:error 2024-03-02 19:29:30: Publish 'set' 'state' to 'Home Office Window and Lights' failed: 'Error: Value: 'CLOSE' not found in: [ON, OFF]'
03/02/2024 19:29:30 Zigbee2MQTT:debug 2024-03-02 19:29:30: Error: Value: 'CLOSE' not found in: [ON, OFF]
03/02/2024 19:29:30 at Object.getFromLookup (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:580:15)
03/02/2024 19:29:30 at Object.to (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:434:53)
03/02/2024 19:29:30 at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:1153:57)
03/02/2024 19:29:30 at Publish.onMQTTMessage (/app/lib/extension/publish.ts:259:52)
03/02/2024 19:29:30 at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:23)
03/02/2024 19:29:30 at EventEmitter.emit (node:events:529:35)
03/02/2024 19:29:30 at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:115:22)
03/02/2024 19:29:30 at MQTT.onMessage (/app/lib/mqtt.ts:141:27)
03/02/2024 19:29:30 at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:130:27)
03/02/2024 19:29:30 at WebSocket.emit (node:events:517:28)
```

Click on STOP:
```
03/02/2024 19:29:30 Zigbee2MQTT:debug 2024-03-02 19:29:30: Received MQTT message on 'zigbee2mqtt/Home Office Window and Lights/set' with data '{"state":"STOP"}'
03/02/2024 19:29:30 Zigbee2MQTT:debug 2024-03-02 19:29:30: Publishing 'set' 'state' to 'Home Office Window and Lights'
03/02/2024 19:29:30 Zigbee2MQTT:error 2024-03-02 19:29:30: Publish 'set' 'state' to 'Home Office Window and Lights' failed: 'Error: Value: 'STOP' not found in: [ON, OFF]'
03/02/2024 19:29:30 Zigbee2MQTT:debug 2024-03-02 19:29:30: Error: Value: 'STOP' not found in: [ON, OFF]
03/02/2024 19:29:30 at Object.getFromLookup (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:580:15)
03/02/2024 19:29:30 at Object.to (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:434:53)
03/02/2024 19:29:30 at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:1153:57)
03/02/2024 19:29:30 at Publish.onMQTTMessage (/app/lib/extension/publish.ts:259:52)
03/02/2024 19:29:30 at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:23)
03/02/2024 19:29:30 at EventEmitter.emit (node:events:529:35)
03/02/2024 19:29:30 at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:115:22)
03/02/2024 19:29:30 at MQTT.onMessage (/app/lib/mqtt.ts:141:27)
03/02/2024 19:29:30 at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:130:27)
03/02/2024 19:29:30 at WebSocket.emit (node:events:517:28)
```

Change POSITION:
```
[... cut ...]
03/02/2024 19:29:32 at EventEmitter.emit (node:events:529:35)
03/02/2024 19:29:32 at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:115:22)
03/02/2024 19:29:32 at MQTT.onMessage (/app/lib/mqtt.ts:141:27)
03/02/2024 19:29:32 at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:130:27)
03/02/2024 19:29:32 at WebSocket.emit (node:events:517:28)
03/02/2024 19:29:32 at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1209:20)
03/02/2024 19:29:32 at Receiver.emit (node:events:517:28)
```

elad-bar, 2024-03-02:
checked again, seems that `_TZE200_p6vz3wzt` is working - this is more simple component - just cover,
the other 2 are cover + 1/2 switch and are throwing the errors,
if it helps, i noticed that when i restart z2m, for the `_TZE200_p6vz3wzt` i have log message of:
```
03/02/2024 19:42:33 Zigbee2MQTT:debug 2024-03-02 19:42:33: Received Zigbee message from 'Home Office Window', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":70}' from endpoint 1 with groupID 0
03/02/2024 19:42:33 Zigbee2MQTT:info  2024-03-02 19:42:33: MQTT publish: topic 'zigbee2mqtt/Home Office Window', payload '{"backlight_mode":"ON","calibration":null,"child_lock":null,"last_seen":1709401353381,"linkquality":152,"motor_steering":null,"position":30,"state":"STOP"}'
```

for the other type i couldn't find any log related to it on startup

Koenkk, 2024-03-02:
Fixed it in https://github.com/Koenkk/zigbee2mqtt/issues/21646

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-03-02 by Koenkk

elad-bar, 2024-03-03:
thanks, it works great.