Koenkk/zigbee2mqtt #21647: Some thermostat report state as separate entity

joaquinvacas, 2024-03-02:
### What happened?

Some Thermostats, like mines ( https://www.zigbee2mqtt.io/devices/TYBAC-006.html#tuya-tybac-006 ) report state as a separate entity.

While it may look like nothing, this is some of a dealbreaker for some cases like Areas treating them as normal switches while they are not.
They are being stuck in a "System mode" even if they are off, while other thermostats (the older Z-Wave ones I had) show off as another mode, which usually is the normal way of working.

Maybe some screenshots will help:

This is what's expected:

[image]
[image]

And this is what happens right now, there's no "off" mode:

[image]

While its state is "off" right now.

[image]

And while maybe in a fancoil thermostat doesn't seem like too much due to having 3 status (plus off), on a standard thermostat that does only heating, the standard status should be:

[image]

And now looks like this:
[image]

Even if it's state is off.

[image]

With no way of turning it off easily using the climate service, where it belongs.

### What did you expect to happen?

State should be part of "System mode"

[image]

Instead of being a single entity

[image]

So it can be controlled as a whole climate entity and therefore, fixing the climate card's way of working (treating off as another mode, which is what the device does physically, you can't have heat and be turned off at the same time!)

[image]

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

6.7.8.0 build 373

### Adapter

LAN EZSP, but not adapter related

### Setup

Add-on on Home Assistant

lumilooms, 2024-05-04:
The BAC-006 WiFi version and Zigbee climate thermostat looks like this in Home Assistant. 

**WiFi - BAC-006**

[image]

[image]

**Zigbee - BAC-006**

[image]
[image]

As joaquinvacas request, State should be part of "System mode" in the Zigbee version. This messes up the climate thermostat function as the BAC-006 is forever turned on in the climate card.

joaquinvacas, 2024-05-04:
Reported this issue two months ago and it's still not fixed.

I'm not a developer (DevOps mainly, not into JS/TS) but I may have a look into the zigbee-herdsman-converters code and try to make a Pull Request for that :)

joaquinvacas, 2024-05-04:
Okay @Koenkk I don't want to mess up with PR requests on herdsman-converters (and I just have my work machine right now with me so no GH login and blahblah, if I can try the damn converter I may do the pull request when I have my personal laptop) and you may know the best way of proceeding with this:

So there area a lot of already-working thermostats

[image]

The changes I made were those ones that I've took from BAC-003 which has the off mode and datapoints seems to be the same for TY-BAC006.

I've tried to create an external converter but I can't manage to get it working since tuya.ts is TypeScript and Z2M expects JS, which is out of scope for me :)

Also, I'm not sure about the differences between "BAC-003" and "TYBAC-006" so maybe is the same device in the end?
I'm not sure if there any more devices that have the same issue.

Koenkk, 2024-05-05:
@joaquinvacas what you can do to try other converters is to stop z2m, change the `manufacturerName` of your device to the one of the other definition and start z2m. (no external converters required 😄 )

joaquinvacas, 2024-05-26:
> @joaquinvacas what you can do to try other converters is to stop z2m, change the `manufacturerName` of your device to the one of the other definition and start z2m. (no external converters required 😄 )

Did it.

Now recognizes it as BAC-003:
[image]

And as expected, it does report "off" as another system_mode:
[image]

Therefore, it's possible to turn it off from the Home Assistant climate cards:

[image]
[image]

And can confirm it's working fine :smile:

joaquinvacas, 2024-06-08:
> @joaquinvacas what you can do to try other converters is to stop z2m, change the `manufacturerName` of your device to the one of the other definition and start z2m. (no external converters required 😄 )

Well, after some testing, I can say that _system_mode_ is the correct one being used on BAC-003 but the rest of the exposed entities need the BAC-006 ones, as BAC-003 is a more limited device, so a mix between BAC-003 system_mode and the rest of the already working BAC-006 features :)

Koenkk, 2024-06-10:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/a7a69ac8ef4294e2122217443449d9af): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

lumilooms, 2024-06-13:
Thanks for the file. I've tried it but it does not work for me :(
Note: This is my first time using an external converter, so I'm not sure if I've done it correctly.
I stopped and restarted z2mqtt and also rebooted home assistant. 

[image]
[image]

z2mqtt log:
[2024-06-13 20:02:03] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Zigbee BAC-006', payload '{"child_lock":"UNLOCK","current_heating_setpoint":15,"deadzone_temperature":1,"eco_mode":"OFF","fan_mode":"high","linkquality":54,"local_temperature":28,"local_temperature_calibration":0,"manual_mode":"ON","max_temperature":35,"max_temperature_limit":15,"min_temperature":15,"min_temperature_limit":24,"schedule_friday":"06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0","schedule_monday":"06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0","schedule_saturday":"06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0","schedule_sunday":"06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0","schedule_thursday":"06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0","schedule_tuesday":"06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0","schedule_wednesday":"06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0","state":"ON","system_mode":"cool","valve":"OPEN"}'

DBunevich, 2024-07-02:
> Thanks for the file. I've tried it but it does not work for me :( Note: This is my first time using an external converter, so I'm not sure if I've done it correctly. I stopped and restarted z2mqtt [...]

Can you please advise if you were able to fix it?
I've done the same, but nothing has happened.

pcrednet, 2024-09-13:
> Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/a7a69ac8ef4294e2122217443449d9af):
>
> * save this as file next to `configuration.yaml` as [...]

Hello, good morning.

@Koenkk  I confirm that your [converter](https://gist.github.com/Koenkk/a7a69ac8ef4294e2122217443449d9af) works with the bac-006.

the model I have is this one:, i onlye changed this value.

_TZE204_mpbki2zm

But only works from Z2m to device. When i switch off device, in z2m only change state to off, system mode don't change.

[image]

[image]

Thx

DBunevich, 2024-09-13:
> But only works from Z2m to device. When i switch off device, in z2m only change state to off, system mode don't change.

I have the same problem.

pcrednet, 2024-09-13:
Another bug I see is that home assistant does not show when it is cooling or heating.

It only shows that the valve entity is open or close, but home assistant climate does not see that.
[image]

Is there any way to indicate if it is heating or cooling from Z2M?

Thanks

joaquinvacas, 2024-11-24:
> Just came across this issue randomly, I noticed BAC-003 is referenced.
> 
> The reason why BAC-003 works in the climate card is because I encountered the same issue and implemented a virtual `off` system_mode, which is the behavior you are seeing.
> 
> You can find the pull request here: [Koenkk/zigbee-herdsman-converters#7257](https://github.com/Koenkk/zigbee-herdsman-converters/pull/7257)
> 
> Doing the same for the BAC-006 module should fulfill your requirements.

@Koenkk Replacing on the .db/config files my model with this allows me to turn it off using system_mode (but messes up with some other config available on BAC-006)

Your converter some comments up there seems to not be working for me :S

pcrednet, 2024-11-24:
we have an installation with 30 bac-006 with @koenkk's converter.
we have had to make several networks because the flow of messages is very high and putting more than 10 in one network makes it saturate and block the coordinator.
we have tried with slzb-06, slzb-06P7, several channels, several locations... and it is random, there are devices that one day do not saturate and others if... I have not managed to see what makes that happen.

Is there any way to indicate that the thermostat does not send so much information?

joaquinvacas, 2024-11-29:
> we have an installation with 30 bac-006 with @Koenkk's converter. we have had to make several networks because the flow of messages is very high and putting more than 10 in one network makes it [...]

I don't think this is related to this issue.

Anyhow, you should check debounce and/or throttle options.

joaquinvacas, 2025-01-30:
So is this fixed @Koenkk @robvanoostenrijk ?

Then, will close as soon as I can test it 😉

robvanoostenrijk, 2025-01-30:
> So is this fixed [@Koenkk](https://github.com/Koenkk) [@robvanoostenrijk](https://github.com/robvanoostenrijk) ?
> 
> Then, will close as soon as I can test it 😉

It should be included in the next zigbee-herdsman-converters npm package. If you are not on dev branch you have to wait until next zigbee2mqtt release.

Koenkk, 2025-02-01:
It is in the 2.1.0 release! Assuming this can be closed now, thanks @robvanoostenrijk

closed 2025-02-01 by Koenkk

rodrigogbs, 2025-02-01:
Hi, I agree with the expected behavior of the CLIMATE entity, showing the correct status in the thermostat card, but, is it possible just to keep the SWITCH entity and make it be syncronized with the CLIMATE entity status?

In other words, if I turn the thermostat on via CLIMATE entity, it turns the SWITCH entity on automatically, and viceversa.

I am asking this because I have an installation with 5 coordinators and more than 50 thermostats in all of them, a very complex environment to change everything right now.

Thanks in advance!

robvanoostenrijk, 2025-02-02:
@rodrigogbs,

I'll see if I can create a quick external converter version for you which does just that. I'd normally not be opposed to exposing the switch entity as well, but I had problems with exposing the device to matter and it showing as a switch instead of a climate control when doing that. So I'd rather not change it upstream.

If you want to take a look yourself, there is an other user who left a pretty decent external converter approach here:
https://github.com/Koenkk/zigbee2mqtt/issues/25718#issuecomment-2612972603
You can easily adapt that with your change.

joaquinvacas, 2025-02-02:
> Hi, I agree with the expected behavior of the CLIMATE entity, showing the correct status in the thermostat card, but, is it possible just to keep the SWITCH entity and make it be syncronized with [...]

https://imgs.xkcd.com/comics/workflow.png

rodrigogbs, 2025-02-02:
I really didn't understand your intention with your comment @joaquinvacas 

You did a change by yourself in the equipment behavior and are judging third part needs?

🤦🏼‍♂️

This is not a community expected behavior...

joaquinvacas, 2025-02-02:
> I really didn't understand your intention with your comment [@joaquinvacas](https://github.com/joaquinvacas)
> 
> You did a change by yourself in the equipment behavior and are judging third part needs?
> 
> 🤦🏼‍♂️
> 
> This is not a community expected behavior...
> 
> By default, I am managing the switch entity to turn on / off all the thermostats daily, in more than 200 automations...and out of nowhere there is no more the switch entity, because you decided to end up with this...
> 
> There is no conflict if the behavior that I described be implemented, no problem returning...just attending to different needs, sir!

Ah, real sorry to bother. 
Internet talk makes everything worse as sounds aggressive and/or without "tone".

It was just a joke that remind me of that specific xkcd comic, nothing to be offended as it was not the case, not judging or anything else, for real. ❤️

rodrigogbs, 2025-02-02:
Hi @robvanoostenrijk i am currently using model TYBAC-006 manufacturer _TZE204_mpbki2zm

robvanoostenrijk, 2025-02-02:
> TYBAC-006

Give the following external converter a try. I do not have a BAC-006, so I can't test it sadly.
Mind that at some point you'd have to update your automations though in order to keep up with Z2M updates.

    const exposes = require('zigbee-herdsman-converters/lib/exposes');
    const e = exposes.presets;
    const ea = exposes.access;
    const tuya = require('zigbee-herdsman-converters/lib/tuya');

    const definition = {
        fingerprint: tuya.fingerprint('TS0601', ['_TZE204_mpbki2zm']),
        model: 'TYBAC-006',
        vendor: 'Tuya',
        description: 'Wall-mount thermostat for 2-pipe fan-coil unit',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        onEvent: tuya.onEventSetLocalTime,
        configure: tuya.configureMagicPacket,
        exposes: [
            e.binary('state', ea.STATE_SET, 'ON', 'OFF').withDescription('Turn the thermostat ON/OFF'),
            e.child_lock(),
            e
                .climate()
                .withLocalTemperature(ea.STATE)
                .withSetpoint('current_heating_setpoint', 5, 35, 1, ea.STATE_SET)
                .withSystemMode(['off', 'cool', 'heat', 'fan_only'], ea.STATE_SET)
                .withFanMode(['low', 'medium', 'high', 'auto'], ea.STATE_SET)
                .withLocalTemperatureCalibration(-5, 5, 0.5, ea.STATE_SET),
            e.min_temperature().withValueMin(5).withValueMax(15),
            e.max_temperature().withValueMin(15).withValueMax(45),
            e.binary('eco_mode', ea.STATE_SET, 'ON', 'OFF').withDescription('ECO mode ON/OFF'),
            e.max_temperature_limit().withDescription('ECO Heating energy-saving temperature (default: 20 ºC)').withValueMin(15).withValueMax(30),
            e.min_temperature_limit().withDescription('ECO Cooling energy-saving temperature (default: 26 ºC)').withValueMin(15).withValueMax(30),
            e.deadzone_temperature().withValueMin(0).withValueMax(5).withValueStep(1),
            e.binary('valve', ea.STATE, 'OPEN', 'CLOSE').withDescription('3-Way Valve State'),
            e.binary('manual_mode', ea.STATE_SET, 'ON', 'OFF').withDescription('Manual = ON or Schedule = OFF'),
            ...tuya.exposes.scheduleAllDays(ea.STATE_SET, 'HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C'),
        ],
        meta: {
            tuyaDatapoints: [
                [
                    1,
                    null,
                    {
                        from: (v, meta) => {
                            return v === true
                                ? {state: 'ON', system_mode: meta.state.system_mode_device ? meta.state.system_mode_device : 'cool'}
                                : {state: 'OFF', system_mode: 'off'};
                        },
                    },
                ],
                [
                    null,
                    'system_mode',
                    {
                        // Extend system_mode to support 'off' in addition to 'cool', 'heat' and 'fan_only'
                        to: async (v, meta) => {
                            const entity = meta.device.endpoints[0];

                            // Power State
                            await tuya.sendDataPointBool(entity, 1, v !== 'off', 'dataRequest', 1);

                            switch (v) {
                                case 'cool':
                                    await tuya.sendDataPointEnum(entity, 2, 0, 'dataRequest', 1);
                                    break;
                                case 'heat':
                                    await tuya.sendDataPointEnum(entity, 2, 1, 'dataRequest', 1);
                                    break;
[... 22 lines cut ...]
                [4, 'eco_mode', tuya.valueConverter.onOff],
                [16, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
                [19, 'max_temperature', tuya.valueConverter.divideBy10],
                [24, 'local_temperature', tuya.valueConverter.divideBy10],
                [26, 'min_temperature', tuya.valueConverter.divideBy10],
                [27, 'local_temperature_calibration', tuya.valueConverter.localTemperatureCalibration],
                [28, 'fan_mode', tuya.valueConverterBasic.lookup({ low: tuya.enum(0), medium: tuya.enum(1), high: tuya.enum(2), auto: tuya.enum(3) })],
                [36, 'valve', tuya.valueConverterBasic.lookup({ OPEN: 0, CLOSE: 1 })],
                [40, 'child_lock', tuya.valueConverter.lockUnlock],
                [103, 'deadzone_temperature', tuya.valueConverter.raw],
                [104, 'min_temperature_limit', tuya.valueConverter.divideBy10],
                [105, 'max_temperature_limit', tuya.valueConverter.divideBy10],
                [106, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
                [107, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
                [108, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
                [109, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
                [110, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
                [111, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
                [112, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(1)],
                [101, 'manual_mode', tuya.valueConverter.onOff],
            ],
        },
    };

    module.exports = definition;

rodrigogbs, 2025-02-02:
> Give the following external converter a try. I do not have a BAC-006, so I can't test it sadly.
> Mind that at some point you'd have to update your automations though in order to keep up with Z2M [...]

Thanks a lot!

I will give a try tomorrow in a test environment and will give the feedback.

pcrednet, 2025-02-02:
> Hi [@robvanoostenrijk](https://github.com/robvanoostenrijk) i am currently using model TYBAC-006 manufacturer _TZE204_mpbki2zm

I have an installation with 35 devices like that.
I have it divided into 5 z2mqtt because if I put them all together in one, when reporting every few minutes, the system crashes.
One of the problems I have is that it randomly starts reporting constantly, without stopping, does something similar happen to you?

I also have automations to turn on the thermostat when the climate turns on and turn it off when it turns off because it doesn't do it on its own.
If that works now, it would be ideal to remove automations.

As soon as I can; I'll try what have put @robvanoostenrijk 

Thank you so much!

pcrednet, 2025-02-02:
> It is in the 2.1.0 release! Assuming this can be closed now, thanks [@robvanoostenrijk](https://github.com/robvanoostenrijk)

I installed version 2.1-1 today.
The thermostat TY-BAC006 is not working properly.
From HA if you turn on/off climate turns off/on the thermostat but if you do it from the thermostat nothing changes in HA, I have gone back to the previous version.

Has anyone else tried it?

Thx in advance.

robvanoostenrijk, 2025-02-03:
Looks like its having the same issue I was seeing on my BAC-003 when upgrading to the 2.x series.
I didn't do this change for the BAC-006, because I don't have one and I do not know how many other BAC-xxx have the same issue, hence did only the requested `system_mode`.

There is a new option under the metadata of the device which needs to be set: `publishDuplicateTransaction: true`.
Without this option messages with a duplicate sequenceId are ignored.

This is because the BAC-003 (and according to your feedback presumably the BAC-006 as well), sends messages out with a sequence ID of `1` repeatedly, like so:

```
// DP01: on/off
{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":1} // off
{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":1} // on
```

I updated the above (untested) external converter to set the property. Test it out?

rodrigogbs, 2025-02-03:
> Looks like its having the same issue I was seeing on my BAC-003 when upgrading to the 2.x series. I didn't do this change for the BAC-006, because I don't have one and I do not know how many other BAC-xxx have the same issue, hence did only the requested `system_mode`.
> 
> There is a new option under the metadata of the device which needs to be set: `publishDuplicateTransaction: true`. Without this option messages with a duplicate sequenceId are ignored.
> 
> This is because the BAC-003 (and according to your feedback presumably the BAC-006 as well), sends messages out with a sequence ID of `1` repeatedly, like so:
> 
> ```
> // DP01: on/off
> {"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":1} // off
> {"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":1} // on
> ```
> 
> I updated the above (untested) external converter to set the property. This additionally contains device specific configurations for exposing device state and/or heating mode.
> 
> Once tested and confirmed working I can raise a pull request for BAC-006.

@robvanoostenrijk I can confirm that with your external converter all the manual commands directly in the thermosart is working perfectly!

However, the SWITCH entity to turn it ON/OFF didn't appear...

pcrednet, 2025-02-03:
@robvanoostenrijk I can confirm that your external converter works like a charm.
it would be interesting to put this converter instead of the one in tuya.ts
The tuya.ts converter does not work with the changes made in the physical thermostat.

pcrednet, 2025-02-03:
> > Looks like its having the same issue I was seeing on my BAC-003 when upgrading to the 2.x series. I didn't do this change for the BAC-006, because I don't have one and I do not know how many other BAC-xxx have the same issue, hence did only the requested `system_mode`.
> > There is a new option under the metadata of the device which needs to be set: `publishDuplicateTransaction: true`. Without this option messages with a duplicate sequenceId are ignored.
> > This is because the BAC-003 (and according to your feedback presumably the BAC-006 as well), sends messages out with a sequence ID of `1` repeatedly, like so:
> > ```
> > // DP01: on/off
> > {"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":1} // off
> > {"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":1} // on
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > I updated the above (untested) external converter to set the property. This additionally contains device specific configurations for exposing device state and/or heating mode.
> > Once tested and confirmed working I can raise a pull request for BAC-006.
> 
> [@robvanoostenrijk](https://github.com/robvanoostenrijk) I can confirm that with your external converter all the manual commands directly in the thermosart is working perfectly syncronizing the information with HA.
> 
> However, the SWITCH entity to turn it ON/OFF didn't appear...I gave a try with a Helper and it worked like a charm, but I'll need to create more than 50 helpers for this and remember to create a new helper every single new thermostat that I install, it would me amazing to have this automatically created by Z2M.

You have an option to enable switch:

[image]

then:

[image]

Great workd @robvanoostenrijk 

thank you so much!!

pcrednet, 2025-02-03:
@robvanoostenrijk 
I checked with publishDuplicateTransaction: true and false.

I find no difference in their use

every 15 seconds it reports and 2 messages appear in the log.

`info 2025-02-03 23:17:48z2m:mqtt: MQTT publish: topic 'z2m0/termostato', payload '{"child_lock":"UNLOCK","current_heating_setpoint":20,"deadzone_temperature":1,"eco_mode":"OFF","fan_mode":"auto","last_seen":"2025-02-03T22:17:47.890Z","linkquality":134,"local_temperature":21,"local_temperature_calibration":-3,"manual_mode":"ON","max_temperature":26,"max_temperature_limit":0,"min_temperature":5,"min_temperature_limit":26,"schedule_friday":"06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0","schedule_monday":"06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0","schedule_saturday":"06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0","schedule_sunday":"06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0","schedule_thursday":"06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0","schedule_tuesday":"06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0","schedule_wednesday":"06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0","system_mode":"heat","system_mode_device":"heat","valve":"CLOSE"}'
`

`info 2025-02-03 23:17:48z2m:mqtt: MQTT publish: topic 'z2m0/termostato', payload '{"child_lock":"UNLOCK","current_heating_setpoint":20,"deadzone_temperature":1,"eco_mode":"OFF","fan_mode":"auto","last_seen":"2025-02-03T22:17:48.017Z","linkquality":131,"local_temperature":21,"local_temperature_calibration":-3,"manual_mode":"ON","max_temperature":26,"max_temperature_limit":0,"min_temperature":5,"min_temperature_limit":26,"schedule_friday":"06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0","schedule_monday":"06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0","schedule_saturday":"06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0","schedule_sunday":"06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0","schedule_thursday":"06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0","schedule_tuesday":"06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0","schedule_wednesday":"06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0","system_mode":"heat","system_mode_device":"heat","valve":"CLOSE"}'`

Is there any way that can i change reporting interval to 1 minute, for example?

Thx in advance.

robvanoostenrijk, 2025-02-04:
> However, the SWITCH entity to turn it ON/OFF didn't appear...I gave a try with a Helper and it worked like a charm, but I'll need to create more than 50 helpers for this and remember to create a new [...]

You can enable the switch entity as a device specific option:
[image]

If you want to update multiple device settings for all your existing thermostats easily, you can post the configuration updates to MQTT in a batch:

`mosquitto_pub -t 'zigbee2mqtt/bridge/request/device/options' -m '{ "id": "Airconditioning - Living", "options":{ "expose_device_state": true } }' -P <password> -u <user>`

This will enable the device state entity for the targeted device.

robvanoostenrijk, 2025-02-04:
> [@robvanoostenrijk](https://github.com/robvanoostenrijk) I checked with publishDuplicateTransaction: true and false.
> 
> I find no difference in their use
> 
> every 15 seconds it reports and 2 messages appear in the log.
> 

@pcrednet,

The property `publishDuplicateTransaction` affects manual control actions / changes from the device itself not reflecting in the zigbee2mqtt interface. Enabling this resolves that issue as now confirmed by others testing. It does not change the reporting interval.

I haven't looked into changing the reporting interval, perhaps this is possible.

You can however prevent the duplicate messages in zigbee2mqtt for the same state by using the [debounce](https://www.zigbee2mqtt.io/guide/configuration/devices-groups.html#generic-device-options) or [throttle](https://www.zigbee2mqtt.io/guide/configuration/devices-groups.html#generic-device-options) option.

rodrigogbs, 2025-02-04:
> You have an option to enable switch:
>
> [image]
>
> then:
>
> [image]
>
> Great workd [@robvanoostenrijk](https://github.com/robvanoostenrijk)
>
> thank you so much!!

You are right!

I tested here and it is working like a charm!

This converter MUST BE the new pattern of the device!

@robvanoostenrijk I saw that you opened a PR for this, GREAT job and THANKS a lot!

luismipr, 2025-02-07:
Hy All.
Is there any way to show on Home Asistant when thermostat is heating or cooling??

[image]

In history stats appears, but on thermostat card not appear...

[image]

thanks All.

luismipr, 2025-02-10:
I created a new converters based on @robvanoostenrijk that show when thermostat is cooling or heating in home assistant:

```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_mpbki2zm']),
    model: 'TYBAC-006',
    vendor: 'Tuya',
    description: 'Wall-mount thermostat for 2-pipe fan-coil unit',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetLocalTime,
    configure: tuya.configureMagicPacket,
    options: [
        e
            .enum('control_sequence_of_operation', ea.SET, ['cooling_only', 'cooling_and_heating_4-pipes'])
            .withDescription('Operating environment of the thermostat'),
        e
            .binary('expose_device_state', ea.SET, true, false)
            .withDescription('Expose device power state as a separate property when enabled.')
    ],
    exposes: (device, options) => {
        const system_modes = ['off', 'cool', 'heat', 'fan_only'];
        // Device can operate either in 2-pipe or 4-pipe configuration
        // For 2-pipe configurations remove 'heat' mode
        switch (options?.control_sequence_of_operation) {
            case 'cooling_only':
                system_modes.splice(2, 1);
                break;
        }

        const exposes = [
            e
                .climate()
                .withLocalTemperature(ea.STATE)
                .withSetpoint('current_heating_setpoint', 5, 35, 1, ea.STATE_SET)
                .withSystemMode(system_modes, ea.STATE_SET)
                .withFanMode(['low', 'medium', 'high', 'auto'], ea.STATE_SET)
                .withLocalTemperatureCalibration(-5, 5, 0.5, ea.STATE_SET)
                .withRunningState(['idle', 'heat'], ea.STATE),
            e.child_lock(),                
            e.min_temperature().withValueMin(5).withValueMax(15),
            e.max_temperature().withValueMin(15).withValueMax(45),
            e.binary('eco_mode', ea.STATE_SET, 'ON', 'OFF').withDescription('ECO mode ON/OFF'),
            e.max_temperature_limit().withDescription('ECO Heating energy-saving temperature (default: 20 ºC)').withValueMin(15).withValueMax(30),
            e.min_temperature_limit().withDescription('ECO Cooling energy-saving temperature (default: 26 ºC)').withValueMin(15).withValueMax(30),
            e.deadzone_temperature().withValueMin(0).withValueMax(5).withValueStep(1),
            e.binary('manual_mode', ea.STATE_SET, 'ON', 'OFF').withDescription('Manual = ON or Schedule = OFF'),
            ...tuya.exposes.scheduleAllDays(ea.STATE_SET, 'HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C')
        ]

        if (options?.expose_device_state === true) {
            exposes.unshift(e.binary('state', ea.STATE_SET, 'ON', 'OFF').withDescription('Turn the thermostat ON or OFF'));
        }

        return exposes;
    },
    meta: {
        publishDuplicateTransaction: true,
        tuyaDatapoints: [
            [
                1,
                'state',
                {
                    to: async (v, meta) => {
[... 82 lines cut ...]
            [106, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
            [107, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
            [108, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
            [109, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
            [110, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
            [111, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
            [112, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(1)],
            [101, 'manual_mode', tuya.valueConverter.onOff],
        ],
    },
};

module.exports = definition;

```

[image]

[image]

[image]

If anyone can test... 

Thx in advance.

robvanoostenrijk, 2025-02-10:
@luismipr,

Nice addition! You could rewrite it in the following way:

```
            [36, 'running_state', {
                from: (v, meta) => {
                    if (!v) {
                        return meta.state.system_mode;
                    }

                    return 'idle';
                },
            }],  

```

datapoint 36 is a read-only variable, so you'd never use the `to:`, but only `from:`.
Ideally if the value returned is 0 (not true), the valve is open and the controller is expected to execute the current `system_mode`, otherwise its just `idle`.

I cannot test it for you though, as my controller does not have dp36, though my screen has a valve open indicator 😢

luismipr, 2025-02-10:
It was my first contribution, I've never ever touched JS...

@robvanoostenrijk 

```js
[36, 'running_state', {
                from: (v, meta) => {
                    if (!v) {
                        return meta.state.system_mode;
                    }

                    return 'idle';
                },
            }],  
```
Working perfectly!!!

Thanks for sharing!