Koenkk/zigbee2mqtt #21648: Exception while calling fromZigbee converter: Value '4' is not allowed, expected one of [object Object],[object Object],[object Object],[object Object]}

charsiufan, 2024-03-02:
### What happened?

Using a Sonos USB 3.0 P Zigbee Controller and a Tuya based presence detector.  When the presence detector is activated it triggers multiple errors in Zigbee2MQTT. All show: 
Exception while calling fromZigbee converter: Value '4' is not allowed, expected one of [object Object],[object Object],[object Object],[object Object]}

### What did you expect to happen?

No errors generating

### How to reproduce it (minimal and precise)

Anytime the motion sensor has a change of state

### Zigbee2MQTT version

 1.35.3 commit: fe0742a

### Adapter firmware version

20230507

### Adapter

Sonoff USB 3.0 P

### Setup

Docker container, Synology DSM7.1

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14469415/log.txt)

Koenkk, 2024-03-02:
This is a `radar_scene` mode, 0 = default, 1 = bathroom, 2 = bedroom, 3 = sleeping, do you have any clue what 4 should be mapped to? If not, we can map it to `unknown`.

Koenkk, 2024-03-04:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-03-04 by Koenkk