Koenkk/zigbee2mqtt #21651: Error in log for GZCGQ01LM after updating to 1.36.0

Calimerorulez, 2024-03-02:
### What happened?

Updated to 1.36.0 and now this error shows:

`2024-03-02 16:52:05Failed to configure 'Lichtsensor voorkant', attempt 1 (Error: Bind 0x04cf8cdf3c78bf0f/1 msIlluminanceMeasurement from '0x00124b001ca5ea40/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Object.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:692:28 at Queue.execute (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Endpoint.bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:551:13) at Object.bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:43:9) at configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/lumi.ts:2229:13) at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/index.ts:141:25) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:122:13) at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:174:17))`

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

20230507

### Adapter

CC1352P

### Setup

Docker LXC

mrskycriper, 2024-03-02:
@Calimerorulez have you tried requesting an interview and waking it up using a button? Z2M probably wants to re-interview it because of converter changes

Mettchen, 2024-03-09:
Getting the same issue on a variety of different sensors
- Motion (Aqara)
- Sonoff ZBMini
- Lux (Aqara)
- Motion (IKEA)

I just tried to repair the devices fully since I wanted to move to a different band anyway, and that did not change the situation.  Adapter CC2652RB (Slae.sh)

Running in a VM:
HA Core: 2024.3.0
Supervisor: 2024.02.1
HA Operating System: 12
Z2M: 1.36.0-1

RubenKelevra, 2024-09-02:
Duplicate to https://github.com/Koenkk/zigbee2mqtt/issues/21842

closed 2024-09-02 by Koenkk