Koenkk/zigbee2mqtt #21656: [New device support]: Tuya ZP01 Motion Sensor TS0202,  _TZ3000_y56pgpgs

jpmorrison, 2024-03-03:
### Link

https://www.aliexpress.com/item/1005006010327384.html

### Database entry

{"id":5,"type":"EndDevice","ieeeAddr":"0xa4c138a50d6d824c","nwkAddr":17821,"manufId":4417,"manufName":"_TZ3000_y56pgpgs","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,1280,3,0],"outClusterList":[4096,6,25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"ssIasZone":{"attributes":{"61441":1,"iasCieAddr":"0x00124b0024c0d743","zoneState":1,"currentZoneSensitivityLevel":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1709422278504,"defaultSendRequestWhen":"immediate"}

### Comments

I changed the manufacturer to _TZ3000_y56pgpgs and it works for occupancy - model ZP01 from aliexpress.
Didn't this device, sorry if it's a dupe.

https://github.com/Koenkk/zigbee2mqtt/issues/19105

I see LQI in Zigbee2MQTT but not in homeassistant. Not sure about battery, sometimes it's green, sometimes it's blinking or question mark. Dont see battery, voltage in HA either

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
            fingerprint: [
                            {modelID: 'TS0202', manufacturerName: '_TZ3000_y56pgpgs'},
                            ],
            model: 'ZP01',
            vendor: 'TuYa',
            description: 'Motion sensor',
            fromZigbee: [fz.ias_occupancy_alarm_1, fz.battery, fz.ignore_basic_report, fz.ias_occupancy_alarm_1_report],
            toZigbee: [],
            exposes: [e.occupancy(), e.battery_low(), e.battery(), e.battery_voltage()],
            configure: async (device, coordinatorEndpoint, logger) => {
                            const endpoint = device.getEndpoint(1);
                            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
                            try {
                                                await reporting.batteryPercentageRemaining(endpoint);
                                            } catch (error) {/* Fails for some https://github.com/Koenkk/zigbee2mqtt/issues/13708*/}
                        },
};

module.exports = definition;
```

jpmorrison, 2024-03-05:
No, I didn't see the device  in Homeassistant, and when I looked at the Zigbee2Mqtt logs I saw this:

```
warn  2024-03-01 23:01:23: Received message from unsupported device with Zigbee model 'TS
0202' and manufacturer name '_TZ3000_y56pgpgs'
warn  2024-03-01 23:01:23: Please see: https://www.zigbee2mqtt.io/advanced/support-new-de
vices/01_support_new_devices.html
```

I’m running 1.30.4

Koenkk, 2024-03-05:
Thats an old version, updating to the latest should fix it

closed 2024-03-05 by Koenkk