Koenkk/zigbee2mqtt #21659: [Wrong device]: MiBoxer FUT037Z+ is shown as Tuya TS0505B_1

grezco, 2024-03-03:
### Link

https://miboxer.com/fr/produit/controleur-led-3-en-1-zigbee-3-0-2-4g

### Model

TS0505B_1

### Description

Contrôleur LED 3 en 1 (Zigbee 3.0 + 2.4G)

### Vendor

MiBoxer

### Picture (link)

https://miboxer.com/wp-content/uploads/a1-50.jpg

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0xa4c138f4a59f5e0f","nwkAddr":61911,"manufId":4417,"manufName":"_TZB210_417ikxay","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":29}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1842547789},"lastSeen":1709455071435}

AnderVentura, 2024-03-22:
I'm getting this error using Zigbee2MQTT 1.36.0-1 in Home Assistant 2024.3.1. Any way I can fix it?
[image]
[image]

AnderVentura, 2024-05-10:
Any way we can force detection as the correct device?

thefox69, 2024-05-13:
I guess Z2M is supprting FUT037Z without the plus.  Means only RGB.
The newer (different) model with +, can switch between three different modes /RGB, RGBW and RGB+CCT.
Maybe thats the reason why not identified correctly

ppanagiotis, 2024-05-16:
Any update about this? Or workaround ?

Koenkk, 2024-05-18:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-05-18 by Koenkk

ppanagiotis, 2024-06-15:
Can we also fix this please https://github.com/Koenkk/zigbee2mqtt/issues/21185 ? Looks similar problem

BastiKA84, 2025-01-29:
Hi all,
seeing this Topic, I assume, that the Miboxer FUT037Z+ is now full compatible with zigbee2mqtt. But when I look at this Topic [https://github.com/Koenkk/zigbee2mqtt/issues/21761](url) it seems to be still valid as the last entry there is from the 27th december. 
I am in the decision making of buying a LED panel, that is nearly as good as a Philipps HUE but for around half the price. That said LED panel uses the Miboxer FUT037Z+ as the Zigbee Controller.
So I would like to know, if all is working fine, then I could purchase the cheaper one. If not, I would need to go with the HUE one.

NickBuilder, 2025-11-26:
I'm also having this issue.
Updated my z2m installation to hopefully remedy this but the issue persists.

Zigbee2MQTT-version [2.6.3]
Frontend-version [2.2.3]
zigbee-herdsman-converters version [25.59.0]
zigbee-herdsman version [6.3.2]