Koenkk/zigbee2mqtt #21661: From version 1.36.0 WXKG01LM single press is passed in MQTT as Long press

engal89, 2024-03-03:
### What happened?

If i short press the button, Z2M pass in the MQTT topic click with "long" value. Action is correct passed as single.

[image]

No problem with long and double click.

### What did you expect to happen?

I expect that click is passed as single, as worked before that version.

### How to reproduce it (minimal and precise)

Short press the button, Z2M pass in the MQTT topic click with "long" value.

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

20211217

### Adapter

SONOFF Zigbee USB dongle -  zStack3x0

### Setup

Home Assistant OS container

### Debug log

 2024-03-03 11:43:15MQTT publish: topic 'zigbee2mqtt/Pulsante ingresso', payload '{"action":"single","battery":100,"click":null,"linkquality":185,"power_outage_count":317,"voltage":3012}'
Info 2024-03-03 11:43:15MQTT publish: topic 'zigbee2mqtt/Pulsante ingresso', payload '{"action":"","battery":100,"click":null,"linkquality":185,"power_outage_count":317,"voltage":3012}'
Info 2024-03-03 11:43:15MQTT publish: topic 'zigbee2mqtt/Pulsante ingresso/action', payload 'single'
Info 2024-03-03 11:43:16MQTT publish: topic 'zigbee2mqtt/Pulsante ingresso', payload '{"action":null,"battery":100,"click":"long","linkquality":185,"power_outage_count":317,"voltage":3012}'
Info 2024-03-03 11:43:16MQTT publish: topic 'zigbee2mqtt/Pulsante ingresso', payload '{"action":null,"battery":100,"click":"","linkquality":185,"power_outage_count":317,"voltage":3012}'
Info 2024-03-03 11:43:16MQTT publish: topic 'zigbee2mqtt/Pulsante ingresso/click', payload 'long'

dangerdanik, 2024-03-03:
Same problem. A short click on the wireless switch (WXKG01LM) is processed as a long click on the wireless switch.

GregoryZet, 2024-03-03:
I confirm - the same problem:

After update zigbee2mqtt Addon to v1.36.0, the button Mi wireless switch (WXKG01LM) started to interpret a single click as a long press. Automations triggered by a single click no longer work.
The situation occurs on all Mi wireless switch (WXKG01LM)
Addon version

v1.36.0

Platform - HomeAssistant
Core 2024.2.4
Supervisor 2024.02.1
Operating System 12.0
Logs of the issue (if applicable)

In Z2M log "single click" looks like this (MQTT publish two different action topics) :

info 2024-03-02 19:01:29MQTT publish: topic 'zigbee2mqtt/switch_1', payload '{"action":"single","battery":100,"click":null,"linkquality":54,"power_outage_count":95,"voltage":3022}'
info 2024-03-02 19:01:29MQTT publish: topic 'zigbee2mqtt/switch_1', payload '{"action":"","battery":100,"click":null,"linkquality":54,"power_outage_count":95,"voltage":3022}'
info 2024-03-02 19:01:30MQTT publish: topic 'zigbee2mqtt/switch_1/action', payload 'single'
info 2024-03-02 19:01:30MQTT publish: topic 'zigbee2mqtt/switch_1', payload '{"action":null,"battery":100,"click":"long","linkquality":54,"power_outage_count":95,"voltage":3022}'
info 2024-03-02 19:01:30MQTT publish: topic 'zigbee2mqtt/switch_1', payload '{"action":null,"battery":100,"click":"","linkquality":54,"power_outage_count":95,"voltage":3022}'
info 2024-03-02 19:01:30MQTT publish: topic 'zigbee2mqtt/switch_1/click', payload 'long'

nethomesh, 2024-03-03:
I confirm with me is the same error.

Fierys, 2024-03-03:
I confirm - the same problem

0WL007, 2024-03-03:
Same problem.

Koenkk, 2024-03-04:
Could someone provide the herdsman debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

engal89, 2024-03-04:
[Z2M.log](https://github.com/Koenkk/zigbee2mqtt/files/14482143/Z2M.log)

Hope this is what you need! :)

Thank you

Koenkk, 2024-03-04:
Fixed!

Note however that the click is deprecated, it's recommended to set [`legacy: false`](https://www.zigbee2mqtt.io/devices/WXKG01LM.html#options) for this device.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-03-04 by Koenkk

james-fry, 2024-03-15:
Thanks for the fix. Will need to revert my workaround after this drops.