Koenkk/zigbee2mqtt #21667: Newer whitelabel Sunricher SR-ZG9092A devices with different clusters

treetip, 2024-03-03:
### What happened?

I bought one of these ZG9092A thermostats from a local seller ~a year ago to test that it would work for me.

I recently bought 9 more from a different seller, after the seller reported that in zigbee2mqtt the device reports the same firmware version.

But in previous Z2M version (used as an addon in HA) the device set up always failed, I'd get an error but the device worked fine for my purposes.

Today I updated to the latest addon, 1.36.0 to do some testing to report to the seller, and i no longer get the errors, the (newer) device binds nothing and sets up no reporting at all. I could not find the code change for this, i guess it fails and then reverts?

The cause seems to be that the older device reports:

    Core: V01.30
    Zigbee: V20.00

during a factory reset, where the newer one reports:

    Core: V01.32
    Zigbee: V23.00

So even though they report the same firmware version, they aren't on the same software/hardware stack or whatever.

The newer version seems to be missing an output cluster which caused the setup errors in previous version of Z2M.

[image]

### What did you expect to happen?

Both devices would get the same binds and reporting settings.

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

20230507

### Adapter

uzg-01

### Setup

home assistant addon, x86_64 intel nuc

treetip, 2024-03-03:
Both versions of the device report 

    Firmware build date:   NULL
    Firmware version:      6.9.1.0_r4

treetip, 2024-03-03:
Spent some time looking around all the code and type definitions and it would be trivial to just check for the endpoints existence during the configure call in the herdsman-converter.

Different question if you want to do it like that, or if a mechanism already exist that could do this? The binds for example seem to be hard coded and do not check if they are actually available, this could allow for a version branch in the code, or just plain filtering out the ones that are not present in this revision.

treetip, 2024-03-06:
fixed it for myself with a custom converter that basically just try/catches this part:

            await endpoint.configureReporting('hvacThermostat', [{
                attribute: 'occupancy',
                minimumReportInterval: 0,
                maximumReportInterval: constants.repInterval.HOUR,
                reportableChange: null,
            }]);

as it seems the newer version no longer supports this attribute, seems to work fine, all devices configured nicely

ollimyllari, 2024-07-30:
@treetip could you provide the whole custom converter file? Thanks!

treetip, 2024-07-31:
```

const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const constants = require('zigbee-herdsman-converters/lib/constants');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const {} = require('zigbee-herdsman-converters/lib/tuya');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;
const ota = require("zigbee-herdsman-converters/lib/ota");

async function syncTime(endpoint) {
    try {
        const time = Math.round(((new Date()).getTime() - constants.OneJanuary2000) / 1000 + ((new Date()).getTimezoneOffset() * -1) * 60);
        const values = { time: time };
        await endpoint.write('genTime', values);
    }
    catch (error) { /* Do nothing*/ }
}
const definition = 
    {
        zigbeeModel: ['ZG9092', 'HK-LN-HEATER-A'],
        model: 'SR-ZG9092A',
        vendor: 'Sunricher',
        description: 'Touch thermostat',
        ota: ota.zigbeeOTA,
        fromZigbee: [fz.thermostat, fz.namron_thermostat, fz.metering, fz.electrical_measurement, fz.namron_hvac_user_interface],
        toZigbee: [tz.thermostat_occupied_heating_setpoint, tz.thermostat_unoccupied_heating_setpoint, tz.thermostat_occupancy,
            tz.thermostat_local_temperature_calibration, tz.thermostat_local_temperature, tz.thermostat_outdoor_temperature,
            tz.thermostat_system_mode, tz.thermostat_control_sequence_of_operation, tz.thermostat_running_state,
            tz.namron_thermostat, tz.namron_thermostat_child_lock],
        exposes: [
            e.numeric('outdoor_temperature', ea.STATE_GET).withUnit('°C')
                .withDescription('Current temperature measured from the floor sensor'),
            e.climate()
                .withSetpoint('occupied_heating_setpoint', 0, 40, 0.1)
                .withSetpoint('unoccupied_heating_setpoint', 0, 40, 0.1)
                .withLocalTemperature()
                .withLocalTemperatureCalibration(-3, 3, 0.1)
                .withSystemMode(['off', 'auto', 'heat'])
                .withRunningState(['idle', 'heat']),
            e.binary('away_mode', ea.ALL, 'ON', 'OFF')
                .withDescription('Enable/disable away mode'),
            e.binary('child_lock', ea.ALL, 'UNLOCK', 'LOCK')
                .withDescription('Enables/disables physical input on the device'),
            e.power(), e.current(), e.voltage(), e.energy(),
            e.enum('lcd_brightness', ea.ALL, ['low', 'mid', 'high'])
                .withDescription('OLED brightness when operating the buttons.  Default: Medium.'),
            e.enum('button_vibration_level', ea.ALL, ['off', 'low', 'high'])
                .withDescription('Key beep volume and vibration level.  Default: Low.'),
            e.enum('floor_sensor_type', ea.ALL, ['10k', '15k', '50k', '100k', '12k'])
                .withDescription('Type of the external floor sensor.  Default: NTC 10K/25.'),
            e.enum('sensor', ea.ALL, ['air', 'floor', 'both'])
                .withDescription('The sensor used for heat control.  Default: Room Sensor.'),
            e.enum('powerup_status', ea.ALL, ['default', 'last_status'])
                .withDescription('The mode after a power reset.  Default: Previous Mode.'),
            e.numeric('floor_sensor_calibration', ea.ALL)
                .withUnit('°C')
                .withValueMin(-3).withValueMax(3).withValueStep(0.1)
                .withDescription('The tempearatue calibration for the external floor sensor, between -3 and 3 in 0.1°C.  Default: 0.'),
            e.numeric('dry_time', ea.ALL)
                .withUnit('min')
                .withValueMin(5).withValueMax(100)
                .withDescription('The duration of Dry Mode, between 5 and 100 minutes.  Default: 5.'),
            e.enum('mode_after_dry', ea.ALL, ['off', 'manual', 'auto', 'away'])
                .withDescription('The mode after Dry Mode.  Default: Auto.'),
[... 152 lines cut ...]
                    minimumReportInterval: 0,
                    maximumReportInterval: constants.repInterval.HOUR,
                    reportableChange: 0
                }], options);
            // Away Mode Set
            await endpoint.configureReporting('hvacThermostat', [{
                    attribute: { ID: 0x2002, type: 0x30 },
                    minimumReportInterval: 0,
                    maximumReportInterval: constants.repInterval.HOUR,
                    reportableChange: null
                }], options);
            // Device does not asks for the time with binding, we need to write time during configure
            syncTime(endpoint);
            // Trigger initial read
            await endpoint.read('hvacThermostat', ['systemMode', 'runningState', 'occupiedHeatingSetpoint']);
            await endpoint.read('hvacThermostat', [0x1000, 0x1001, 0x1002, 0x1003], options);
            await endpoint.read('hvacThermostat', [0x1004, 0x1005, 0x1006, 0x1007], options);
            await endpoint.read('hvacThermostat', [0x1008, 0x1009, 0x100A, 0x100B], options);
            await endpoint.read('hvacThermostat', [0x2001, 0x2002], options);
        },
    };

module.exports = definition;

```

treetip, 2024-12-30:
If anyone has any suggestions on how to implement this for the converter properly, I could do a PR.

Koenkk, 2024-12-30:
You can do an if/else based on the stackVersion or zclVersion in the `configure`, dummy example:

```js
        configure: async (device, coordinatorEndpoint) => {
            if (device.stackVersion == 9) {
                const endpoint = device.getEndpoint(1);
                await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'closuresWindowCovering']);
                await reporting.currentPositionLiftPercentage(endpoint);
                await reporting.currentPositionTiltPercentage(endpoint);
            }
        },
```

closed 2025-03-10 by github-actions[bot]