Koenkk/zigbee2mqtt #21668: [New device support]: Aqara Z1 PRO ZNQBKG45LM

Pickllo, 2024-03-04:
### Link

https://www.aqara.com/en/product/smart-wall-switch-z1-pro-sea/

### Database entry

ieeeAddr:0x54ef441000b2640e

### Comments

This is similar to the Z1, but with the addition of the touch slider on the left. Anyone able to help get it into HA?

### External definition

```shell
const {deviceEndpoints, electricityMeter, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['lumi.switch.acn059'],
    model: 'lumi.switch.acn059',
    vendor: 'Aqara',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2,"3":3,"4":4,"21":21}}), electricityMeter(), onOff({"powerOnBehavior":false,"endpoints":{"1":1,"2":2,"3":3}})],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

mrmaximas, 2024-03-04:
please provide database entry for ieeeAddr: 0x54ef441000b2640e from zigbee2mqtt_DIR/data/database.db file located depend of your os.

Pickllo, 2024-03-04:
> please provide database entry for ieeeAddr: 0x54ef441000b2640e from zigbee2mqtt_DIR/data/database.db file located depend of your os.

Apologies, updated with the correct entry

mrskycriper, 2024-03-05:
@Pickllo @MadMirza we are trying to figure out the slider in this [issue](https://github.com/Koenkk/zigbee2mqtt/issues/21687). Your help will be very much welcome.

closed 2024-03-08 by Koenkk