Koenkk/zigbee2mqtt #21669: Develco HESZB-120 keeps interviewing after OTA Update

Migaddo, 2024-03-04:
### What happened?

After updating Develco HESZB-120
From: FW 3.4.25
To: FW 4.0.8
Trough OTA, the device keeps in interviewing mode. It does pair successful, however it keeps leaving the network, creating a pairing loop.

### What did you expect to happen?

Upgrade to new firmware, pair and stay stable.

### How to reproduce it (minimal and precise)

Upgrade to new firmware version. 
Als tried rebooting the host, the add On. nothing helped

### Zigbee2MQTT version

 1.36.0 

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Zigbee USB Dongle Plus ( EZSP v8)

### Setup

Add On x86-64 HP T620 Thin Client

### Debug log

Info 2024-03-04 09:25:23MQTT publish: topic 'zigbee2mqtt/Brandmelder 1st Floor', payload '{"alarm":null,"battery":80,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0xe0798dfffe77bf89","last_seen":"2024-03-04T09:25:23+01:00","linkquality":196,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":19.37,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"},"update_available":null,"voltage":2900,"zone_id":23}'
Info 2024-03-04 09:25:23MQTT publish: topic 'zigbee2mqtt/Brandmelder 1st Floor', payload '{"alarm":null,"battery":80,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0xe0798dfffe77bf89","last_seen":"2024-03-04T09:25:23+01:00","linkquality":196,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":19.37,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"},"update_available":null,"voltage":2900,"zone_id":23}'
Info 2024-03-04 09:25:29MQTT publish: topic 'zigbee2mqtt/Brandmelder 1st Floor', payload '{"alarm":null,"battery":80,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0xe0798dfffe77bf89","last_seen":"2024-03-04T09:25:29+01:00","linkquality":196,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":19.37,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"},"update_available":null,"voltage":2900,"zone_id":23}'
Info 2024-03-04 09:25:29MQTT publish: topic 'zigbee2mqtt/Brandmelder 1st Floor', payload '{"alarm":null,"battery":80,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0xe0798dfffe77bf89","last_seen":"2024-03-04T09:25:29+01:00","linkquality":196,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":19.37,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"},"update_available":null,"voltage":2900,"zone_id":23}'
Info 2024-03-04 09:25:29MQTT publish: topic 'zigbee2mqtt/Brandmelder 1st Floor', payload '{"alarm":null,"battery":80,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0xe0798dfffe77bf89","last_seen":"2024-03-04T09:25:29+01:00","linkquality":200,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":19.37,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"},"update_available":null,"voltage":2900,"zone_id":23}'
Info 2024-03-04 09:25:29MQTT publish: topic 'zigbee2mqtt/Brandmelder 1st Floor', payload '{"alarm":null,"battery":80,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0xe0798dfffe77bf89","last_seen":"2024-03-04T09:25:29+01:00","linkquality":204,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":19.37,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"},"update_available":null,"voltage":2900,"zone_id":23}'
Info 2024-03-04 09:25:29MQTT publish: topic 'zigbee2mqtt/Brandmelder 1st Floor', payload '{"alarm":null,"battery":80,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0xe0798dfffe77bf89","last_seen":"2024-03-04T09:25:29+01:00","linkquality":204,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":19.37,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"},"update_available":null,"voltage":2900,"zone_id":23}'
Info 2024-03-04 09:26:00MQTT publish: topic 'zigbee2mqtt/Brandmelder 1st Floor', payload '{"alarm":null,"battery":80,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0xe0798dfffe77bf89","last_seen":"2024-03-04T09:26:00+01:00","linkquality":204,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":19.37,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"},"update_available":null,"voltage":2900,"zone_id":23}'
Info 2024-03-04 09:26:00MQTT publish: topic 'zigbee2mqtt/Brandmelder 1st Floor', payload '{"alarm":null,"battery":80,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0xe0798dfffe77bf89","last_seen":"2024-03-04T09:26:00+01:00","linkquality":204,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":19.37,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"},"update_available":null,"voltage":2900,"zone_id":23}'
Info 2024-03-04 09:26:05MQTT publish: topic 'zigbee2mqtt/Brandmelder 1st Floor', payload '{"alarm":null,"battery":80,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0xe0798dfffe77bf89","last_seen":"2024-03-04T09:26:05+01:00","linkquality":204,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":19.37,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"},"update_available":null,"voltage":2900,"zone_id":23}'
Info 2024-03-04 09:26:05MQTT publish: topic 'zigbee2mqtt/Brandmelder 1st Floor', payload '{"alarm":null,"battery":80,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0xe0798dfffe77bf89","last_seen":"2024-03-04T09:26:05+01:00","linkquality":204,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":19.37,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"},"update_available":null,"voltage":2900,"zone_id":23}'
Info 2024-03-04 09:26:05MQTT publish: topic 'zigbee2mqtt/Brandmelder 1st Floor', payload '{"alarm":null,"battery":80,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0xe0798dfffe77bf89","last_seen":"2024-03-04T09:26:05+01:00","linkquality":204,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":19.37,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"},"update_available":null,"voltage":2900,"zone_id":23}'
Info 2024-03-04 09:26:05MQTT publish: topic 'zigbee2mqtt/Brandmelder 1st Floor', payload '{"alarm":null,"battery":80,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0xe0798dfffe77bf89","last_seen":"2024-03-04T09:26:05+01:00","linkquality":192,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":19.37,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"},"update_available":null,"voltage":2900,"zone_id":23}'
Info 2024-03-04 09:26:05MQTT publish: topic 'zigbee2mqtt/Brandmelder 1st Floor', payload '{"alarm":null,"battery":80,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0xe0798dfffe77bf89","last_seen":"2024-03-04T09:26:05+01:00","linkquality":192,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":19.37,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"},"update_available":null,"voltage":2900,"zone_id":23}'

sjorge, 2024-03-04:
Might be https://github.com/Koenkk/zigbee-herdsman/commit/b004cf827da04eb78e7658776bd2726289797b16 since you are using a EZSP based coordinator.

srett, 2024-03-04:
There seems to be issues like this popping up left and right, like [the random one I posted in yesterday.](https://github.com/Koenkk/zigbee2mqtt/issues/21036) I have a zStack3x0-based coordinator though.

Migaddo, 2024-03-04:
> Might be [Koenkk/zigbee-herdsman@b004cf8](https://github.com/Koenkk/zigbee-herdsman/commit/b004cf827da04eb78e7658776bd2726289797b16) since you are using a EZSP based coordinator.

All my 90+ devices work perfectly, only the 1 device HESZB-120 that i upgraded trough OTA has this problem. the other 3 HESZB-120 devices that I didnt upgrade stayed stable. OTA update for these devices were recently implemented in Z2MQTT

srett, 2024-03-06:
I downgraded to 1.33.2, the plug paired on the second attempt (when I only allowed pairing via the nearest router). After that I switched back to current dev branch and the plug is still paired. Fingers crossed.

Xento, 2024-05-05:
I'm having the same issue.
After OTA to 4.0.8 
I already updated Zigbee2Mqtt and the SonOff Stick to the latest firmware and switched to the new ember adapter.
Tried to learn it directly on the coordinator but this doesn't work, too.

Zigbee2MQTT-Version: [1.37.0] commit: [46f34c8]
Coordinator-Typ: EmberZNet
Coordinator-Version: 7.4.2 [GA]

Zigbee Herdsman Konverter Version: 19.32.0
Zigbee Herdsman Version: 0.45.0

Xento, 2024-05-17:
Any news on this?

I'm getting this error:
`Info 2024-05-17 12:00:03Starting interview of '0x0015bc00340018de'
Error 2024-05-17 12:00:27Interview failed for '0x0015bc00340018de with error 'Error: Interview failed because of failed IAS enroll (zoneState didn't change ('0x0015bc00340018de')'
Error 2024-05-17 12:00:27Failed to interview '0x0015bc00340018de', device has not successfully been paired`

Migaddo, 2024-05-19:
From my point of view I no longer have this issue.
I have paired it again, but on the nearest router and not directly on the coordinator. I hope this will also work for you.

tichachm, 2024-05-28:
I have the same issue, updated both of my HESZB-120 to 4.0.8, after that they seem to have a much higher battery consumption.
So I resetted one and now it no longer joins the network:
`Info 2024-05-28 07:35:40Interview for '0x0015bc0034000bf2' started
Info 2024-05-28 07:35:40Device '0x0015bc0034000bf2' joined
Info 2024-05-28 07:35:40Starting interview of '0x0015bc0034000bf2'
Error 2024-05-28 07:35:51Interview failed for '0x0015bc0034000bf2 with error 'Error: AREQ - ZDO - simpleDescRsp after 10000ms'
Error 2024-05-28 07:35:51Failed to interview '0x0015bc0034000bf2', device has not successfully been paired`
[1.37.1]commit: [ea39d86] with zStack3x0 (20230507)

Also paring from a router does not work

tichachm, 2024-05-30:
Yes I tried some more times now it worked.

mamrai1, 2024-06-13:
There is definitely a problem with this devices and ember driver...(fails to configure device and high power consumption). Iswitched back to ezsp and repaired the device .

JanDetrez, 2024-09-03:
Also resolved here after a factory reset of the device and repairing using the latest  1.40.0-1

closed 2025-01-07 by github-actions[bot]