Koenkk/zigbee2mqtt #21670: Zigbee unstable since 1.34

1nfinity84, 2024-03-04:
### What happened?

Until 1.33 (which I rolled back to) my zigbee network worked fine, with an ocassional need to restart (like once in a few months). After updating to 1.34 it'd go completely mental, devices falling off the network, becomming unavailable, generally the whole network would go wonky. 

The weirdest thing is that the logs only contain errors which (I think) are related to the quirks of aqara devices. 

Any ideas? Pretty please.

### What did you expect to happen?

I'd just love it to work as before. Don't want to stay on a legacy version.

### How to reproduce it (minimal and precise)

To be honest, no idea.

### Zigbee2MQTT version

1.36 currently

### Adapter firmware version

6.10.3.0 build 297 (EZSP v8)

### Adapter

Sonoff Dongle-E

### Setup

Addon on Home assistant

### Debug log

 at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
Zigbee2MQTT:error 2024-03-04 11:51:51: Failed to configure 'mamka tvoc', attempt 1 (Error: {"address":64739,"clusterId":32801,"sequence":133} after 10000ms
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
    at Driver.zdoRequest (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:659:43)
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:574:13
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:551:13)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:43:9)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/lumi.ts:2537:13)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:141:25)
    at Configure.configure (/app/lib/extension/configure.ts:122:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
Zigbee2MQTT:error 2024-03-04 11:54:03: Failed to configure 'vojta tvoc', attempt 1 (Error: {"address":5612,"clusterId":32801,"sequence":154} after 10000ms
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
    at Driver.zdoRequest (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:659:43)
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:574:13
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:551:13)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:43:9)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/lumi.ts:2537:13)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:141:25)
    at Configure.configure (/app/lib/extension/configure.ts:122:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
Zigbee2MQTT:error 2024-03-04 11:58:07: Failed to configure 'obyvak-ht-tvoc', attempt 2 (Error: {"address":26194,"clusterId":32801,"sequence":185} after 10000ms
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
    at Driver.zdoRequest (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:659:43)
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:574:13
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:551:13)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:43:9)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/lumi.ts:2537:13)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:141:25)
    at Configure.configure (/app/lib/extension/configure.ts:122:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
Zigbee2MQTT:error 2024-03-04 12:00:30: Failed to configure 'mamka tvoc', attempt 2 (Error: {"address":64739,"clusterId":32801,"sequence":205} after 10000ms
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
[... 39 lines cut ...]
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:141:25)
    at Configure.configure (/app/lib/extension/configure.ts:122:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
Zigbee2MQTT:error 2024-03-04 12:06:40: Failed to configure 'vojta tvoc', attempt 3 (Error: {"address":5612,"clusterId":32801,"sequence":33} after 10000ms
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
    at Driver.zdoRequest (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:659:43)
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:574:13
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:551:13)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:43:9)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/lumi.ts:2537:13)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:141:25)
    at Configure.configure (/app/lib/extension/configure.ts:122:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
Zigbee2MQTT:error 2024-03-04 12:14:46: Failed to configure 'Ht citron', attempt 1 (Error: {"address":56612,"clusterId":32801,"sequence":139} after 10000ms
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
    at Driver.zdoRequest (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:659:43)
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:574:13
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:551:13)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:43:9)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/lumi.ts:1256:13)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:141:25)
    at Configure.configure (/app/lib/extension/configure.ts:122:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

mrskycriper, 2024-03-04:
It looks like Z2M wants to reconfigure those devices, but they are sleepy and don't answer in time. Have you tried helping by waking them up using pairing button so they can finish the process? I think that is the problem here

1nfinity84, 2024-03-04:
Yes, but I never know when it will try to do this. It spams the log like this absolutely randomly, I don't know if it could be the reason that the add-on sometimes get stuck?

mrskycriper, 2024-03-04:
It should complete the interview successfully once and then stop spaming

abergui, 2024-03-04:
> > @mrskycriper Do you think it'd be fixed by re-pairing them?
> 
> This should definitely fix the issue

With 1.36 I have the same issue and unfortunately when I tried re-pairing, z2m lost the devices and cannot able to re-pair them.
And then z2m have lost the triggers of all my Button scene switch thus not any automation was working.

I had to roll-back to 1.35.1 and had must to re-pair temperature devices I had lost by trying to re-pair with 1.36 and that now OK

All triggers has reappears after push all functions of each Scene switch but some has disappeared in Automations.

jymorel, 2024-03-04:
same issue here since updating to 1.36.0.1. Zigbee network unstable. Some Sonoff switches and plugs frequently become unavailable (sonoff zigbee dongle plus-p). I have tried to delete, recreate et re-pair some devices. Same issue. Looks like a reduction of the dongle power

1nfinity84, 2024-03-05:
Any of you tried flashing stick to a newer firmware? I feel tempted to use the web flasher and use ember driver

abergui, 2024-03-05:
> Any of you tried flashing stick to a newer firmware? I feel tempted to use the web flasher and use ember driver

I was tempted too but I preferred to roll-back z2m because it was working well before and I was not have enough time to test it really

russilker, 2024-03-06:
> Any of you tried flashing stick to a newer firmware? I feel tempted to use the web flasher and use ember driver

Doesn't help unfortunately, I upgraded firmware on my zb-dongle E and am using Ember, and it's still as unstable as it's been the last few releases

zcaalock, 2024-03-06:
same here, i try every update and need to go back to 1.35.1-1. The only new thing with new update is that after few messages of "successful configured" on my [TS0505B_1] switches, zigbee2mqtt completely crashes

lazynoda, 2024-03-08:
After almost half a day, everything is working as expected. Some battery devices took around 30min to send updates, but you can quickly press the button with a pin to "force" it (not resetting the device).

1nfinity84, 2024-05-04:
Solved by update to 1.37 and updating do gle firmware to 7.4.2.0

closed 2024-05-04 by 1nfinity84