Koenkk/zigbee2mqtt #21675: Devlce : Develco KEYZB-110

Dabbadoeber, 2024-03-04:
### What happened?

Try to pair with Zigbee2mqtt.
Pressing the small opening on the back, first until 0ne flash, not pairing
Pressing the small opening on the back, first until two flashes, not pairing
Pressing the small opening on the back, first until multiple flashes, reset, after this no pairing

### What did you expect to happen?

i would pair

Tried an diffrent Zigbee networks, still not working.
Tried on Channel : 15, 20 and 25
With coordinator ezsp and zstack both not working. 
Asked for a replacement, device. Tried to pair this one with zstack on ZHA  still no luck.
Any one else got issues with these devices?
Any help would be appreciated

### How to reproduce it (minimal and precise)

Pressing the small opening on the back, first until 0ne flash, not pairing
Pressing the small opening on the back, first until two flashes, not pairing
Pressing the small opening on the back, first until multiple flashes, reset, after this no pairing

### Zigbee2MQTT version

 1.36.0 

### Adapter firmware version

20230507

### Adapter

SLZB-06

### Setup

Intel NUC (VMWare)

### Debug log

x

RobLeighton67, 2024-03-05:
My recommendation would be remove back cover, then batteries for a few seconds, replace the batteries and then press the button inside the circular hole (the square hole is the LED, the metal bar is tamper detection) between the batteries for 15s so that the LED has flashed once, then twice , and finally three times - the device is then reset and looking to join any available network.

JSteunou, 2024-07-19:
I'm using a SLZB-06M which is slightly different but it works.