Koenkk/zigbee2mqtt #21676: Aqara Dual Relay T2 (LLKZMK12LM ) Configuration Failure

morhimi, 2024-03-04:
### What happened?

I get configuration failure message after adding the device to the network today.

### What did you expect to happen?

configuration should succeed

### How to reproduce it (minimal and precise)

add the device to zigbee network

### Zigbee2MQTT version

1.36.0-dev commit: aaed8ab

### Adapter firmware version

7.4.1 [GA]

### Adapter

SLZB-07 

### Setup

Add-on on HAOS running on Mini PC (Intel)

### Debug log

```
Debug 2024-03-04 22:44:26Received MQTT message on 'z2medge/bridge/request/device/configure' with data '{"id":"Aqara T2","transaction":"yhw5r-1"}'
Info 2024-03-04 22:44:26Configuring 'Aqara T2'
Debug 2024-03-04 22:44:26Configure reporting: false, read: true for 0x54ef4410009b3f3b/1 manuSpecificLumi [{"attribute":{"ID":10,"type":32},"min":-1,"max":-1,"change":-1}]
Debug 2024-03-04 22:44:26Received Zigbee message from 'Aqara T2', type 'readResponse', cluster 'manuSpecificLumi', data '{"10":3}' from endpoint 1 with groupID 0
Debug 2024-03-04 22:44:26LLKZMK12LM: Processed data into payload {}
Debug 2024-03-04 22:44:26Configure reporting: false, read: true for 0x54ef4410009b3f3b/2 manuSpecificLumi [{"attribute":{"ID":10,"type":32},"min":-1,"max":-1,"change":-1}]
Info 2024-03-04 22:44:26MQTT publish: topic 'z2medge/Aqara T2', payload '{"action":null,"consumption":0.09999999403953552,"current":0,"device_temperature":34,"energy":0.1,"interlock":"OFF","led_disabled_night":false,"linkquality":200,"mode":"power","operation_mode_l1":"decoupled","operation_mode_l2":"decoupled","power":0,"power_on_behavior":"off","power_on_behavior_l1":"off","power_outage_count":6,"pulse_length":200,"state":"OFF","state_l1":"OFF","state_l2":"OFF","switch_type":"none","switch_type_l1":"none","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":213.2}'
Debug 2024-03-04 22:44:26Received Zigbee message from 'Aqara T2', type 'readResponse', cluster 'manuSpecificLumi', data '{}' from endpoint 2 with groupID 0
Debug 2024-03-04 22:44:26LLKZMK12LM: Processed data into payload {}
Error 2024-03-04 22:44:26Exception while calling fromZigbee converter: Expected one of: 1, 2, 3, got: 'undefined'}
Debug 2024-03-04 22:44:26Error: Expected one of: 1, 2, 3, got: 'undefined' at getFromLookupByValue (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:592:15) at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:386:64) at Receive.onDeviceMessage (/app/lib/extension/receive.ts:153:51) at processTicksAndRejections (node:internal/process/task_queues:95:5) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)
Error 2024-03-04 22:44:26Failed to configure 'Aqara T2', attempt 5 (Error: Read 0x54ef4410009b3f3b/2 manuSpecificLumi([10], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:319:28) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:469:22) at processTicksAndRejections (node:internal/process/task_queues:95:5) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:65:13) at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:86:17) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:141:25) at Configure.configure (/app/lib/extension/configure.ts:122:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
Info 2024-03-04 22:44:26MQTT publish: topic 'z2medge/bridge/response/device/configure', payload '{"data":{"id":"Aqara T2"},"error":"Failed to configure (Read 0x54ef4410009b3f3b/2 manuSpecificLumi([10], {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4447,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE'))","status":"error","transaction":"yhw5r-1"}'
```

morhimi, 2024-03-04:
forgot to mention that I'm running `ember`

morhimi, 2024-03-24:
Anything I can do to assist in debugging this?

morhimi, 2024-05-04:
any assistance will be appreciated...

Koenkk, 2024-05-05:
Can you go to the exposes tab and change the `switch_type`, does this work?

BeFygo, 2024-08-06:
I'm having a simular problem.

`error 2024-08-06 06:44:51z2m: Exception while calling fromZigbee converter: Expected one of: 1, 2, 3, got: 'undefined'}
error 2024-08-06 06:44:51z2m: Exception while calling fromZigbee converter: Expected one of: 0, 1, 2, 3, got: 'undefined'}
error 2024-08-06 06:44:52z2m: Exception while calling fromZigbee converter: Expected one of: 0, 1, 3, got: 'undefined'}
error 2024-08-06 06:44:52z2m: Exception while calling fromZigbee converter: Value is not a number, got undefined (undefined)}`

changing the `switch_type` doesn't seem to resolve the issue

swannet, 2024-10-07:
I've the same problem,

Switching the switch_type seems to work (it's changed in the state json of the device), but for me the switch_type seems to work incorrectly (I don't know if it its related). I've set the switch_type to momentary but it acts like a toggle switch (even when the state page tell me the switch_type is set tot momentary).

I've enabled debug logging and found this in the logs:

[2024-10-07 20:18:53] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-10-07 20:18:53] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=1 frmTx=1)
[2024-10-07 20:18:53] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=4](frmRx=4) Added to rxQueue
[2024-10-07 20:18:53] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=5](ackRx=1)
[2024-10-07 20:18:53] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=40 Len=59]
[2024-10-07 20:18:53] debug: 	zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":64704,"sourceEndpoint":1,"destinationEndpoint":1,"options":320,"groupId":0,"sequence":127}], [packetInfo:{"senderShortId":54035,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":180,"lastHopRssi":-55,"lastHopTimestamp":0}], [messageContents=1c5f117e0af2ff4119aa751384bd8d0741000f0c23010100040a2308000c640b2003]
[2024-10-07 20:18:53] debug: 	zh:controller: Received payload: clusterID=64704, address=54035, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=180, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":4447,"transactionSequenceNumber":126,"commandIdentifier":10},"payload":[{"attrId":65522,"dataType":65,"attrData":{"type":"Buffer","data":[170,117,19,132,189,141,7,65,0,15,12,35,1,1,0,4,10,35,8,0,12,100,11,32,3]}}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2024-10-07 20:18:53] debug: 	z2m: Received Zigbee message from 'werkkamer_rolluik', type 'attributeReport', cluster 'manuSpecificLumi', data '{"65522":{"data":[170,117,19,132,189,141,7,65,0,15,12,35,1,1,0,4,10,35,8,0,12,100,11,32,3],"type":"Buffer"}}' from endpoint 1 with groupID 0
[2024-10-07 20:18:53] debug: 	zhc:lumi: LLKZMK12LM: unknown key 65522 with value �u���A # 
# d 
[2024-10-07 20:18:53] debug: 	zhc:lumi: LLKZMK12LM: Processed data into payload {}

If this isn't helping, is there anything else I can do to provide some help?

closed 2025-01-07 by github-actions[bot]