Koenkk/zigbee2mqtt #21679: Aqara wall switch H1 (WS-EUK04) - not syncing state, not showing power measurement anymore

adynis, 2024-03-05:
### What happened?

My kids played with one of the ~10 wall switches -  it was showing "red" leds ... 😡 it was removed from Z2M. (probably they pressed more than 6 seconds)
Device: [WS-EUK04](https://www.zigbee2mqtt.io/devices/WS-EUK04.html#aqara-ws-euk04) (lumi.switch.n2aeu1)
I added it back to Z2M (1000 times) and I can not make it work properly:

1. When I press the physical switch button, I **don't get the updated state value in Z2M**, it appears only if I manually press "refresh" icon to get the value, but not automatically. 
2. for some info fields, e.g.: **power, energy, device_temperature I have N/A** . If I press the "refresh icon" I get the error:

```
Zigbee2MQTT:debug 2024-03-05 02:08:30: Publishing get 'get' 'power' to 'copii_aqara'
Zigbee2MQTT:error 2024-03-05 02:08:30: Publish 'get' 'power' to 'copii_aqara' failed: 'TypeError: Cannot read properties of undefined (reading 'read')'
Zigbee2MQTT:debug 2024-03-05 02:08:30: TypeError: Cannot read properties of undefined (reading 'read')
```

Workaround for issue 2. :
I found somewhere on internet this screenshot with the suggestion to add "Reporting" :
[image]
I don't know if this is something ok to do, it looks like it has some slower reaction times then before, I'm not sure if this way I make any process to consume "energy" just to check from second to second the status of the switch ... **somehow I feel that it's not ok to have this workaround** ...

Some remarks:

- the other switches work fine
- I was having this issue also in the past (previous Z2M versions) , but usualy after "dancing with tambourines" in a coupple of triles of removing from network and re-joining (eventually joining to an aqara hub) , I was able to make it work. This time I couldn't after ~1h of trials of remove/re-join, restart HA ... etc 
- I recently changed to 1.36 and ember

Final thought: I'm not sure if it is from aqara device itself (since I was having this kind of issues also in the past), or from Z2M , or from something else ... but it would be helpful to have a solution when it happens. Also ... like I already wrote, the workaround I used doesn't "smell nice" to me now (unless someone confirms to me that it's ok :) )

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

7.4.1 [GA]

### Adapter

Home Assistant Yellow - EZSP v13

### Setup

HomeAssistant on HomeAssistantYellow

mojiro, 2024-03-05:
Hello @adynis,
I have a similar issue.

Is it possible to remember what versions your setup had when you paired them?

At which month/date maybe? - That could tell us which versions you might had..

adynis, 2024-03-07:
oh ... @mojiro , sorry .. I missed your message. I observed it now by accident. I will answer to you immediately after my intended message. 
So .... today .. very strange experiences:

1.) Something ... less important but I can't avoid mentioning it ...
Something very strange which I CAN NOT REPLICATE anymore :( My kid pressed 6 times quickly on the switch and ... "magic" ... for ~3 minutes the relay started to go ON/OFF/ON/OFF/ON/OFF at around 2 changes per second ... Unfortunately I didn't have the idea to start loging it :( My wide made a short video ... When I got home I also tried, it did the same (after 6 clicks it started to on/off/on/off....) and I just checked that when I restarted the Z2M addon from my HA instance, the on/off/on/off stopped. After that I jsut forced a re-join of the switch and .... from that moment I couldn't replicate the issue anymore ....

2.) This I consider it IMPORTANT: 
After the problem from point 1 , the state of the switch was not synced anymore 😠 😢 ... also power was NULL.... I tried agan to rejoin , restart HA, remove, force remove, add to aqara hub, remove, re-add, re-join, etc etc .... nothing.
Then ... I changed back to ezsp !!!
Then I tried to rejoin the device for ~10 minutes (it took me so long because of z2m crashing #21118 , but finally I succeded it)  and ... surprize: state is syncing,  power measurement is working !!!
So ... my conclusion : there's a problem with ember & aqara switch for state syncing & power measurements !

wastez, 2024-03-07:
@Nerivec 
Could that be the same reason like on the ikea motion sensor which we found today?
Maybe aqara do such a same dumb thing on this device?

adynis, 2024-03-07:
> Hello @adynis, I have a similar issue.
> 
> Is it possible to remember what versions your setup had when you paired them?
> 
> At which month/date maybe? - That could tell us which versions you might had..

Sorry for not answering, I missed the mail when you initially wrote.
I do not remember Z2M version, but ... : majority of switches ware installed in ~October.2023
Just that some of them (especially the one from my office-room) I tested a lot and it was removed/re-added multiple times. 

In the same time, like in the poiint 2.) written above, with some effort I was usually succeeding to join it with power & state-sync working. I couldn't yet understand which was the trick to make it work ... I was trying various things until it was working ... . Things I was trying , independently or combined :

- keep 6 seconds pressed on the switch with PermitJoin = OFF
- keep 6 seconds pressedo n the switch with PermitJoin = ON
- Remove it from the z2m button then re-join
- Remove it with force switch , then re-join
- remove it, restart z2m (or HA) then re-join
- add it to an aqara hub, then back to z2m
- this is the most "strange" assumption I was hoping for and testing (I had the feeling it worked once but probably it was jsut a conincidence) : Adding The "Reporting" options like in the picture from the ticket description, then remove it somehow (check previous points) then re-join the device (hoping to somhow maintain the reporting :-D )

The problem was that all these tests were DIFFICULT because of the Z2M restart issue ( https://github.com/Koenkk/zigbee2mqtt/issues/21118 ) . But somehow in ... ~5-30 minutes of testing I was usually able to make it work ... despite the z2m restarts in the middle of "device configuration" or despite all the "non-understandings" which I have with regards to all previous points while testing

wastez, 2024-03-07:
@adynis 
Can you post logs with herdsman debug logging on?
https://www.zigbee2mqtt.io/guide/usage/debug.html

adynis, 2024-03-07:
Yes, but what do you think it would worth "adding" in the log:
- ember vs ezsp
- normal functionality vs not working functionality
- the joining moment
?

Nerivec, 2024-03-07:
@adynis wastez may be right. Hopefully, the [pending ember PR](https://github.com/Koenkk/zigbee-herdsman/pull/959) should also take care of this issue. All newly reported issues (since `ember` release) are being fixed in dev branch at the moment, to allow some testing before release (every 1st of the month). If you can, try with dev/edge version after that PR is merged and available, and let me know if you still encounter the issue.

As for the `magic on/off`, it will be hard to say without debug logs. It almost sounds like the device got into a "special state" by triggering it in a very specific way (kind of like some devices that need to be plugged/unplugged 5 times at regular interval to reset them...).

adynis, 2024-03-07:
@Nerivec : I would be glad to test ! Jsut that .. being in home assistant (on a "productive environment" ) I ask first: can I just change to Z2M-edge addon in HA and all devices will still be there ?

wastez, 2024-03-07:
@adynis 
I wanted to have a look if the problem is the same thing like on ikea motion sensor because it’s very probably. 
But I didn’t know that @Nerivec is sooooooo fast. 
So the best thing is you try the edge release as soon the PR is pushed to it. 
I‘m pretty sure it will solve it.

wastez, 2024-03-07:
@adynis 
Yes everything is there, the same db is used. You can swap between them without problems.

adynis, 2024-03-08:
In the meantime here's one log:

```
log ember 5:
 - 01:51 EET - hobby_aqara (was working fine) -> pressed 6 seconds on the physical switch to force it re-join
	-> PermitJoin was previously enabled; Device was succesfuylly added back with the same name
	-> state not syncing anymore
	-> power measurements not working anymore (but still shows 0 in browser, not NULL, maybe because of a browser caching ? )
 - 01:54 EET - Press "remove device button in Z2M (Permit Join still on)
	-> Device removed and added back succesfully (added with the default name 0x54ef441000720a04 )
	-> onOff not synced, Power , Energy, DeviceTemperature not synced. 
	-> If I press the "refresh" icon in Z2M for onoff state then it works. If I press for Power/Energy/DeviceTemperature then it gives error.
```
[log_ember_5.txt](https://github.com/Koenkk/zigbee2mqtt/files/14531406/log_ember_5.txt)

Nerivec, 2024-03-08:
Flash https://github.com/darkxst/silabs-firmware-builder/blob/ember-nohw/firmware_builds/yellow/ncp-uart-hw-v7.4.1.0-yellow-115200.gbl again, just to be sure, then definitely set `rtscts: false` in config. I'm seeing strange behavior in your logs, that's likely the cause. Then try getting logs again please.

adynis, 2024-03-08:
EZSP logs, 2 files combined, since it was a Z2M restart in the middle:
```
log_ezsp_1_and_2:
 - 2:14 EET - removed device from Z2M interface
  - various things: Z2M crashed once, I pressed again on the physical switch for 6 seconds , then it didn't finish the interview, then I pressed again 6 seconds, then it worked ... 
 - ~2:19 EET - succcesfully added, and things are sincing correctly
 ```
[log_ezsp_1_and_2.txt](https://github.com/Koenkk/zigbee2mqtt/files/14531638/log_ezsp_1_and_2.txt)

adynis, 2024-03-08:
Oh, I've seen you message now, @Nerivec . Oky , I reflash, set rtscts to false and re do the log 👍

adynis, 2024-03-08:
Firmware log:
```
universal-silabs-flasher --device /dev/ttyAMA1 flash --allow-cross-flashing --firmware ncp-uart-hw-v7.4.1.0-yellow-115200.gbl --force 2>&1 | tee log_fw_no_hw3.log
2024-03-08 02:29:28.324 a0d7b954-ssh universal_silabs_flasher.flash INFO Extracted GBL metadata: NabuCasaMetadata(metadata_version=1, sdk_version='4.4.1', ezsp_version='7.4.1.0', ot_rcp_version=None, cpc_version=None, fw_type=<FirmwareImageType.NCP_UART_HW: 'ncp-uart-hw'>, baudrate=115200)
2024-03-08 02:29:28.325 a0d7b954-ssh universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2024-03-08 02:29:30.329 a0d7b954-ssh universal_silabs_flasher.flasher INFO Probing ApplicationType.EZSP at 115200 baud
2024-03-08 02:29:31.570 a0d7b954-ssh universal_silabs_flasher.flasher INFO Detected ApplicationType.EZSP, version '7.4.1.0 build 0' (7.4.1.0.0) at 115200 baudrate (bootloader baudrate None)
2024-03-08 02:29:37.789 a0d7b954-ssh universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2024-03-08 02:29:38.806 a0d7b954-ssh universal_silabs_flasher.flasher INFO Detected bootloader version '2.0.1'
2024-03-08 02:29:38.807 a0d7b954-ssh universal_silabs_flasher.flasher INFO Detected ApplicationType.GECKO_BOOTLOADER, version '2.0.1' at 115200 baudrate (bootloader baudrate 115200)
ncp-uart-hw-v7.4.1.0-yellow-115200.gbl
```
Does this mean that ... it worked ... or not ...  ❓

adynis, 2024-03-08:
a new log:

- firmware:  https://github.com/darkxst/silabs-firmware-builder/blob/ember-nohw/firmware_builds/yellow/ncp-uart-hw-v7.4.1.0-yellow-115200.gbl , with flashing log in the previous comment
- Z2M version 1.36.0 (I will move also to edge, if it helps I can re-do the log on edge)
- ember
- rtscts: false

log_ember_6:
```
 - 2:38 EET 
	- Permit Join
	- 0x54ef441000720a04 device (which was working ok, state synced, power measurements working), press 6 seconds on phisical button
	- device removed and successfully re-added
	- onOff not synced , power measurements not working
 - 2:40 EET
	- press also "remove device" from Z2M interface
	- device successfully removed and re-added
	- onOff not synced , power measurements not working
```

[log_ember_6.txt](https://github.com/Koenkk/zigbee2mqtt/files/14531807/log_ember_6.txt)

Let me know what other logs are helpful to make (ezsp? something else?)

Nerivec, 2024-03-08:
This device is terrible, awful. I'd take a hammer to it if it was mine... 😞
It's spamming the network with duplicate messages for no reason whatsoever; it's like it's trying really hard to crash the network or something...

With `ezsp`, I'd bet the `setManufacturerCode` workaround is the reason behind the crashes; it's triggering this dozens of times. I'm sure the coordinator doesn't like that...
@kirovilya Seems there is a need to throttle the triggering of `setManufacturerCode` somehow when a xiaomi/lumi device joins the network? I'm guessing that's the cause of the crashes...

`ember` manages to survive, but I'm not surprised things aren't working well after that. I'll have to dig deeper...

adynis, 2024-03-08:
Thanks @Nerivec for looking into it ! 
If you need any kind of test/log and you estimate I can do it, please let me know, I'm willing to help :)

kirovilya, 2024-03-08:
@Nerivec Sorry, I don't understand what's wrong?
in the `ezsp` adapter `setManufacturerCode `is set separately to `0x115F` for Lumi/Xiaomi/Aqara devices when they connect to the network

Nerivec, 2024-03-08:
@kirovilya Check the logs [here](https://github.com/Koenkk/zigbee2mqtt/issues/21679#issuecomment-1984832501). You will see the `setManufacturerCode` is triggered in rapid succession from the join handler. For 3 joins, I count 25 `setManufacturerCode` triggered...

kirovilya, 2024-03-08:
@Nerivec hmmm... So many `trustCenterJoinHandler` events - so many `setManufacturerCode`... 
I saw this - the device repeatedly tried to connect to the network, but the coordinator did not provide a key to connect (I looked at this with a sniffer). As a result, the device went offline. And then again...
but I still didn’t understand the reasons.
If you install a new network, it connects normally.

This also happened when connecting via routers - it was solved like this https://github.com/Koenkk/zigbee-herdsman/pull/938

Nerivec, 2024-03-08:
@kirovilya The log in the post I linked is with `ezsp` driver with 1.36.0 (herdsman 0.35.1) so the PR you linked is included (0.34.9). It must be something else; but this Aqara device is definitely no good, so it could be a lot of problems... 😓

kirovilya, 2024-03-08:
I saw similar behavior not with Aqara devices, but with Tuya.
It's as if the coordinator goes into closed mode and stops sending the key. maybe it was a firmware bug

Nerivec, 2024-03-12:
@adynis I was waiting for a few related changes to be released, since you swapped to edge, before coming back to this. Can you update your edge version so you get herdsman 0.38.0, then let the network settle for an hour or so after the update (it changes a few configs, so just to make sure the mesh readjusts as needed).
After that, if you can:

- Get fresh new logs before doing anything more, and after each below step.
- Try to re-pair the device with the coordinator (when permit join, allow only on `Coordinator`), see if you notice changes in behavior.
- Try to re-pair the device with a specific router that is not Aqara/Xiaomi/Lumi, see if you notice changes in behavior.

adynis, 2024-03-12:
Yes. 
Just to confirm what I will do (please correct me if I got it wrong) :
 - I uninstall edge and reinstall edge (to get the latest edge version)
 - I wait ~1h
 - I start loging (eventually I restart Z2M to have a fresh log)
 - I re-pair my Aqara switch (lumi) with the coordinator directly, and check if OnOff is synced and if Power measurements are shown
 - I re-pair my Aqara switch (lumi) with another non-aqara router (despite the ~16 aqara/lumi routers I have, I only have some MiBoxer FUT037Z+ which are (wrongly?) seen as TS0505B ) and again check OnOff sync and Power measurements
  - I put here the log

adynis, 2024-03-12:
so ... It's a big log, but I tried to write my steps to allow you "focus" only on what you find useful:

HomeAssistant Yellow , with Zigbee2Mqtt Edge addon.
 > Zigbee2MQTT version 1.36.0-dev commit: [56feb77](https://github.com/Koenkk/zigbee2mqtt/commit/56feb77)
 > Coordinator revision 7.4.1 [GA]
 > zigbee_herdsman_converters_version 19.1.0
 > zigbee_herdsman_version 0.38.0
Z2m addon config:
 > adapter: ember
 > port: /dev/ttyAMA1
 > baudrate: 115200
 > rtscts: false

Device Under Test: 
 > "0x54ef441000720a04" with name "hobby_aqara"
  >  Before starting the test device was in the network in a "not-working" state with respect to syncing OnOff state from physical buttons and showing measurements.

What I did:
After updating Z2m edge to the newest version, I let it 1h (PS. I have also that log if you need), then I restarted Z2M Edge addon and .. started what's written below:
```
00:55 EET:
 - Permit join on Coordinator ONLY
 - pressed 6 seconds on physical button
 - device removed/readded quickly
 - device still does NOT work (no sync, no measurements)
 - disable join coordinator
00:58 EET:
 - Permit join on non-aqara router
 - device removed but .... not added !?
 - 00:59 pressed on more time 4 6 seconds (it's still counting down the join on this another router)
 - nothing .... 
01:01 EET:
 - Trying another non-aqara router
 - nothing ....
01:03 EET:
 - trying another AQARA (lumi) router (another identical device like the one I'm playing with )
 - nothing ....
01:04 EET:
 - PermitJoin again on coordinator
 - device immediately added with the "hobby_aqara" name
 - Disable "PermitJoin"
01:05 EET:
 - Remove device from "remvoe button" in Z2m (without force)
 - device removed (interesting, I was expecting ~20 "device removed" messages)
01:06 EET:
 - PermitJoin on non-aqara router 
 - nothing ....
01:08 EET:
 - PermitJoin ALL
 - Device added with "0x54ef441000720a04" name
 - Device still doesn't work (no sync, no power mewasurements)
 - DisableJoin ALL
01:09 EET:
 - Remove device from Z2M
1:10 EET:
 - PermitJoin on non-aqara router
 - nothing ...
01:11 EET:
 - thinking what else to test ....
01:11 EET:
 - PermitJoin on coordinator again
 - device added immediately as "0x54ef441000720a04"
01:13 EET:
 - I renamed the device back to hobby_aqara
 - Device still not syncing onOff state from physical button (only if I press "refresh" icon) and  not showing measurements
01:16 EET:
 - something offtopic: a loratap button seen as <...> doesn't work anymore (it was working on previous edge version I had; Maybe I need to re-join also that device ... didn't yet try)
 ```
 
[log_ember_edge_3.txt](https://github.com/Koenkk/zigbee2mqtt/files/14579817/log_ember_edge_3.txt)

Nerivec, 2024-03-13:
Thanks! Details much appreciated!
Can you re-upload the log file? Seems github lost it, I'm getting a 404.
Can you also post a network map while we're at it?

_It tells you just how much that device's firmware is broken when it doesn't even want to join with another Aqara router..._ 😞

adynis, 2024-03-13:
Something is very strange with uploading files ... I can not make it work ... I try also in this comment.
Regarding the network map ... it's big and hard to see ... let me know what do you exactly need, to zoom there :) 
[image]
[log_ember_edge_3.txt](https://github.com/Koenkk/zigbee2mqtt/files/14580199/log_ember_edge_3.txt)

Nerivec, 2024-03-13:
> Something is very strange with uploading files ... I can not make it work ... I try also in this comment.

Might be some temporary github problem.

> let me know what do you exactly need, to zoom there :)

_I've found the best way is usually to make an outer circle like you did with green devices (end devices), grouping them near their router, and then a smaller circle inside with blue devices (routers/coordinator)._ I'm most interested in the links between those Aqara routers (including the coordinator).

adynis, 2024-03-13:
Now I'm adding more screenshots (I challange github while attachments do not work properly ... ) 
PS. I hope you were able to get the log from the "external" link I've added in that previous post https://github.com/Koenkk/zigbee2mqtt/issues/21679#issuecomment-1992734515

After I arranged as you suggested me, I found 3 things as interesting :
 - Coordinator is connected to all routers
 - all routers are connected ALMOST to all other routers
 - that green "end device" which I pus in the left side is very "special" ... it's actually LIKE another existing router, BUT looking now after seing this map it seems it's the "CN" version and it does not appear with blue, as a router. It is [DLKZMK11LM](https://www.zigbee2mqtt.io/devices/DLKZMK11LM.html#aqara-dlkzmk11lm) , it appears as an End Device, but it seems connected to almost all other routers and it's very similar to [SSM-U01](https://www.zigbee2mqtt.io/devices/SSM-U01.html#aqara-ssm-u01) which is "blue" in my case. 

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Nerivec, 2024-03-13:
I got the log file yes. Much cleaner with new 0.38.0 version (no more duplicate "device left"), so that's a plus. Unfortunately, it isn't very descriptive in the problems without herdsman debug enabled. I can tell you however, that all the routers you tried to pair the Aqara device to, flat out denied it from joining.

```log
[TRUST CENTER] Device 24426:0x54ef441000720a04 was denied joining via 59335.
[TRUST CENTER] Device 6500:0x54ef441000720a04 was denied joining via 45709.
[TRUST CENTER] Device 55155:0x54ef441000720a04 was denied joining via 59218.
```

Any chance you could try [enabling herdsman debug](https://www.zigbee2mqtt.io/guide/usage/debug.html#home-assistant-os-supervised-addon), then re-do one remove/pairing with Coordinator only, and one remove/pairing with a specific router (one from your previous tests, doesn't matter which)? Then you can disable herdsman debug and restart Z2M.

> Coordinator is connected to all routers

That's normal, through the mesh, they all eventually go back to the coordinator.

> that green "end device" which I pus in the left side is very "special"

Likely another Aqara firmware "mishap" where the device is reporting as an end device, even though it is a router... 😞

_Offtopic: I can also tell you from your network map, that most of your end devices have pretty poor connections (below or near 100 LQI). I'm not sure if that makes sense (I'm guessing not!) based on the areas/distances to routers/coordinator, but on top of possibly creating network issues, that can seriously affect battery life. If you have a strategic place where it could be put (need a socket and an old phone charger or something like that), I'd suggest getting a coordinator that you can flash as dedicated router (example [here](https://www.zigbee2mqtt.io/devices/ZBDongle-E.html#how-to-create-a-sonoff-zbdongle-e-router)). The device can be quickly paid off by not spending $$$ on batteries... and should increase reliability by a large margin in the area it covers._

adynis, 2024-03-13:
Oh ... I forgot to enable hardsman debug in the edge addon 🤦‍♂️ sorry.
Yes, I'll re-do the test tonight (in ~10 hours) , no problem. 

Related to LQI ... interesting ... in the Z2M main page, with the list of devices (Except ~4 exceptions) all numbers are much bigger .. 
All above 100, Majority around 200. I'll check tonight some concrete examples but now it looks like there are different LQI numbers in Z2M devices page vs. LQI numbers from the map 🤔

adynis, 2024-03-13:
I re-did the test.
The only differences in settings comparing to https://github.com/Koenkk/zigbee2mqtt/issues/21679#issuecomment-1992734515 are:
 - I reinstalled edge so no I have version:
> 1.36.0-dev commit: [720fc49](https://github.com/Koenkk/zigbee2mqtt/commit/720fc49)
 - I enabled zigbee_herdsman_debug

```
00:14 EET:
 - Permit Join Coordinator
 - Device removed and readded quickly ("hobby_aqara" )
 - Device not working (no onoff sync, no power measurements)
 - Disable join
00:16 EET:
 - Permit Join - non-aqara router 1
 - (waiting all ~4 minutes just to be sure)
 - Pressing one more time 6 seconds on phisical button 
 - device NOT added !
00:21 EET:
 - Trying another non-aqara router 2
 - nothing ... device not added
00:22 EET:
 - Trying a similar aqara router
 - nothing ... (PS. this router is ... ~1.5m distance away)
00:23 EET:
 - Permit join (ALL)
 - immediately added (in ~1 sec)
 - Disable join
00:25 EET:
 - remove device in Z2M (without force)
 - Permit Join (again) : non-aqara router 1
 - nothing ....
00:28 EET:
 - Permit join (Coordinator)
 - Added instantly as "0x54ef441000720a04"
 - Device not working (no OnOff state sync, no power measurements)
00:31 EET:
 - tested also the button , I still have that error : " No converter available for 'TS0042' with cluster 'genOnOff' and type 'commandTuyaAction' and data '{"value":0}'"
00:32 EET:
 - Rewmove device from Z2m with "force" option
 - Permit Join Coordinator
 - rename back to "hobby_aqara"
 - Device not working (no OnOff state sync, no power measurements)
00:34 EET:
 - I stopped Z2M addon
 ```
 
[log_ember_edge_5.txt](https://github.com/Koenkk/zigbee2mqtt/files/14594531/log_ember_edge_5.txt)

adynis, 2024-03-18:
Something I just observerd:
Shortly:
on EZSP mode it did work only 2nd or 3rd time, not necessary from the 1st time, but I got a log with the 2 failed trials and 1 successfull trial of joining the device on ezsp so that OnOff are synced and measurements are shown.

Longer:
I had some strange behavior on ember (one door sensor did not update since 1 week. I re-added to network and ... quite immediately 3 OTHER devices left the network :-| :-| :-| (2 switches like the one in this thread and one MiBoxer RGBCCT controller). Since those switches had some automations on phisical-button-pressed, I moved back to ezsp to add them in that way when they work propertly. On both ... I observed something: 1st re-join it still didn't work, but 2nd re-join it worked. So ... I said ... let's do a log with this example, maybe it helps seeing the difference (on ezsp) for when it doesn't work versus when it works. So ... here it is:

 - HA with Z2M edge with the version from ~4days ago (the time of my last comment) :
> 1.36.0-dev commit: [720fc49](https://github.com/Koenkk/zigbee2mqtt/commit/720fc49)

 - zigbee_herdsman_debug enabled

 - configuration in HA Z2M addon: EZSP

```
~02:30:
 - Permit Join All
 - Press 6 seconds on the switch to re-join
 - immediately removed and re-joined, but still N/A, and still no OnOff sync
~02:31:
 - Press 6 seconds on the switch to re-join
 - immediately removed and re-joined, but still N/A, and still no OnOff sync
~02:32:
 - Press 6 seconds on the switch to re-join
 - immediately removed and re-joined, and it WORKS !! Power, Energy and Device Temperature have values. State is updated in Z2M as I press the buttons on the wall-switch.
 - Disable Permit Join

After that I stoped Z2M, changed back to ember, and the switch still works. 
```

[log_ezsp_edge_2.txt](https://github.com/Koenkk/zigbee2mqtt/files/14629584/log_ezsp_edge_2.txt)

Nerivec, 2024-03-18:
The `ember` fix (hopefully) should be included in next herdsman version (the PR is merged but the version not yet).
https://github.com/Koenkk/zigbee-herdsman/pull/975

Let me know how it goes after it is available 😉

adynis, 2024-03-20:
Initial quick & partial test:
> zigbee_herdsman_version 0.39.0

IT WORKS !!!!!!!!!! 🎉 🎉 🎉  

I removed my "hobby_aqara" and re-joined it 2 times (like previously in last weeks) and now it immediately worked each time: OnOff is synced, measurements values are shown !! 🥳  I need to test more tomorrow evening (because now I can't stay anymore too much) but at least on the first quick, short test it worked. Thank you, @Nerivec . I will come back after a longer testing.

adynis, 2024-03-24:
After more days of testing, I can really confirm some things:

- Joining the mentioned Aqara switches works fine !! (at least on ember, I didn't test ezsp anymore)  (joining is very rapid, OnOff gets quickly synced, those measurements work fine !! Great update, thank you !!! 🎉 
- just that @Nerivec asked me previously, I con confirm something which does NOT yet work for me (altough not relevant, i think, for this ticket) : I am unable to join WS-EUK04 on anything else than the coordinator ! in case it's needed, I add a log below.

Log for showing the error of trying to joini on a router, not on coordintor:

```
2024-03-24T00:47:39.081Z zigbee-herdsman:adapter:ember:ezsp ezspTrustCenterJoinHandler(): callback called with: [newNodeId=14428], [newNodeEui64=0x54ef441000720a04], [status=STANDARD_SECURITY_UNSECURED_JOIN], [policyDecision=DENY_JOIN], [parentOfNewNodeId=14880]
[TRUST CENTER] Device 14428:0x54ef441000720a04 was denied joining via 14880.
2024-03-24T00:47:39.082Z zigbee-herdsman:adapter:ember:uart:ash:parser <<<< [FRAME raw=75a6b1a9702a498a5d9e3825ba7d317d5d1d9d4c079188e77e]
2024-03-24T00:47:39.082Z zigbee-herdsman:adapter:ember:uart:ash <--- [FRAME type=DATA]
2024-03-24T00:47:39.082Z zigbee-herdsman:adapter:ember:uart:ash <--- [FRAME type=DATA ackNum=5]
2024-03-24T00:47:39.082Z zigbee-herdsman:adapter:ember:uart:ash <--- [FRAME type=DATA ackNum=5 frmNum=7] Added to rxQueue
2024-03-24T00:47:39.084Z zigbee-herdsman:adapter:ember:uart:ash ---> [FRAME type=ACK frmRx=0]
2024-03-24T00:47:39.085Z zigbee-herdsman:adapter:ember:uart:ash:writer >>>> [FRAME raw=8070787e]
2024-03-24T00:47:39.086Z zigbee-herdsman:adapter:ember:ezsp <=== [FRAME: ID=36:"TRUST_CENTER_JOIN_HANDLER" Seq=228 Len=19]
2024-03-24T00:47:39.087Z zigbee-herdsman:adapter:ember:ezsp ezspTrustCenterJoinHandler(): callback called with: [newNodeId=14428], [newNodeEui64=0x54ef441000720a04], [status=STANDARD_SECURITY_UNSECURED_JOIN], [policyDecision=DENY_JOIN], [parentOfNewNodeId=14880]
```
full log (search for : 'was denied joining via' ) :

[log_ember_edge_22.txt](https://github.com/Koenkk/zigbee2mqtt/files/14734053/log_ember_edge_22.txt)

One last question: should I be worried about those strange messages ? (especially those with many not-ASCII values) like: 
```
WS-EUK03: unknown key 223 with value  #6...
WS-EUK03: unknown key 229 with value  �ev$....
WS-EUK04: unknown key 14 with value 0
WS-EUK04: unknown key 220 with value 맏-�C  K���
WS-EUK04: unknown key 223 with value ! 
WS-EUK04: unknown key 252 with value 0
RTCGQ14LM: unknown key 19 with value 0
RTCGQ14LM: unknown key 20 with value 0
SP-EUC01: unknown key 155 with value 1
```

Nerivec, 2024-03-24:
The workaround to make these Aqara devices work is to make them think the coordinator is the Aqara Hub, so I'm not surprised that they don't pair properly with other devices. Basically Aqara is trying to prevent you from using their devices without their hub...
_My advice: when you add/replace devices from now on, stay far away from Aqara..._

> should I be worried about those strange messages ?

These are just unknown values that are not hexified before being printed to logs. Not a problem, just unsupported attributes for a device model. When the device has all expected attributes working in Z2M, that usually means the brand is using extra attributes to exchange with their hub & co, that Z2M doesn't care about.

I'll let you close this issue if that's alright with you. Thanks ;)

adynis, 2024-03-24:
Thanks a lot @Nerivec for fixing this topic !  🎉 🥳 🥇 
The problem stated at the begining is now fixed (tested in edge version, soon will be in public release) so I'm closing the issue .

closed 2024-03-24 by adynis