Koenkk/zigbee2mqtt #21685: Z2M Home Assistant Addon settings are ignored

Stooovie, 2024-03-05:
### What happened?

I have rebuilt my Z2M network from scratch (taking five hours), setting channel 25 in the Z2M Home Assistant addon Settings. After setting everything up, I noticed this settings is completely ignored, and I'm still apparently using channel 11.

### What did you expect to happen?

For Z2M network to use Channel 25 as instructed.

### How to reproduce it (minimal and precise)

Install Z2M addon in HA, go to Addons/Zigbee2MQTT/Settings, Edit in YAML, edit Channel

### Zigbee2MQTT version

1.36.0-1

### Adapter firmware version

20230507

### Adapter

Slae'sh 2652RB

### Setup

Add-on for Home Assistant (HA installedProxmox VM

lux73, 2024-03-06:
AFAIK the HA Addon will not provided by @Koenkk - you should get in touch with HA Developers

it shares some Code Base with z2m, that's all...

Stooovie, 2024-03-06:
Will do, thanks, it's egregious

brdcg, 2024-03-07:
I have a similar Problem.
Installed Z2M as Home Assistant Addon. Now I wanted to change the MQTT Server from mosquitto to EMQX.
When I change the mqtt server settings in configuraton.yaml, Z2M still tries to connect to mosquitto.
I have the suspicion that the configuration.yaml under config/zigbee2mqtt is being ignored.

closed 2024-10-04 by github-actions[bot]