Koenkk/zigbee2mqtt #21688: Luminance door sensor TZE200_pay2byax

malac1, 2024-03-05:
### What happened?

I have 8 pieces Luminance door sensor Zigbee Model: TS0601, Zigbee Manufacturer: _TZE200_pay2byax, Manufacturer: TuYa
The sensors were integrated into my HomeAssistant system with Zigbee2MQTT without problems, the interview was done without any error messages. But I see one drawback. The contact works without problems, the status and entities "binary_sensor.door_contact" can be seen. These entities can be used to create automations.
The problem is with the luminance sensor. I can see the luminance value in Z2M - Exposes, but there are no entities associated with the luminance value.
[image]
[image]
[image]
[image]

### What did you expect to happen?

I would like to use the luminance value to create automations. For example "Turn on the light when the luminance value is less than 500lx". As it is, with this integration into HA the Luminance sensor cannot be used for anything.

### Zigbee2MQTT version

1.36.0-1

### Adapter firmware version

Firmware: 1.36.0

### Adapter

ZigBee 3.0 Stick CC2652P Sonoff

### Setup

Add-on on Home Assistant OS on Intel x64

Koenkk, 2024-03-05:
[image]

They are there but disabled by default, click on them to enable

closed 2024-03-05 by Koenkk

malac1, 2024-03-05:
Thanks Koenkk for the quick response and help! I'm new to HA and sometimes a detail gets stuck. I hope you understand. Kind regards and thanks again.