Koenkk/zigbee2mqtt #21691: New with 1.36.0: State of Ubisys S1 not properly reflected / reported from and to Home Assistant

m0wlheld, 2024-03-06:
### What happened?

When changing state of an Ubisys S1 in Home Assistant, the state is not properly reflected in Z2M or reported by it. Turning the switch 'on' via HA UI will display it as 'on' for a second in HA, then turn back to 'off'. In the device's detail view in Z2M, the State is not updated to 'on' unless I hit the refresh button next to it. 

Check the debug log:

although the device itself turns 'on', the state is yet reported as 'off' with the exception of state_l1, with is 'on'. 

```
MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":2.98,"linkquality":18,"power":8,"power_on_behavior":"off","state":"OFF","state_l1":"ON","update":{"installed_version":37749810,"latest_version":37749810,"state":"idle"},"update_available":null}'
`´`

clicking state refresh button in Z2M UI afterwards will make it report the proper state:

```
MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":2.98,"linkquality":14,"power":8,"power_on_behavior":"off","state":"ON","state_l1":"ON","update":{"installed_version":37749810,"latest_version":37749810,"state":"idle"},"update_available":null}'
```

* The device will actually change state as requested.
* That issue has been introduced with 1.36.0
* Reverting to 1.35.3 will make it disappear.
* Seems to be limited to Ubisys S1, other device (ICZB-IW11SW) is not affected

### What did you expect to happen?

Changing switche's state using HA UI or Z2M is properly recognized.

### How to reproduce it (minimal and precise)

Variant 1: 
* in Z2M UI open device's detail page
* change state of device (on to off or off to on, doesn't matter)
* check if the state is properly reflected in HA's UI (it is not)
* click the refresh button in Z2M, next to the device's state
* check if the state is property reflected in HA's UI (it is now)

Variant 2:
* in HA UI change the state of the device. UI will reflect the new state for a second, then return to previous state. 
* check if the state is properly reflected in Z2M UI's device details (it is not)
* click the refresh button in Z2M, next to the device's state. This will not change the state display in Z2M, it is still wrong
* check if the state is property reflected in HA's UI (it is now)

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

20230507

### Adapter

Slaesh's CC2652RB

### Setup

Z2M container on RPi 3B, HA container on x86. MQTT integration in HA

### Debug log

# Variant 1

## Changing state via Z2M device details (off to on)

```
Zigbee2MQTT:debug 2024-03-06 07:48:13: Received MQTT message on 'zigbee2mqtt/0x001fee0000006cd5/set' with data '{"state":"ON"}'
Zigbee2MQTT:debug 2024-03-06 07:48:13: Publishing 'set' 'state' to '0x001fee0000006cd5'
Zigbee2MQTT:info  2024-03-06 07:48:13: MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":2.98,"linkquality":18,"power":0,"power_on_behavior":"off","state":"OFF","state_l1":"ON","update":{"installed_version":37749810,"latest_version":37749810,"state":"idle"},"update_available":null}'
Zigbee2MQTT:debug 2024-03-06 07:48:14: Received Zigbee message from '0x001fee0000006cd5', type 'attributeReport', cluster 'seMetering', data '{"instantaneousDemand":8}' from endpoint 3 with groupID 0
Zigbee2MQTT:info  2024-03-06 07:48:14: MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":2.98,"linkquality":18,"power":8,"power_on_behavior":"off","state":"OFF","state_l1":"ON","update":{"installed_version":37749810,"latest_version":37749810,"state":"idle"},"update_available":null}'
[... 50 lines cut ...]
Zigbee2MQTT:debug 2024-03-06 07:43:54: Received MQTT message on 'zigbee2mqtt/0x001fee0000006cd5/get' with data '{"state":""}'
Zigbee2MQTT:debug 2024-03-06 07:43:54: Publishing get 'get' 'state' to '0x001fee0000006cd5'
Zigbee2MQTT:debug 2024-03-06 07:43:54: Received Zigbee message from '0x001fee0000006cd5', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2024-03-06 07:43:54: MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":2.98,"linkquality":47,"power":8,"power_on_behavior":"off","state":"ON","state_l1":"ON","update":{"installed_version":37749810,"latest_version":37749810,"state":"idle"},"update_available":null}'
```

## Changing state in HA UI (on to off):

```
Zigbee2MQTT:debug 2024-03-06 07:44:23: Received MQTT message on 'zigbee2mqtt/0x001fee0000006cd5/set' with data 'OFF'
Zigbee2MQTT:debug 2024-03-06 07:44:23: Publishing 'set' 'state' to '0x001fee0000006cd5'
Zigbee2MQTT:info  2024-03-06 07:44:24: MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":2.98,"linkquality":47,"power":8,"power_on_behavior":"off","state":"ON","state_l1":"OFF","update":{"installed_version":37749810,"latest_version":37749810,"state":"idle"},"update_available":null}'
Zigbee2MQTT:debug 2024-03-06 07:44:24: Received Zigbee message from '0x001fee0000006cd5', type 'attributeReport', cluster 'seMetering', data '{"instantaneousDemand":0}' from endpoint 3 with groupID 0
Zigbee2MQTT:info  2024-03-06 07:44:24: MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":2.98,"linkquality":51,"power":0,"power_on_behavior":"off","state":"ON","state_l1":"OFF","update":{"installed_version":37749810,"latest_version":37749810,"state":"idle"},"update_available":null}'
Zigbee2MQTT:debug 2024-03-06 07:44:24: Received Zigbee message from '0x001fee0000006cd5', type 'readResponse', cluster 'seMetering', data '{"currentSummDelivered":[0,2984]}' from endpoint 3 with groupID 0
Zigbee2MQTT:info  2024-03-06 07:44:24: MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":2.98,"linkquality":47,"power":0,"power_on_behavior":"off","state":"ON","state_l1":"OFF","update":{"installed_version":37749810,"latest_version":37749810,"state":"idle"},"update_available":null}'
```

## Clicking refresh state button in Z2M afterwards:

```
Zigbee2MQTT:debug 2024-03-06 07:44:52: Publishing get 'get' 'state' to '0x001fee0000006cd5'
Zigbee2MQTT:debug 2024-03-06 07:44:52: Received Zigbee message from '0x001fee0000006cd5', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2024-03-06 07:44:52: MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":2.98,"linkquality":51,"power":0,"power_on_behavior":"off","state":"OFF","state_l1":"OFF","update":{"installed_version":37749810,"latest_version":37749810,"state":"idle"},"update_available":null}'
```

chris-1243, 2024-03-06:
I do have to S2 and I don't have such problem.
A channel of one is bound directly to a light with specific configuration to able to dim a bulb via a "normal" switch.

As the S2 has two channels, I must use state_l1/l2 when triggering them. Here may be the difference.

sjorge, 2024-03-06:
My S1 and S2 still toggle and report the state fine (I don't use homeassistant though, so that is over mqtt or via the frontend)

The fact that your S1 has a state and state_l1 is a bit weird. It should not have a state_l1

m0wlheld, 2024-03-06:
The S1 is single line, yet there is „state“ and „state_l1“. Don‘t know, if this is new.

sjorge, 2024-03-06:
It definitely should not use state_l1 as the S1/S1-R do not have the multiEndpoint: true set in the meta var (which is the correct setting)

S1 (no multiEndpoint, so only state): https://github.com/Koenkk/zigbee-herdsman-converters/blob/604bf5a7e1ea1659836ee8972e68c92318187762/src/devices/ubisys.ts#L553
S2 (has multiEndpoint -> state_l1 and state_l2): https://github.com/Koenkk/zigbee-herdsman-converters/blob/604bf5a7e1ea1659836ee8972e68c92318187762/src/devices/ubisys.ts#L652

m0wlheld, 2024-03-06:
> It definitely should not use state_l1 as the S1/S1-R do not have the multiEndpoint: true set in the meta var (which is the correct setting)
>
> S1 (no multiEndpoint, so only state): [...]

I am sorry, but I don't understand that. Is there something that should be configured?

chris-1243, 2024-03-06:
@sjorge It may be a bit drastic but would you think @m0wlheld should remove and pair again its device ? It may solve is problem.

I use Z2M version 1.36.0 and homeassistant as well. My S2 are both working perfectly and the reposrting is fine as well

sjorge, 2024-03-06:
> @sjorge It may be a bit drastic but would you think @m0wlheld should remove and pair again its device ? It may solve is problem.

That might help, but I'm puzzle why there is a `state_l1` in the state payload for the device, it definitely should not be there. Which hints at something being wrong somewhere. It might also help to stop z2m and clear the cached state for the devices.

chris-1243, 2024-03-06:
I can not help more as I do not have any S1 in my network. Only S2 and C4.

m0wlheld, 2024-03-06:
The state_l1 attribute for my S1 is already present with Z2M 1.35.3, yet it does not negatively affect the proper function using Z2M UI or HA UI. 

This is the debug output when turning the device 'on' via Z2M UI:

```
Zigbee2MQTT:debug 2024-03-06 12:05:58: Received MQTT message on 'zigbee2mqtt/0x001fee0000006cd5/set' with data '{"state":"ON"}'
Zigbee2MQTT:debug 2024-03-06 12:05:58: Publishing 'set' 'state' to '0x001fee0000006cd5'
Zigbee2MQTT:info  2024-03-06 12:05:58: MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":2.99,"linkquality":29,"power":0,"power_on_behavior":"off","state":"ON","state_l1":"OFF","update":{"installed_version":37749810,"latest_version":37749810,"state":"idle"},"update_available":null}'
Zigbee2MQTT:debug 2024-03-06 12:05:59: Received Zigbee message from '0x001fee0000006cd5', type 'attributeReport', cluster 'seMetering', data '{"instantaneousDemand":8}' from endpoint 3 with groupID 0
Zigbee2MQTT:info  2024-03-06 12:05:59: MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":2.99,"linkquality":29,"power":8,"power_on_behavior":"off","state":"ON","state_l1":"OFF","update":{"installed_version":37749810,"latest_version":37749810,"state":"idle"},"update_available":null}'
Zigbee2MQTT:debug 2024-03-06 12:05:59: Received Zigbee message from '0x001fee0000006cd5', type 'readResponse', cluster 'seMetering', data '{"currentSummDelivered":[0,2986]}' from endpoint 3 with groupID 0
Zigbee2MQTT:info  2024-03-06 12:05:59: MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":2.99,"linkquality":25,"power":8,"power_on_behavior":"off","state":"ON","state_l1":"OFF","update":{"installed_version":37749810,"latest_version":37749810,"state":"idle"},"update_available":null}'
```

Same thing, but with 1.36.0:

```
Zigbee2MQTT:debug 2024-03-06 07:48:13: Received MQTT message on 'zigbee2mqtt/0x001fee0000006cd5/set' with data '{"state":"ON"}'
Zigbee2MQTT:debug 2024-03-06 07:48:13: Publishing 'set' 'state' to '0x001fee0000006cd5'
Zigbee2MQTT:info  2024-03-06 07:48:13: MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":2.98,"linkquality":18,"power":0,"power_on_behavior":"off","state":"OFF","state_l1":"ON","update":{"installed_version":37749810,"latest_version":37749810,"state":"idle"},"update_available":null}'
Zigbee2MQTT:debug 2024-03-06 07:48:14: Received Zigbee message from '0x001fee0000006cd5', type 'attributeReport', cluster 'seMetering', data '{"instantaneousDemand":8}' from endpoint 3 with groupID 0
Zigbee2MQTT:info  2024-03-06 07:48:14: MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":2.98,"linkquality":18,"power":8,"power_on_behavior":"off","state":"OFF","state_l1":"ON","update":{"installed_version":37749810,"latest_version":37749810,"state":"idle"},"update_available":null}'
Zigbee2MQTT:debug 2024-03-06 07:48:14: Received Zigbee message from '0x001fee0000006cd5', type 'readResponse', cluster 'seMetering', data '{"currentSummDelivered":[0,2984]}' from endpoint 3 with groupID 0
Zigbee2MQTT:info  2024-03-06 07:48:14: MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":2.98,"linkquality":18,"power":8,"power_on_behavior":"off","state":"OFF","state_l1":"ON","update":{"installed_version":37749810,"latest_version":37749810,"state":"idle"},"update_available":null}'
```

Pay attention, that with Z2M 1.36.0 changing the state via Z2M UI obviously affect `state_l1`, while with 1.35.3 it affects `state`.

> It might also help to stop z2m and clear the cached state for the devices.

How?

> I use Z2M version 1.36.0 and homeassistant as well. My S2 are both working perfectly and the reposrting is fine as well

This issue is with S1.

m0wlheld, 2024-03-08:
> That might help, but I'm puzzle why there is a `state_l1` in the state payload for the device, it definitely should not be there. Which hints at something being wrong somewhere. It might also help [...]

@sjorge How do I clear the cache?

m0wlheld, 2024-03-08:
Thanks, did as recommended and deleted that device specific state. However - it did not solve the issue.

I did
 
1. stop Z2M 1.35.3
1. removed the device specific state from cache
1. started Z2m 1.35.3 - this did not alter `state.json`
1. stopped ZM2 1.35.3 again 

Stopping Z2M again did add an object to cache file for that Ubisys S1 switch:

```json
    "0x001fee0000006cd5": {
        "state": "OFF",
        "power": 0,
        "energy": 3.05,
        "linkquality": 47
    }
```

Next, I updated the Z2M image tag in my Docker compose file to 1.36.0 and re-deployed. This includes stopping the Z2M 1.35.3 container instance, pull the 1.36.0 container image and start a 1.36.0 container instance. 

Without any interaction, neither with the Z2M UI nor Home Assistant, I then stopped the Z2M 1.36.0 container again. This did not alter the entry in cache file, still no sight of `state_l1`. 

After having re-started the Z2M 1.36.0 container instance, I switched the device 'ON' using Z2M's device detail view. The logging output did then contain `state_l1`:

```
Zigbee2MQTT:debug 2024-03-08 08:38:50: Received MQTT message on 'zigbee2mqtt/0x001fee0000006cd5/set' with data '{"state":"ON"}'
Zigbee2MQTT:debug 2024-03-08 08:38:50: Publishing 'set' 'state' to '0x001fee0000006cd5'
Zigbee2MQTT:info  2024-03-08 08:38:50: MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":3.05,"linkquality":47,"power":0,"power_on_behavior":null,"state":"OFF","state_l1":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Zigbee2MQTT:debug 2024-03-08 08:38:51: Received Zigbee message from '0x001fee0000006cd5', type 'attributeReport', cluster 'seMetering', data '{"instantaneousDemand":8}' from endpoint 3 with groupID 0
Zigbee2MQTT:info  2024-03-08 08:38:51: MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":3.05,"linkquality":40,"power":8,"power_on_behavior":null,"state":"OFF","state_l1":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Zigbee2MQTT:debug 2024-03-08 08:38:51: Received Zigbee message from '0x001fee0000006cd5', type 'readResponse', cluster 'seMetering', data '{"currentSummDelivered":[0,3052]}' from endpoint 3 with groupID 0
Zigbee2MQTT:info  2024-03-08 08:38:51: MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":3.05,"linkquality":43,"power":8,"power_on_behavior":null,"state":"OFF","state_l1":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
```

Consequently, when shutting down the Z2M 1.36.0 container, the `state_l1` was added to the device's cache object:

```json
    "0x001fee0000006cd5": {
        "state": "OFF",
        "power": 8,
        "energy": 3.05,
        "linkquality": 47,
        "state_l1": "ON"
    }
```

So, I must conclude that this issue was introduced with Zigbee2MQTT 1.36.0 (or any daily build after 1.35.3). It is caused by Z2M, as it happens even without Home Assistant being involved.

I am back on 1.35.3 now and happy to provide more debug or logging output to solve this issue.

m0wlheld, 2024-03-08:
> It definitely should not use state_l1 as the S1/S1-R do not have the multiEndpoint: true set in the meta var (which is the correct setting)
>
> S1 (no multiEndpoint, so only state): [...]

The "multiendpoint" key is missing in line 553, but present and set to `true` in line 652. But - did the default value for this attribute change, so that is has to explicitely set to `false` for the S1?

Just a thought ...

sjorge, 2024-03-08:
Nope, still default to undefined, which is treated the same as false.

https://github.com/Koenkk/zigbee-herdsman-converters/blob/78c85167ca24c834c22687efcd21f48d1bf1f66b/src/lib/utils.ts#L142

@Koenkk anything I am overlooking where all of a sudden in some cases a endpoint prefix is always added? I can't reproduce this on dev with my S1's. (I have hass integration disabled though)

m0wlheld, 2024-03-08:
Using Z2M 1.36.0-dev commit: [0e0eccb](https://github.com/Koenkk/zigbee2mqtt/commit/0e0eccb) container image, `state_l1` is still present, but not used any more. I think it is present due to it's cache object in `state.json` (created by Z2M 1.36.0).  However it does not alter anymore, the device's state is transported in `state` and properly reflected in Home Assistant's UI.

Turning device on using Z2M 1.36.0-dev UI:

```
Zigbee2MQTT:debug 2024-03-08 10:06:24: Received MQTT message on 'zigbee2mqtt/0x001fee0000006cd5/set' with data '{"state":"ON"}'
Zigbee2MQTT:debug 2024-03-08 10:06:24: Publishing 'set' 'state' to '0x001fee0000006cd5'
Zigbee2MQTT:info  2024-03-08 10:06:24: MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":3.05,"linkquality":58,"power":0,"power_on_behavior":null,"state":"ON","state_l1":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Zigbee2MQTT:debug 2024-03-08 10:06:25: Received Zigbee message from '0x001fee0000006cd5', type 'attributeReport', cluster 'seMetering', data '{"instantaneousDemand":8}' from endpoint 3 with groupID 0
Zigbee2MQTT:info  2024-03-08 10:06:25: MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":3.05,"linkquality":51,"power":8,"power_on_behavior":null,"state":"ON","state_l1":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Zigbee2MQTT:debug 2024-03-08 10:06:25: Received Zigbee message from '0x001fee0000006cd5', type 'readResponse', cluster 'seMetering', data '{"currentSummDelivered":[0,3053]}' from endpoint 3 with groupID 0
Zigbee2MQTT:info  2024-03-08 10:06:25: MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":3.05,"linkquality":29,"power":8,"power_on_behavior":null,"state":"ON","state_l1":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
```

Turning off again:

```
Zigbee2MQTT:debug 2024-03-08 10:08:12: Received MQTT message on 'zigbee2mqtt/0x001fee0000006cd5/set' with data '{"state":"OFF"}'
Zigbee2MQTT:debug 2024-03-08 10:08:12: Publishing 'set' 'state' to '0x001fee0000006cd5'
Zigbee2MQTT:info  2024-03-08 10:08:12: MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":3.05,"linkquality":54,"power":8,"power_on_behavior":null,"state":"OFF","state_l1":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Zigbee2MQTT:debug 2024-03-08 10:08:13: Received Zigbee message from '0x001fee0000006cd5', type 'attributeReport', cluster 'seMetering', data '{"instantaneousDemand":0}' from endpoint 3 with groupID 0
Zigbee2MQTT:info  2024-03-08 10:08:13: MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":3.05,"linkquality":58,"power":0,"power_on_behavior":null,"state":"OFF","state_l1":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Zigbee2MQTT:debug 2024-03-08 10:08:13: Received Zigbee message from '0x001fee0000006cd5', type 'readResponse', cluster 'seMetering', data '{"currentSummDelivered":[0,3053]}' from endpoint 3 with groupID 0
Zigbee2MQTT:info  2024-03-08 10:08:13: MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":3.05,"linkquality":54,"power":0,"power_on_behavior":null,"state":"OFF","state_l1":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
```

After stopping Z2M latest-dev, removing the device object from cache and restarting, the `state_l1` attribute is gone:

```
Zigbee2MQTT:debug 2024-03-08 10:11:10: Received MQTT message on 'zigbee2mqtt/0x001fee0000006cd5/set' with data '{"state":"ON"}'
Zigbee2MQTT:debug 2024-03-08 10:11:10: Publishing 'set' 'state' to '0x001fee0000006cd5'
Zigbee2MQTT:info  2024-03-08 10:11:10: MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":null,"linkquality":58,"power":null,"power_on_behavior":null,"state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Zigbee2MQTT:debug 2024-03-08 10:11:11: Received Zigbee message from '0x001fee0000006cd5', type 'attributeReport', cluster 'seMetering', data '{"instantaneousDemand":8}' from endpoint 3 with groupID 0
Zigbee2MQTT:info  2024-03-08 10:11:11: MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":null,"linkquality":61,"power":8,"power_on_behavior":null,"state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Zigbee2MQTT:debug 2024-03-08 10:11:11: Received Zigbee message from '0x001fee0000006cd5', type 'readResponse', cluster 'seMetering', data '{"currentSummDelivered":[0,3053]}' from endpoint 3 with groupID 0
Zigbee2MQTT:info  2024-03-08 10:11:11: MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":3.05,"linkquality":58,"power":8,"power_on_behavior":null,"state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
```

Turning off

```
Zigbee2MQTT:debug 2024-03-08 10:12:14: Received MQTT message on 'zigbee2mqtt/0x001fee0000006cd5/set' with data '{"state":"OFF"}'
Zigbee2MQTT:debug 2024-03-08 10:12:14: Publishing 'set' 'state' to '0x001fee0000006cd5'
Zigbee2MQTT:info  2024-03-08 10:12:15: MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":3.05,"linkquality":58,"power":8,"power_on_behavior":null,"state":"OFF","update":{"installed_version":-1,"latest_version":-1 [... cut ...]
Stopping Z2M 1.36.0-dev commit: [0e0eccb](https://github.com/Koenkk/zigbee2mqtt/commit/0e0eccb) does not recreate `state_l1` in the cache:

```json
    "0x001fee0000006cd5": {
        "state": "OFF",
        "power": 0,
        "energy": 3.05
    }
```

So, the issue is present with [1.36.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.36.0) commit: [86ed71c](https://github.com/Koenkk/zigbee2mqtt/commit/86ed71c), but not in 1.36.0-dev commit: [0e0eccb](https://github.com/Koenkk/zigbee2mqtt/commit/0e0eccb)

sjorge, 2024-03-08:
https://github.com/Koenkk/zigbee2mqtt/commit/66fb672492c8f049368d5dbda3925500941cd534 probably what fixed it.

I expect it to be fixed in the next release.

Koenkk, 2024-03-08:
@sjorge https://github.com/Koenkk/zigbee2mqtt/commit/66fb672492c8f049368d5dbda3925500941cd534 indeed should have fixed this.

m0wlheld, 2024-03-09:
Okay, thanks. So this issue is resolved with latest-dev / 0e0eccb. I'll wait until 1.36.1 is release on April.

Thanks for the support.

closed 2024-03-09 by m0wlheld