Koenkk/zigbee2mqtt #21692: permit_join switch in Home Assistant does not enable the timer

McGiverGim, 2024-03-06:
### What happened?

Since some time ago, I've observed a new zigbee2mqtt device in Home Assistant. It contains the sensors that until then, I configured using the "package" option.
One of them, is a switch to "permit join", and other a sensor with the time left to permit join (a timer). If I enable the permit join from the zigbee2mqtt frontend, the permit join switch is enable, and the timer contains the countdown. But if I enable the join directly from the Home Assistant switch, then there is no timer.
Is that intended or it must use the timer too? I expected so.

### What did you expect to happen?

When I press the "permit join" switch on Home Assistant, the timer must start the countdown to disable it automatically.

### How to reproduce it (minimal and precise)

Simply press the "permit join" switch and look at the timer to permit join sensor.

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

 20230507

### Adapter

Egony Stick Ebyte

### Setup

Add-on on Home Assistant, Raspberry Pi 4b

### Debug log

I suppose it does not help too much, but here is the log after pressing the permit join switch:
```
debug 2024-03-06 09:21:04: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data 'true'
info  2024-03-06 09:21:04: Zigbee: allowing new devices to join.
info  2024-03-06 09:21:04: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"value":true},"status":"ok"}'
```

Koenkk, 2024-03-06:
Thats expected, you will have to manually turn it off again

McGiverGim, 2024-03-06:
Then the sensor with the countdown is only useful when enabling it from the frontend? This has not too much sense.

closed 2024-10-03 by github-actions[bot]

talondnb, 2024-10-06:
> Then the sensor with the countdown is only useful when enabling it from the frontend? This has not too much sense.

I agree, what is the point of the slider @Koenkk in the [example docs](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#controlling-zigbee2mqtt-via-home-assistant)?

talondnb, 2024-10-06:
@McGiverGim, I've configured my YAML to suit, basically just add these sections to the example doc:

```
# Timer
timer:
  zigbee_permit_join:
    name: Zigbee2MQTT Bridge Permit join timeout

automation:
# Automation to start timer when enable join is turned on
  - id: 'zigbee_join_enabled'
    alias: Zigbee Join Enabled
    trigger:
      platform: state
      entity_id: switch.zigbee2mqtt_bridge_permit_join
      to: "on"
    action:
      service: timer.start
      entity_id: timer.zigbee_permit_join
      data:
        duration: "{{ states('input_number.zigbee2mqtt_join_minutes') | int(0) * 60 }}"

# Automation to stop timer when switch turned off and turn off switch when timer finished
  - id: 'zigbee_join_disabled'
    alias: Zigbee Join Disabled
    trigger:
      - platform: event
        event_type: timer.finished
        event_data:
          entity_id: timer.zigbee_permit_join
      - platform: state
        entity_id: switch.zigbee2mqtt_bridge_permit_join
        to: "off"
    action:
      - service: timer.cancel
        data:
          entity_id: timer.zigbee_permit_join
      - service: switch.turn_off
        entity_id: switch.zigbee2mqtt_bridge_permit_join
```

McGiverGim, 2024-10-06:
> @McGiverGim, I've configured my YAML to suit, basically just add these sections to the example doc:
>
> ```
> # Timer
> timer:
>   zigbee_permit_join:
>     name: Zigbee2MQTT Bridge Permit join timeout [...]

Yes, I've an automation too. That's not the problem.

Koenkk, 2024-10-06:
Will be part of https://github.com/Koenkk/zigbee2mqtt/pull/24245 (https://github.com/Koenkk/zigbee2mqtt/discussions/24198)

Teeseeone, 2025-02-14:
But now its not possible to do permit to join without timer?