Koenkk/zigbee2mqtt #21697: Zigbee network down: Publish 'set' 'state' to 'Kantoor' failed: 'Error: Command 3

avdgaag1984, 2024-03-06:
### What happened?

Last day's i have an issue with the lights to turn on or off, it is not working. When i take a look in the log i get this error: 
Publish 'set' 'state' to 'Kantoor' failed: 'Error: Command 3 genOnOff.on({}) failed (SREQ '--> AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x0000000000000003","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":6,"transid":75,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[17,39,1]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))'

When i restart the zigbee2mqtt addon the problem is solved. But after 1 or 2 hours the problem is back
I run home assistant in an virtual machine on my synology nas. I reboot the NAS, the problem is solved and than after 1 hour it is back. 

Zigbee2mqtt version: 1.36.0-1 
Coordinator version: 20220219
Zigbee stick: CC2652RB

### Zigbee2MQTT version

1.36.0-1 

### Adapter firmware version

20220219

### Adapter

CC2652RB

### Setup

Virtual Machine With Synology NAS

aschmere, 2024-03-06:
Not sure if it is related, but I currently seem to have a similar issue. I have multiple Nous and Innr plugs attached to my Home Assistant Yellow Zigbee2Mqtt installation. Recently I regularly have the issue, that after a while (say a few hours, I would guess 12 hours or so) I cannot control the plugs anymore, e.g. switching on and off. They still appear online in Zigbee2Mqtt and power/energy values are still reported accurately, but I cannot control them anymore. The issue can be resolved by either reboot the HA Yellow or specifically restart the Zigbee2Mqtt integration. This way things are working again for a few hours until they fail again.

avdgaag1984, 2024-03-08:
> Not sure if it is related, but I currently seem to have a similar issue. I have multiple Nous and Innr plugs attached to my Home Assistant Yellow Zigbee2Mqtt installation. Recently I regularly have [...]

I think we have the same problem. Some day's i must restart the zigbee2mqtt addon 4 times. and some day's 1 time. it's getting very annoying now. I am glad that I have many lamps in my house in hue, otherwise I would have had an argument with the wife at home

avdgaag1984, 2024-03-08:
Log from the zigbee2mqtt addon when i want to turn of a lamp: 
Zigbee2MQTT:info  2024-03-08 13:19:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Trapkast' failed: 'Error: Command 0x003c84fffe1640a8/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (SREQ '--> AF - dataRequest - {\"dstaddr\":57241,\"destendpoint\":1,\"srcendpoint\":1,\"clusterid\":6,\"transid\":31,\"options\":0,\"radius\":30,\"len\":3,\"data\":{\"type\":\"Buffer\",\"data\":[1,18,0]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))'","meta":{"friendly_name":"Trapkast"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:debug 2024-03-08 13:19:45: Received MQTT message on 'zigbee2mqtt/Trapkast/set' with data '{"state":"OFF"}'
Zigbee2MQTT:debug 2024-03-08 13:19:45: Publishing 'set' 'state' to 'Trapkast'
Zigbee2MQTT:error 2024-03-08 13:19:45: Publish 'set' 'state' to 'Trapkast' failed: 'Error: Command 0x003c84fffe1640a8/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> AF - dataRequest - {"dstaddr":57241,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":32,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,19,0]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))'
Zigbee2MQTT:debug 2024-03-08 13:19:45: Error: Command 0x003c84fffe1640a8/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> AF - dataRequest - {"dstaddr":57241,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":32,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,19,0]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))
    at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:320:27
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at ZStackAdapter.dataRequest (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:965:9)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:387:35)
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
    at Endpoint.command (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:746:28)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:46:17)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:1119:32)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:259:36)
Zigbee2MQTT:info  2024-03-08 13:19:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Trapkast' failed: 'Error: Command 0x003c84fffe1640a8/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (SREQ '--> AF - dataRequest - {\"dstaddr\":57241,\"destendpoint\":1,\"srcendpoint\":1,\"clusterid\":6,\"transid\":32,\"options\":0,\"radius\":30,\"len\":3,\"data\":{\"type\":\"Buffer\",\"data\":[1,19,0]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))'","meta":{"friendly_name":"Trapkast"},"type":"zigbee_publish_error"}'

Migaddo, 2024-03-12:
Seems very similar as a problem I'm experiencing. Mostly in the morning i need to reboot to be able to control my zigbee devices again.

avdgaag1984, 2024-03-19:
I have fixed it, i have change my zigbee channel to 25. And now is my zigbee network stable for a week. No problems more

closed 2024-10-16 by github-actions[bot]