Koenkk/zigbee2mqtt #21700: [New device support]: Tuya no-spam battery-powered mmwave+pir+lumen sensor

borgqueenx, 2024-03-06:
### Link

https://de.aliexpress.com/item/1005006492967626.html

### Database entry

how??

### Comments

I tried adding the device through the tuya device add docs, but im getting the following data and i dont know how to fill the code up. Im not a coder so im having quite a bad time.
This device could be the holy grail of tuya sensors. since it's battery powered, it's not spamming our networks to near-death. I have already seen the incoming messages are not much of the device, and it's also not a router.

here is the logs it outputs with debug on:
```
Debug 2024-03-06 17:42:24Received Zigbee message from '0xa4c13857d998b93d', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,11,33],"type":"Buffer"},"datatype":2,"dp":106}],"seq":16384}' from endpoint 1 with groupID 0
Debug 2024-03-06 17:42:25Received Zigbee message from '0xa4c13857d998b93d', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":34547}' from endpoint 1 with groupID 0
Debug 2024-03-06 17:42:29Received Zigbee message from '0xa4c13857d998b93d', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[3],"type":"Buffer"},"datatype":4,"dp":101}],"seq":16896}' from endpoint 1 with groupID 0
Debug 2024-03-06 17:42:41Received Zigbee message from '0xa4c13857d998b93d', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":101}],"seq":17152}' from endpoint 1 with groupID 0
Debug 2024-03-06 17:42:42Received Zigbee message from '0xa4c13857d998b93d', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,9,216],"type":"Buffer"},"datatype":2,"dp":106}],"seq":17408}' from endpoint 1 with groupID 0
Debug 2024-03-06 17:42:42Received Zigbee message from '0xa4c13857d998b93d', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":34014}' from endpoint 1 with groupID 0
```
As you can see, only 3 things. 2 DP's, 106 and 101. And the msIlluminanceMeasurement that does not show a DP value.
I have the .js configuration, but dont know how to populate it and test things/make it work.

Help would be appriciated!

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_2aaelwxk',
        },
    ],
    model: 'TS0601_new',
    vendor: 'TuYa',
    description: 'Tuya mmwave+pir+illuminance sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        // Here you should put all functionality that your device exposes
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [106, 'illuminance', tuya.valueConverter.raw], // <- mapped the datapoint
        ],
    },
};

module.exports = definition;
```

ArkadiuszNiemiec, 2024-05-24:
I plan to buy a few of these, but I will wait until it's know if it works with zigbee2mqtt.
Have you tested the sensor? How's battery life looking?

ArkadiuszNiemiec, 2024-06-04:
I bought the sensor and it just arrived. The model is [ZG-204ZM] and is already supported: https://github.com/Koenkk/zigbee2mqtt.io/pull/2650
[ZG-204ZM](https://www.zigbee2mqtt.io/devices/ZG-204ZM.html#tuya-zg-204zm)
@Koenkk I guess you can close this issue. Thanks!

closed 2024-06-05 by Koenkk