Koenkk/zigbee2mqtt #21702: Legrand 067773 and Legrand 067774 - no battery percentage report

cobch, 2024-03-06:
### What happened?

[Legrand 067773 ](https://www.zigbee2mqtt.io/devices/067773.html) and [Legrand 067774 ](https://www.zigbee2mqtt.io/devices/067774.html) devices do not send battery values , but battery is exposed). As can I see in device clusters it should support battery information:

[image]
[image]

I'm using these devices for 6 months, so I waited a lot :)

### What did you expect to happen?

Battery level should be reported

### How to reproduce it (minimal and precise)

Just pair it and observe no battery values

### Zigbee2MQTT version

 1.36.0

### Adapter firmware version

7.1.1.0 build 273

### Adapter

Skyconnect EZSP v9

### Setup

Z2M hassio on a raspberry Pi 4

Loic691, 2024-04-03:
It's working for me for 067773.
Z2m 1.26.1 on sonoff dongle E

cobch, 2024-04-04:
Ok interesting. But I have only a question mark instead of the battery level. could you help me to fix that maybe?
[image]

Loic691, 2024-04-04:
Strange, I think i have the same for double gangs... i try to look about that tomorrow

Loic691, 2024-04-05:
Hey
I have a "?" for Home Away 067646

cobch, 2024-04-05:
Do you know how it is possible you get battery status for you but not for me? I have latest z2m and using skyconnect.

Loic691, 2024-04-05:
I don't know. It's obscur for me... Some users get it, others one no...
Legrand is a problem for users because it's frenco french and minority users use it...
Legrand is not very regardless on zigbee protocol...

Loic691, 2024-04-09:
Hi
I updated my last post. Two device don't get battery state....

Loic691, 2024-04-19:
Hey
Very very strange... Same device, same firmware ( 46 )
One device with battery status, other device no....

[image]

closed 2024-11-16 by github-actions[bot]