Koenkk/zigbee2mqtt #21703: TS0601_smart_air_house_keeper over reporting

Dimmers81, 2024-03-06:
### What happened?

I added a TuYa TS0601_smart_air_house_keeper - it functions as intended, but it appears to be over-reporting multiple times a second. I have lowered the precision of temperature and humidity, and it lessened the amount of reporting times per second. In addition to this, I have also attempted to change the frequency of reporting in the Reporting tab, but it always fails to update

### What did you expect to happen?

To report less frequent

### How to reproduce it (minimal and precise)

I turned it on and added it to Zigbee2MQTT

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

7.3.1.0 build 176

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus V2

### Setup

Intel NUC

### Debug log

 
Info 2024-03-07 10:38:43MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":360,"formaldehyd":3,"humidity":61.8,"linkquality":255,"pm25":6,"temperature":27.8,"voc":20}'
Info 2024-03-07 10:39:00MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":360,"formaldehyd":3,"humidity":61.8,"linkquality":244,"pm25":6,"temperature":27.8,"voc":20}'
Info 2024-03-07 10:39:00MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":360,"formaldehyd":3,"humidity":61.8,"linkquality":172,"pm25":6,"temperature":27.8,"voc":20}'
Info 2024-03-07 10:39:00MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":360,"formaldehyd":3,"humidity":61.8,"linkquality":172,"pm25":6,"temperature":27.8,"voc":20}'
Info 2024-03-07 10:39:01MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":360,"formaldehyd":3,"humidity":61.8,"linkquality":108,"pm25":6,"temperature":27.8,"voc":20}'
Info 2024-03-07 10:39:01MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":360,"formaldehyd":3,"humidity":61.8,"linkquality":100,"pm25":6,"temperature":27.8,"voc":20}'
Info 2024-03-07 10:39:01MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":360,"formaldehyd":3,"humidity":61.8,"linkquality":184,"pm25":6,"temperature":27.8,"voc":20}'
Info 2024-03-07 10:39:01MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":360,"formaldehyd":3,"humidity":61.8,"linkquality":184,"pm25":6,"temperature":27.8,"voc":20}'
Info 2024-03-07 10:39:02MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":360,"formaldehyd":3,"humidity":61.8,"linkquality":180,"pm25":6,"temperature":27.8,"voc":20}'
Info 2024-03-07 10:39:02MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":360,"formaldehyd":3,"humidity":61.8,"linkquality":144,"pm25":6,"temperature":27.8,"voc":20}'
Info 2024-03-07 10:39:02MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":360,"formaldehyd":3,"humidity":61.8,"linkquality":144,"pm25":6,"temperature":27.8,"voc":20}'
Info 2024-03-07 10:39:02MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":360,"formaldehyd":3,"humidity":61.8,"linkquality":136,"pm25":6,"temperature":27.8,"voc":20}'
Info 2024-03-07 10:39:06MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":360,"formaldehyd":2,"humidity":61.8,"linkquality":255,"pm25":6,"temperature":27.8,"voc":20}'
Info 2024-03-07 10:39:06MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":360,"formaldehyd":2,"humidity":61.8,"linkquality":240,"pm25":6,"temperature":27.8,"voc":20}'
Info 2024-03-07 10:39:06MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":360,"formaldehyd":2,"humidity":61.8,"linkquality":255,"pm25":6,"temperature":27.8,"voc":20}'
Info 2024-03-07 10:39:06MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":360,"formaldehyd":2,"humidity":61.8,"linkquality":252,"pm25":6,"temperature":27.8,"voc":20}'
Info 2024-03-07 10:39:07MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":360,"formaldehyd":2,"humidity":61.8,"linkquality":255,"pm25":6,"temperature":27.8,"voc":20}'
Info 2024-03-07 10:39:07MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":360,"formaldehyd":2,"humidity":61.8,"linkquality":244,"pm25":6,"temperature":27.8,"voc":20}'
Info 2024-03-07 10:39:07MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":360,"formaldehyd":2,"humidity":61.8,"linkquality":244,"pm25":6,"temperature":27.8,"voc":20}'
Info 2024-03-07 10:39:07MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":360,"formaldehyd":2,"humidity":61.8,"linkquality":244,"pm25":6,"temperature":27.8,"voc":20}'
Info 2024-03-07 10:39:08MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":360,"formaldehyd":2,"humidity":61.8,"linkquality":240,"pm25":6,"temperature":27.8,"voc":20}'
Info 2024-03-07 10:39:08MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":360,"formaldehyd":2,"humidity":61.8,"linkquality":240,"pm25":6,"temperature":27.8,"voc":20}'
Info 2024-03-07 10:39:08MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":360,"formaldehyd":2,"humidity":61.8,"linkquality":240,"pm25":6,"temperature":27.8,"voc":20}'
Info 2024-03-07 10:39:08MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":360,"formaldehyd":2,"humidity":61.8,"linkquality":244,"pm25":6,"temperature":27.8,"voc":20}'
Info 2024-03-07 10:39:09MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":360,"formaldehyd":2,"humidity":61.8,"linkquality":244,"pm25":6,"temperature":27.8,"voc":11}'
Info 2024-03-07 10:39:09MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":359,"formaldehyd":2,"humidity":61.8,"linkquality":255,"pm25":6,"temperature":27.8,"voc":11}'
Info 2024-03-07 10:39:10MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":359,"formaldehyd":2,"humidity":61.8,"linkquality":255,"pm25":5,"temperature":27.8,"voc":11}'
Info 2024-03-07 10:39:10MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":359,"formaldehyd":2,"humidity":61.8,"linkquality":244,"pm25":5,"temperature":27.6,"voc":11}'
Info 2024-03-07 10:39:10MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":359,"formaldehyd":2,"humidity":61.8,"linkquality":255,"pm25":5,"temperature":27.6,"voc":11}'
Info 2024-03-07 10:39:12MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":359,"formaldehyd":2,"humidity":61.8,"linkquality":240,"pm25":5,"temperature":27.5,"voc":11}'
Info 2024-03-07 10:39:12MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":359,"formaldehyd":2,"humidity":61.8,"linkquality":252,"pm25":5,"temperature":27.5,"voc":11}'
Info 2024-03-07 10:39:13MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":359,"formaldehyd":2,"humidity":61.8,"linkquality":244,"pm25":5,"temperature":27.5,"voc":11}'
Info 2024-03-07 10:39:13MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":359,"formaldehyd":2,"humidity":61.8,"linkquality":248,"pm25":5,"temperature":27.5,"voc":11}'
Info 2024-03-07 10:39:13MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":359,"formaldehyd":2,"humidity":61.8,"linkquality":244,"pm25":5,"temperature":27.5,"voc":11}'
Info 2024-03-07 10:39:13MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":359,"formaldehyd":2,"humidity":61.8,"linkquality":240,"pm25":5,"temperature":27.5,"voc":11}'
Info 2024-03-07 10:39:13MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":359,"formaldehyd":2,"humidity":61.8,"linkquality":240,"pm25":5,"temperature":27.5,"voc":11}'
Info 2024-03-07 10:39:14MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":359,"formaldehyd":2,"humidity":61.8,"linkquality":240,"pm25":5,"temperature":27.5,"voc":11}'
Info 2024-03-07 10:39:14MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":359,"formaldehyd":2,"humidity":61.8,"linkquality":244,"pm25":5,"temperature":27.5,"voc":11}'
Info 2024-03-07 10:39:14MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":359,"formaldehyd":2,"humidity":61.8,"linkquality":244,"pm25":5,"temperature":27.5,"voc":11}'
[... 9 lines cut ...]
Info 2024-03-07 10:39:18MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":362,"formaldehyd":3,"humidity":61.8,"linkquality":236,"pm25":5,"temperature":27.5,"voc":17}'
[... cut ...]
Info 2024-03-07 10:39:22MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":362,"formaldehyd":2,"humidity":61.5,"linkquality":252,"pm25":5,"temperature":27.5,"voc":17}'
Info 2024-03-07 10:39:23MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":362,"formaldehyd":2,"humidity":61.5,"linkquality":244,"pm25":5,"temperature":27.5,"voc":21}'
Info 2024-03-07 10:39:23MQTT publish: topic 'zigbee2mqtt/Ensuite Particle Sensor', payload '{"co2":359,"formaldehyd":2,"humidity":61.5,"linkquality":244,"pm25":5,"temperature":27.5,"voc":21}'

phenix1990, 2024-08-01:
I have the same problem, this device is really spamming. Is it possible to build an average of the last x values before publishing to the network?

I tried to change the minimum reporting value but the values are jumping much. Not sure if this happens because I simply pass a lot of values or if the sensor is inaccurate or too sensitive.
[image]

Richardvdw01, 2024-12-02:
You may try the following:
In the settings, set debounce to 5 or 10.
When setting e.g. debounce: 5 and a message from a device is received, Zigbee2MQTT will not immediately publish this message. But it will combine it with other messages that are received max 5 seconds apart from each other. So there needs to be "5 second of silence" from the device before the combined message is sent out.

closed 2025-02-09 by github-actions[bot]