Koenkk/zigbee2mqtt #21704: [New device support]: Frient Air Quality Sensor

NickoPoups, 2024-03-07:
### Link

https://frient.com/products/air-quality-sensor/

### Database entry

0x0015bc00360028ba

### Comments

Hello,

Could you please add this device to zigbee2mqtt please?

Thank you :)

### External definition

```shell
const definition = {
    zigbeeModel: [''],
    model: '',
    vendor: '',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

NickoPoups, 2024-03-07:
error 2024-03-07 08:47:45 Failed to interview '0x0015bc00360028ba', device has not successfully been paired

info 2024-03-07 08:47:45 MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc00360028ba","ieee_address":"0x0015bc00360028ba","status":"failed"},"type":"device_interview"}'

sjorge, 2024-03-07:
These should already be supported. I'm using a few of those.

But looks like it can't figure out what device it is because the interview doesn't complete. Are there any errors before that?

NickoPoups, 2024-03-08:
I tried pairing the device on another hub (Lixee ZLinky), it works fine...
Sorry for the trouble.
My SONOFF ZigBee 3.0 USB Dongle Plus (EFR32MG21) might have a problem.
Closing my request.
Thank you guys :)

closed 2024-10-05 by github-actions[bot]