Koenkk/zigbee2mqtt #21707: HA commands being ignored.

corporategoth, 2024-03-07:
### What happened?

Z2M is ignoring commands from Home Assistant

It is RECEIVING the commands (as evidenced by logs shown below), but no action happens on the associated devices or groups.

### What did you expect to happen?

The commands would change the state of the associated devices / groups.

### How to reproduce it (minimal and precise)

Try to perform any command (eg. turn device on or off).

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

7.4.1.0

### Adapter

Sonoff Dongle-E (ember)

### Setup

Z2M, Mosquitto & HA are all in their own docker containers.

### Debug log

Command for a device:
```
debug 2024-03-07 07:33:52: Received MQTT message on 'zigbee2mqtt/Post 1 Light - Small/set' with data '{"state":"OFF"}'
debug 2024-03-07 07:33:52: Publishing 'set' 'state' to 'Post 1 Light - Small'
```

Command for a group:
```
debug 2024-03-07 07:36:26: Received MQTT message on 'zigbee2mqtt/driveway_lights/set' with data '{"state":"OFF"}'
debug 2024-03-07 07:36:26: Publishing 'set' 'state' to 'driveway_lights'
```

Neither did anything.

Koenkk, 2024-03-08:
Could this be due to the ember driver, can you post the herdsman debug logs of this?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

corporategoth, 2024-03-10:
Strangely, I enabled herdsman debug and things are working.  Not tried disabling again to see if that did something.

Nerivec, 2024-05-15:
Is this still a problem?

closed 2024-11-15 by corporategoth