Koenkk/zigbee2mqtt #21708: OTA: Images currently unavailable for device 'TRADFRI bulb E27 CWS 806lm'

hboltz, 2024-03-07:
### What happened?

I tried to update the firmware of 3 Ikea bulbs and got the following error message: OTA: Images currently unavailable for device 'TRADFRI bulb E27 CWS 806lm', hardwareVersion '1', manufacturerName IKEA of Sweden, {"fieldControl":1,"manufacturerCode":4476,"imageType":10242,"fileVersion":268572245}'

### What did you expect to happen?

Firmware Update

### How to reproduce it (minimal and precise)

klick on "check for firmware updates" in the OTA section

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

0x26580700

### Adapter

ConBee2

### Setup

Add-On on Homeassistant in a Proxmox VM 

### Debug log

[image]

enjoylifenow, 2024-04-24:
Same problem for Ikea LED BULB GU10 (LED2110R3) identified as [LED1923R5/LED1925G6]

OTA: Images currently unavailable for device 'TRADFRI bulb GU10 CWS 345lm', hardwareVersion '1', manufacturerName IKEA of Sweden, {"fieldControl":1,"manufacturerCode":4476,"imageType":10245,"fileVersion":16777272}'

M1K3SD1, 2024-06-14:
Same here, just tried with both the regular server and the OTA test for IKEA, not working.

`Images currently unavailable for device 'PARASOLL Door/Window Sensor', hardwareVersion '1', manufacturerName IKEA of Sweden, {"fieldControl":1,"manufacturerCode":4476,"imageType":12919,"fileVersion":16777241}'`

M1K3SD1, 2024-12-30:
This is still happening

closed 2025-03-09 by github-actions[bot]