Koenkk/zigbee2mqtt #21716: IMOU ZigBee Siren

array81, 2024-03-07:
### Link

https://www.imoulife.com/en/product/specs/imou-home-security-system

### Database entry

{"id":23,"type":"EndDevice","ieeeAddr":"0x8cf681fffecfd98a","nwkAddr":37531,"manufId":21845,"manufName":"MultIR","powerSource":"DC Source","modelId":"ZR1-EN","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1027,"inClusterList":[0,3,6,1280,1282,2821],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"ZR1-EN","manufacturerName":"MultIR","powerSource":4,"zclVersion":3,"appVersion":4,"hwVersion":1,"dateCode":"20220422","swBuildId":"v0.3.1"}},"ssIasZone":{"attributes":{"iasCieAddr":"0xe0798dfffed1f24b","zoneState":1}},"genOnOff":{"attributes":{"onOff":0}},"haDiagnostic":{"attributes":{"lastMessageRssi":-53,"lastMessageLqi":188}},"ssIasWd":{"attributes":{"maxDuration":1800}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffed1f24b","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffed1f24b","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}}},"appVersion":4,"hwVersion":1,"dateCode":"20220422","swBuildId":"v0.3.1","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1709823215714}

### Comments

I'm trying to add support for an IMOU Siren
This is the Tuya sensor which looks like the IMOU one: https://www.zigbee2mqtt.io/devices/TS0216.html
The siren looks just the same except for the logo and fewer speaker holes. This also has a red light along the edge.
I tried using the TS0216 configuration (already supported) but without success. Using the SA100 configuration it is possible to emit sounds (mute, low, medium and high). I also tried using ias_siren but the only thing that worked was the tamper.
Can anyone give me a hand?

### External definition

```shell
const {onOff, identify} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['ZR1-EN'],
    model: 'ZR1-EN',
    vendor: 'MultIR',
    description: 'Automatically generated definition',
    extend: [onOff({"powerOnBehavior":false}), identify()],
    meta: {},
};
```

array81, 2024-03-08:
Some small progress, this code allows the "warning" seems to work but I can't understand how to turn the siren on or off. None of the "warning" properties are posted to Home Assistant (I don't know if this is normal).

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const legacy= require('zigbee-herdsman-converters/lib/legacy');
const tuya= require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['ZR1-EN'],
    model: 'ZR1-EN',
    vendor: 'IMOU',
    description: 'IMOU ZigBee Siren',
        fromZigbee: [fz.battery, fz.ignore_basic_report],
        toZigbee: [tz.warning],
        exposes: [e.battery(), e.warning()],
        meta: {disableDefaultResponse: true},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await reporting.batteryPercentageRemaining(endpoint);
            // Device advertises itself as Router but is an EndDevice
            device.type = 'EndDevice';
            device.save();
        }
};

module.exports = definition;
```
The battery doesn't seem to work but I'm not sure if the siren has a battery although it says in the specs.
Here's some log:

```
Debug 2024-03-08 08:32:50Received MQTT message on 'zigbee2mqtt/0x8cf681fffecfd98a/set' with data '{"warning":{"level":"high","mode":"burglar"}}'
Debug 2024-03-08 08:32:50Publishing 'set' 'warning' to '0x8cf681fffecfd98a'
Debug 2024-03-08 08:32:50Received Zigbee message from '0x8cf681fffecfd98a', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Debug 2024-03-08 08:32:50No converter available for 'ZR1-EN' with cluster 'genOnOff' and type 'attributeReport' and data '{"onOff":1}'
Info 2024-03-08 08:32:50MQTT publish: topic 'zigbee2mqtt/0x8cf681fffecfd98a', payload '{"ac_status":false,"alarm":false,"battery":null,"battery_low":false,"device":{"applicationVersion":4,"dateCode":"20220422","friendlyName":"0x8cf681fffecfd98a","hardwareVersion":1,"ieeeAddr":"0x8cf681fffecfd98a","manufacturerID":21845,"manufacturerName":"MultIR","model":"ZR1-EN","networkAddress":37531,"powerSource":"DC Source","softwareBuildID":"v0.3.1","type":"EndDevice","zclVersion":3},"last_seen":"2024-03-08T08:32:50+01:00","linkquality":232,"restore_reports":true,"state":"OFF","supervision_reports":false,"tamper":true,"test":false,"type":"normal","volume":"high"}'
Debug 2024-03-08 08:32:56Received MQTT message on 'zigbee2mqtt/0x8cf681fffecfd98a/set' with data '{"warning":{"level":"high","mode":"stop"}}'
Debug 2024-03-08 08:32:56Publishing 'set' 'warning' to '0x8cf681fffecfd98a'
Debug 2024-03-08 08:32:56Received Zigbee message from '0x8cf681fffecfd98a', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Debug 2024-03-08 08:32:56No converter available for 'ZR1-EN' with cluster 'genOnOff' and type 'attributeReport' and data '{"onOff":0}'
Info 2024-03-08 08:32:56MQTT publish: topic 'zigbee2mqtt/0x8cf681fffecfd98a', payload '{"ac_status":false,"alarm":false,"battery":null,"battery_low":false,"device":{"applicationVersion":4,"dateCode":"20220422","friendlyName":"0x8cf681fffecfd98a","hardwareVersion":1,"ieeeAddr":"0x8cf681fffecfd98a","manufacturerID":21845,"manufacturerName":"MultIR","model":"ZR1-EN","networkAddress":37531,"powerSource":"DC Source","softwareBuildID":"v0.3.1","type":"EndDevice","zclVersion":3},"last_seen":"2024-03-08T08:32:56+01:00","linkquality":224,"restore_reports":true,"state":"OFF","supervision_reports":false,"tamper":true,"test":false,"type":"normal","volume":"high"}'
```

array81, 2024-03-08:
I think this is the best you can get from this siren:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const legacy= require('zigbee-herdsman-converters/lib/legacy');
const tuya= require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['ZR1-EN'],
    model: 'ZR1-EN',
    vendor: 'IMOU',
    description: 'IMOU ZigBee Siren',
        fromZigbee: [fz.ignore_basic_report, fz.ias_siren],
        toZigbee: [tz.warning],
        exposes: [e.tamper(), e.warning()],
        meta: {disableDefaultResponse: true},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await reporting.batteryPercentageRemaining(endpoint);
            // Device advertises itself as Router but is an EndDevice
            device.type = 'EndDevice';
            device.save();
        }
};

module.exports = definition;
```

Both "warning" and "tamper" work.
To turn the siren on or off you need to publish via MQTT. Below is the code to create a switch on Home Assistant to turn the siren on and off.

```
  switch:
    - name: "Siren Switch"
      unique_id: sirena_switch
      state_topic: "zigbee2mqtt/Siren"
      command_topic: "zigbee2mqtt/Siren/set"
      #availability:
      #  - topic: "zigbee2mqtt/Siren/availability"
      #    payload_available: "online"
      #    payload_not_available: "offline"
      payload_on: '{"warning": {"duration":60,"level":"very_high","mode":"emergency","strobe":true,"strobe_duty_cycle":5,"strobe_level":"very_high"}}'
      payload_off: '{"warning": {"duration":1,"level":"low","mode":"stop","strobe":false,"strobe_level":"low"}}'
      state_on: "ON"
      state_off: "OFF"
      icon: mdi:alarm-light
      qos: 0
      optimistic: true
      retain: true
```

I was inspired by the HS2WD-E siren whose behavior seems identical to that of this siren. Reference to the HS2WD-E siren support page.

array81, 2024-04-07:
The official code released for this device has issues. If you try to send an MQTT command via Home Assistant (as per the code in the message above this) you get the following error:

`Publish 'set' 'warning' to 'Internal Siren' failed: 'ReferenceError: utils is not defined'`

While my code worked. Furthermore, the device has no battery so it is useless to display the battery values.

mrskycriper, 2024-04-11:
What version of Zigbee2MQTT are you using? What does and doesn't work with the current release?

array81, 2024-04-11:
I currently use version 1.36.1 (the latest).
With version 1.36.0 I had created an extension to make this siren work.
I made a pull request with my extension to be included in version 1.36.1 however the code I find in version 1.36.1 is different from mine. If I now try to turn on the siren via MQTT (see the code in the posts above) I get an error (I see it on Zigbee2Mqtt):

`Publish 'set' 'warning' to 'Internal Siren' failed: 'ReferenceError: utils is not defined'`

While if I turn it on via Zigbee2Mqtt it turns on correctly. Using my code the siren also turned on via MQTT.

mrskycriper, 2024-04-11:
Are you sure that you have disabled the external definition? The code in release version should not have this kind of error.

array81, 2024-04-11:
I removed the files with my definitions and removed them from the zigbee2mqtt settings.

Frizzbeat, 2024-04-24:
@array81 I figured out that when you delete everythin related to the strobe the mqqt switch from you code above is working fine.

These are the terms which i deleted:
"strobe":true,"strobe_duty_cycle":5,"strobe_level":"very_high"
"strobe":false,"strobe_level":"low"

closed 2024-11-22 by github-actions[bot]