Koenkk/zigbee2mqtt #21724: Failed to apply calibration to 'power': 'power_calibration' is not a number, got string ()

GerdRuetten, 2024-03-08:
### What happened?

The z2m log is flloded with the message:

`Failed to apply calibration to 'power': 'power_calibration' is not a number, got string ()`

I cannot find the device that is causing the error

### What did you expect to happen?

Not this error

### How to reproduce it (minimal and precise)

Good question

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

7.3.1.0 build 176

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus,TI CC2652P + CP2102(N)

### Setup

Plain on Docker

### Debug log

[_zigbee2mqtt_logs.txt](https://github.com/Koenkk/zigbee2mqtt/files/14541483/_zigbee2mqtt_logs.txt)

dCatfish, 2024-03-13:
I got rid of these errors by adding a zero in the blank calibration fields in "Settings (specific)" for all my devices.

[image]

GerdRuetten, 2024-03-27:
Hi @dCatfish 

sorry i'm bussy

> adding a zero in the blank calibration fields 

Yes, thats solved the problem but is this the solution?

dCatfish, 2024-03-28:
Hi @GerdRuetten 

It's just a workaround. But the issue shouldn't be hard to permanently solve for Koenkk (or some other HA developer). Probably solved in the next Z2M update. 

Hope you're feeling well and having a happy Easter :)

sebastianlorenzen, 2024-07-19:
I have the same issue with 1.39.0.
will this be fixed?

simontims, 2024-09-23:
I have this issue too, in 1.40.1

blooesky, 2024-09-24:
same here 1.40.1 commit: 403d3c09

daubman, 2024-11-12:
The same thing is still happening with other calibration numbers before they are set - I added two common sensors recently and get floods of these in my logs now - I will try the workaround mentioned (setting all to 0) but would be great to handle correctly with defaults if possible
```
error 2024-11-12 07:40:11zhc: Failed to apply calibration to 'humidity': 'humidity_precision' is not a number, got string ()
error 2024-11-12 07:40:11zhc: Failed to apply calibration to 'pressure': 'pressure_calibration' is not a number, got string ()
error 2024-11-12 07:40:31zhc: Failed to apply calibration to 'temperature': 'temperature_calibration' is not a number, got string ()
```

ortofan, 2024-11-19:
Same issue here with everything the latest, but in my case it often crashes the adapter. 

[2024-11-19 05:31:58] error: 	zhc: Failed to apply calibration to 'current': 'current_calibration' is not a number, got string ()
[2024-11-19 05:36:23] error: 	zhc: Failed to apply calibration to 'current': 'current_calibration' is not a number, got string ()
[2024-11-19 05:37:00] error: 	zhc: Failed to apply calibration to 'energy': 'energy_calibration' is not a number, got string ()
[2024-11-19 05:37:10] error: 	zhc: Failed to apply calibration to 'current': 'current_calibration' is not a number, got string ()
[2024-11-19 05:42:13] error: 	zhc: Failed to apply calibration to 'energy': 'energy_calibration' is not a number, got string ()
[2024-11-19 05:42:24] error: 	zhc: Failed to apply calibration to 'current': 'current_calibration' is not a number, got string ()
[2024-11-19 05:42:57] error: 	zh:ember:uart:ash: Port Error: read ECONNRESET
[2024-11-19 05:42:57] error: 	zh:ember: Adapter fatal error: ERROR_SERIAL_INIT
[2024-11-19 05:42:57] error: 	z2m: Adapter disconnected, stopping
[05:42:58] INFO: Preparing to start...
[05:42:58] INFO: Socat not enabled
[05:42:58] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-11-19 05:43:05] error: 	zh:ember:uart:ash: Received ERROR from adapter while connecting, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-11-19 05:47:25] error: 	zhc: Failed to apply calibration to 'energy': 'energy_calibration' is not a number, got string ()
[2024-11-19 05:47:36] error: 	zhc: Failed to apply calibration to 'current': 'current_calibration' is not a number, got string ()
[2024-11-19 05:52:54] error: 	zhc: Failed to apply calibration to 'energy': 'energy_calibration' is not a number, got string ()

mhoogenbosch, 2024-12-26:
Out of the blue, this started happening... Specific field have always been empty. Now errors all over... only thing changed is that i have added a couple of new plugs.

leonidostrovski, 2025-01-11:
For me, it looks solved by removing row:
energy_calibration: " "
from file "devices.yaml"
Source: ChatGpt

simontims, 2025-01-12:
I've just noticed the error has a different prefix to other messages. I only tend to see the error every few minutes.

The error message is prefixed by zhc:
All other normal log entries start with z2m:mqtt

sebastianlorenzen, 2025-01-16:
> For me, it looks solved by removing row: energy_calibration: " " from file "devices.yaml" Source: ChatGpt

It looks like that the devices.yaml does not exist by default.

Where did you found it?

rekcodocker, 2025-02-22:
> > For me, it looks solved by removing row: energy_calibration: " " from file "devices.yaml" Source: ChatGpt
> 
> It looks like that the devices.yaml does not exist by default.

It is not devices.yaml but the configuration yaml from Zigbee2Mqtt, in the config directory of Zigbee2Mqtt.

superpower10, 2025-04-18:
Have the same error.

All devices are set to “power_calibration: 0”. Did not fix the problem.

How do I find the device that causes the error?

michaelsidenius, 2025-04-23:
As soon as you start settings values in "Settings (specific)" this happens. 
And changing the values in the web interface is not enough. You need to set the values in the devices.yaml, and change any values shown as: '' (two single quotes with no values in between) to 0.0 (no quotes).

I guess this is a web front end problem.

Zigbee2MQTT version [2.2.1]

rroblik, 2025-04-23:
@michaelsidenius `devices.yaml` file not exist on my side (v1.42) how to do ?

michaelsidenius, 2025-04-24:
You can use the "File editor"  Add-on (under "Settings" "Add-ons") The files are placed under the folder "zigbee2mqtt".

oxivanisher, 2025-06-13:
I can not find any matching entry in any config:
```bash
# grep -r _calibration /somepath/zigbee2mqtt/*.yaml
#
```

I am unable to re-pair the [IKEA E1525/E1745](https://www.zigbee2mqtt.io/devices/E1525_E1745.html#ikea-e1525-e1745) which worked for a long time without problems. I changed the batteries and tried all the tips on the device page.

The error I get looks like this:
```plaintext
[2025-06-13 22:38:26] error:    z2m: Failed to configure '0xec1bbdfffe7147fc', attempt 1 (Error: Bind 0xec1bbdfffe7147fc/1 genPowerCfg from '0x00124b0021cc012d/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Object.start (/app/node_modules/.pnpm/zigbee-herdsman@4.1.0/node_modules/zigbee-herdsman/src/utils/waitress.ts:67:23)
    at func (/app/node_modules/.pnpm/zigbee-herdsman@4.1.0/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:419:47)
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@4.1.0/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.sendZdoInternal (/app/node_modules/.pnpm/zigbee-herdsman@4.1.0/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:424:43)
    at ZStackAdapter.sendZdo (/app/node_modules/.pnpm/zigbee-herdsman@4.1.0/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:301:16)
    at Endpoint.bind (/app/node_modules/.pnpm/zigbee-herdsman@4.1.0/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:580:30)
    at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.53.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:159:13)
    at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.53.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:222:17)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.53.0/node_modules/zigbee-herdsman-converters/src/index.ts:363:21)
    at Configure.configure (/app/lib/extension/configure.ts:124:13))
```
And I also tried switching back to the old WebUI without any success. Any ideas?

Fleurelle13, 2025-07-01:
Bekomme die selber Fehlermeldung:
error 2025-07-01 20:13:24zhc: Failed to apply calibration to 'energy': 'energy_calibration' is not a number, got string ()

closed 2025-07-29 by Koenkk