Koenkk/zigbee2mqtt #21725: [New device support]: Aqara - Ceiling Light T1M - Model CL-L02D

coniferous1, 2024-03-08:
### Link

https://www.aqara.com/en/product/ceiling-light-t1m/

### Database entry

{"id":24,"type":"Router","ieeeAddr":"0x54ef441000b12703","nwkAddr":60597,"manufId":4447,"manufName":"Aqara","powerSource":"DC Source","modelId":"lumi.light.acn032","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[5,4,3,0,768,8,6,64704],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"61440":2,"modelId":"lumi.light.acn032","manufacturerName":"Aqara","powerSource":4,"zclVersion":3,"appVersion":26,"stackVersion":27,"hwVersion":1,"dateCode":"20231226"}},"lightingColorCtrl":{"attributes":{"245":15513062,"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370,"colorTemperature":153}},"manuSpecificLumi":{"attributes":{"223":{"type":"Buffer","data":[22,33,2,0,23,33,0,0,12,33,5,0,24,33,218,0]},"229":{"type":"Buffer","data":[15,0,15,0,0,4,215,198,36,37,175,91,49,37,223,40,56,37,145,129,65,37,184,180,72,37,85,123,93,37,181,242,98,37,95,175,128,37,179,195,173,37,136,71,191,37,147,162,199,37,92,176,216,37,116,33,239,37,104,55,255,37,32]},"247":{"type":"Buffer","data":[117,32,254,115,35,153,153,204,76]},"65522":{"type":"Buffer","data":[170,117,19,132,78,252,7,65,0,15,12,35,1,6,0,15,10,35,8,0,12,100,11,32,2]}}},"genLevelCtrl":{"attributes":{"245":15512839,"currentLevel":191,"onLevel":254}},"genOnOff":{"attributes":{"245":15512865,"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":258,"inClusterList":[5,4,3,768,8,6,64704],"outClusterList":[],"clusters":{"lightingColorCtrl":{"attributes":{"245":49067264,"colorCapabilities":24,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370,"currentX":19660,"currentY":39321,"colorMode":1,"options":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":26,"stackVersion":27,"hwVersion":1,"dateCode":"20231226","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1709924741206}

### Comments

The main light works right out of the box, temperature and everything. z2m recoginizes that there are two lights and this is a multipoint device, but all commands sent to the second light seem to be sent to the first one instead in the logging. 

Example where I control the main light:

Info 2024-03-08 14:20:07MQTT publish: topic 'zigbee2mqtt/Office_Overhead', payload '{"brightness_1":254,"color":{"x":0.6942,"y":0.2963},"color_1":{"x":0.4066,"y":0.1643},"color_mode":"xy","color_mode_1":"xy","color_options":{"execute_if_off":false},"color_temp":158,"color_temp_1":153,"level_config":{"on_level":254},"linkquality":255,"power_on_behavior":null,"state":null,"state_1":"ON"}'

If I attempt to change the colour of the LED strip in z2m I get this: 

Info 2024-03-08 14:20:47MQTT publish: topic 'zigbee2mqtt/Office_Overhead', payload '{"brightness_1":254,"color":{"x":0.6942,"y":0.2963},"color_1":{"x":0.5524,"y":0.4079},"color_mode":"xy","color_mode_1":"xy","color_options":{"execute_if_off":false},"color_temp":158,"color_temp_1":370,"level_config":{"on_level":254},"linkquality":255,"power_on_behavior":null,"state":null,"state_1":"ON"}'

It's also worth noting that if I try to turn on or off the LED strip the main light turns on or off instead. 

It almost feels like in the external definition I need to define what light belongs to what endpoint, but when looking at modernExtend i can see that under line 252 there is a "endpointNames" argument under "LightArgs", yet when I add it to the external definition it just gives me a syntax error. 

example of what I was trying to do: 

extend: [deviceEndpoints({"endpoints":{"1":1,"2":2}}), light({"colorTemp":{"range":[153,370]}, "endpointNames":"1"}), light({"colorTemp":{"range":[153,370]},"color":true, "endpointNames":"2"})],

I'm likely on the wrong path. Any advice or general guidelines on dealing with multiendpoint devices?

### External definition

```shell
const {deviceEndpoints, light} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {} = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.light.acn032'],
    model: 'CL-L02D',
    vendor: 'Aqara',
    description: 'Ceiling Light T1M',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2}}), light({"colorTemp":{"range":[153,370]}}), light({"colorTemp":{"range":[153,370]},"color":true})],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

coniferous1, 2024-03-08:
And I was absolutely doing something wrong with the syntax. It was expecting an array of strings *headdesk*. This line now allows me to control the brightness and warmness of the LED strip, but I still cannot control colour. 

extend: [deviceEndpoints({"endpoints":{"1":1,"2":2}}), light({"colorTemp":{"range":[153,370]}, "endpointNames":["1"]}), light({"colorTemp":{"range":[153,370]},"color":true, "endpointNames":["2"]})],

decry76, 2024-03-08:
Hello 
I found this post 
https://github.com/Koenkk/zigbee2mqtt/issues/21626

This works fine with t1m, except a bug withthe RGB color light.

hope this can help you

AndreCeridian, 2024-03-08:
> Hello I found this post #21626
> 

I followed the example provided in this post and can now control the T1M. Z2MQTT does not allow control of the rgb ring, but I can control that through HA. Through HA I have control over power, brightness, and color of both lights.

coniferous1, 2024-03-08:
thanks @decry76 and @AndreCeridian! that thread is better. I'm going to close this one.

closed 2024-03-08 by coniferous1