Koenkk/zigbee2mqtt #21731: aqara DJT12LM (lumi.vibration.agl01) Vibration sensor T1 - exposes null

eject-dream, 2024-03-09:
### What happened?

Based on the official documentation and information gathered through online searches, this device seems to only be capable of monitoring "vibrations" and being struck three times. Additionally, it can be set to three levels of sensitivity. I have tested functions such as falls and changes in angle, which are originally supported by DJT11LM (this issue pertains to DJT12LM), but there is no additional information in the debug logs.

This issue has been persistent and not recent. I have currently upgraded to version 1.36.0.

In the Exposes tab, there is only Action and its value is always null.
[image]

In the About tab, the battery also cannot be read correctly
[image]

### How to reproduce it (minimal and precise)

The device is added to z2m, and no matter what you do, there will be no more information

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

20221226

### Adapter

cc2652p

### Setup

x86 docker compose 

### Debug log

Press button
```log
Debug 2024-03-09 20:06:07Received Zigbee message from 'zdtest', type 'attributeReport', cluster 'manuSpecificLumi', data '{"280":1}' from endpoint 2 with groupID 0
Debug 2024-03-09 20:06:08Received Zigbee message from 'zdtest', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.vibration.agl01"}' from endpoint 1 with groupID 0
```
shake
```log
Debug 2024-03-09 20:09:13Received Zigbee message from 'zdtest', type 'attributeReport', cluster 'manuSpecificLumi', data '{"280":1}' from endpoint 2 with groupID 0
```
Strike three times
```log
Debug 2024-03-09 20:10:04Received Zigbee message from 'zdtest', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 2 with groupID 0
```

mrskycriper, 2024-03-10:
How does it report battery? Does it use genPowerCfg?

mrskycriper, 2024-03-10:
Can you also please provide entry from the `zigbee2mqtt/datebase.db` on this device?

eject-dream, 2024-03-11:
@mrskycriper Sorry for the late response, do you need any more information from me?
> How does it report battery? Does it use genPowerCfg?

[image]

> Can you also please provide entry from the `zigbee2mqtt/datebase.db` on this device?

{"id":17,"type":"EndDevice","ieeeAddr":"0x04cf8cdf3c89c90a","nwkAddr":11025,"manufId":4447,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.vibration.agl01","epList":[1,2,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.vibration.agl01","appVersion":25,"manufacturerName":"LUMI","powerSource":3,"zclVersion":3,"stackVersion":2,"hwVersion":1,"dateCode":"20200528","swBuildId":"2019\u0000www."}},"aqaraOpple":{"attributes":{"247":{"type":"Buffer","data":[1,33,195,11,3,40,25,4,33,168,67,5,33,20,0,6,36,1,0,0,0,0,8,33,25,1,10,33,229,181,12,32,1]}}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0029e736c2","zoneState":1}},"genPowerCfg":{"attributes":{"batteryVoltage":30}},"manuSpecificLumi":{"attributes":{"247":{"type":"Buffer","data":[1,33,191,11,3,40,25,4,33,168,19,5,33,22,0,6,36,1,0,0,0,0,8,33,25,1,10,33,30,209,12,32,1]},"252":0}},"genIdentify":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":1026,"inClusterList":[18,1280],"outClusterList":[],"clusters":{"aqaraOpple":{"attributes":{"280":1}},"genMultistateInput":{"attributes":{"presentValue":1}},"manuSpecificLumi":{"attributes":{"280":1}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":1026,"inClusterList":[12],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":25,"stackVersion":2,"hwVersion":1,"dateCode":"20200528","swBuildId":"2019\u0000www.","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1710120256205,"checkinInterval":3600}

mrskycriper, 2024-03-11:
Thanks, I’ll create a PR to fix this. But probably will need your help with testing the changes

eject-dream, 2024-03-12:
Sure, I'd be happy to help with testing the changes. Just let me know what you need me to do and when you need me to do it.

eject-dream, 2024-03-14:
@mrskycriper Thanks for letting me know. I've saved the content of the link as a js file and configured the full path of the file in settings > External converters. I've also restarted the container, but it seems like there are no changes. Could I have made a mistake somewhere? （ I've saved the file without making any modifications.）
[image]

[image]

Here's my docker-compose file for reference.
```yaml
version: '3'
services:
  zigbee2mqtt:
    image: koenkk/zigbee2mqtt:1.36.0
    container_name: zigbee2mqtt
    volumes:
      - ./data/z2m:/app/data
      - /run/udev:/run/udev:ro
    devices:
      - /dev/ttyACM0:/dev/ttyACM0
    restart: always
    ports:
      - "8080:8080"
```

mrskycriper, 2024-03-14:
Any errors in the beginning of the log file?

eject-dream, 2024-03-14:
> Any errors in the beginning of the log file?
No, I'm pretty sure this doesn't have any error logs, I've tried docker restart directly as well as hitting restart on the page

Also, I tried changing all these lines to full paths, but it still didn’t work.
const lumi = require('zigbee-herdsman-converters/lib/lumi');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const exposes = require('zigbee-herdsman-converters/lib/exposes');

mrskycriper, 2024-03-14:
I’m not familiar as to how z2m works in docker. But with home assistant addon the external converters go in the same place as configuration files

eject-dream, 2024-03-14:
> Any errors in the beginning of the log file?

Zigbee2MQTT:error 2024-03-14 22:11:07: Failed to load external converter file '/app/data/DJT12LM.js' (Unexpected identifier)

mrskycriper, 2024-03-14:
Well, it doesn’t like the file location

eject-dream, 2024-03-14:
https://github.com/Koenkk/zigbee2mqtt/blob/master/docker/docker-entrypoint.sh
 z2m configuration file is in /app/data/

---
I tried moving /app/data/DJT12LM.js to /app/DJT12LM.js, but it still didn’t work.
I tried moving /app/data/DJT12LM.js to /app/node_modules/DJT12LM.js, but it still didn’t work.
Let me try the home assistant addon, maybe it will be late, I have to figure out how to avoid letting the retarded devices at home test this thing without being affected

mrskycriper, 2024-03-14:
If they are in the same folder then the converter file path is just its name

eject-dream, 2024-03-14:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/lumi.ts#L1767
If this device is already supported, do External converters have a higher priority than the original one?

mrskycriper, 2024-03-14:
It always has higher priority

mrskycriper, 2024-03-15:
Any luck with testing?

eject-dream, 2024-03-16:
I've realized that my External converters seem to have stopped working after a certain version update, as for a long time I couldn't control my `ZNMHLDJ01LM` (since it supports manual pulling for curtain closure, I didn't pay much attention to it). After the last response to this issue, I discovered that it has now become 'Not supported' (the location of this External converters has always been the same, with no changes, the only change was the version of the Docker image), so I installed a new HAOS and purchased a new Zigbee dongle-e key specifically for additional testing (mainly because the original environment had too many devices, and the multitude of logs was quite disheartening).
`aaa` from this issus  #17046

The new usb key has arrived. I will update this issus as soon as there is the latest news.

eject-dream, 2024-03-16:
I have some bad news. The newly installed HAOS and the new USB key still report 'Failed to load external converter file' upon startup. Below are the screenshots and configuration files.

[image]

[image]

eject-dream, 2024-03-16:
It's a bit strange. I haven't set the environment variable `ZIGBEE2MQTT_DATA`, so according to the definition in the code, my aaa.js should be placed under /app/data, but this doesn't seem to work.
[image]

https://github.com/Koenkk/zigbee2mqtt/blob/2a53b8e59b457e0805318bdb38d2c74de651724d/lib/util/data.ts#L5-L12

eject-dream, 2024-03-18:
@mrskycriper I have some good news to share. I've found the correct location where HA has defined the `data_path`. I should place the script under the `extension` directory here. 

Moreover, I still need your assistance. 

In your script, the thirteenth line should be `const` instead of `cconst`. The `extend` in the `const definition` is missing a `]`. 

After making these modifications, there are still some error messages as follows:
```log
Zigbee2MQTT:error 2024-03-18 23:56:03: Failed to call 'ExternalExtension' 'start' (/app/data/extension/aaa.js:59
        battery({voltageToPercentage: '3V_2850_3000', voltage: true}),
        ^

TypeError: battery is not a function
    at /app/data/extension/aaa.js:59:9
    at Script.runInContext (node:vm:134:12)
    at Script.runInNewContext (node:vm:139:17)
    at Object.runInNewContext (node:vm:291:38)
    at Object.loadModuleFromText (/app/lib/util/utils.ts:139:8)
    at /app/lib/extension/externalExtension.ts:98:42
    at Array.map (<anonymous>)
    at ExternalExtension.loadUserDefinedExtensions (/app/lib/extension/externalExtension.ts:98:14)
    at ExternalExtension.start (/app/lib/extension/externalExtension.ts:19:14)
    at Controller.callExtensions (/app/lib/controller.ts:337:42))
```

mrskycriper, 2024-03-18:
> After making these modifications, there are still some error messages as follows:

I completely forgot that `battery` is not in the main release yet. I replaced it with the older definition for Aqara devices. 

This also presents an opportunity to see, how it is reported. Please give a feedback on an updated converter and be sure the contents of the State tab from the device page in Zigbee2MQTT. There might be some interesting information there.
[
The updated converter is on the same link as before.](https://gist.github.com/mrskycriper/f3078850b627e0441abf7cb1872b3d9c)

eject-dream, 2024-03-18:
new error 
```log
Zigbee2MQTT:info  2024-03-19 07:17:38: Zigbee2MQTT started!
TypeError: ConstructorClass is not a constructor
    at ExternalExtension.loadExtension (/app/lib/extension/externalExtension.ts:91:33)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
```

mrskycriper, 2024-03-19:
> new error

Slightly fixed the indentations, how about now?

eject-dream, 2024-03-19:
```log
Debug 2024-03-19 20:32:36Received MQTT message on 'zigbee2mqtt/bridge/request/extension/save' with data '{"code":"const lumi = require('zigbee-herdsman-converters/lib/lumi');\r\nconst {lumiZigbeeOTA} = lumi.modernExtend;\r\n\r\nconst modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');\r\nconst {battery, quirkCheckinInterval, numeric} = modernExtend;\r\n\r\nconst exposes = require('zigbee-herdsman-converters/lib/exposes');\r\nconst e = exposes.presets;\r\nconst ea = exposes.access;\r\n\r\nconst manufacturerCode = 0x115f;\r\n\r\nconst lumiVibrationT1 = () => {\r\n const exposes = [\r\n e.enum('action', ea.STATE, ['shake', 'strike_3_times']).withDescription('Triggered vibration action'),\r\n ];\r\n\r\n const fromZigbee = [\r\n {\r\n cluster: 'manuSpecificLumi',\r\n type: ['attributeReport', 'readResponse'],\r\n convert: (model, msg, publish, options, meta) => {\r\n if (msg.data.hasOwnProperty(280)) {\r\n return {action: 'shake'};\r\n }\r\n },\r\n },\r\n {\r\n cluster: 'genMultistateInput',\r\n type: ['attributeReport', 'readResponse'],\r\n convert: (model, msg, publish, options, meta) => {\r\n if (msg.data.hasOwnProperty('presentValue')) {\r\n return {action: 'strike_3_times'};\r\n }\r\n },\r\n },\r\n ];\r\n\r\n return {exposes, fromZigbee, isModernExtend: true};\r\n};\r\n\r\nconst definition = {\r\n zigbeeModel: ['lumi.vibration.agl01'],\r\n model: 'DJT12LM',\r\n vendor: 'Aqara',\r\n description: 'Vibration sensor T1',\r\n meta: {battery: {voltageToPercentage: '3V_2850_3000'}},\r\n fromZigbee: [lumi.fromZigbee.lumi_specific],\r\n exposes: [e.battery(), e.battery_voltage()],\r\n meta: {battery: {voltageToPercentage: '3V_2850_3000'}},\r\n configure: async (device, coordinatorEndpoint, logger) => {\r\n const endpoint = device.getEndpoint(1);\r\n await endpoint.read('genPowerCfg', ['batteryVoltage']);\r\n },\r\n extend: [\r\n lumiVibrationT1(),\r\n numeric({\r\n name: 'vibration_sensetivity',\r\n cluster: 'manuSpecificLumi',\r\n attribute: {ID: 0x00FC, type: 0x39},\r\n valueMin: 0,\r\n valueMax: 10,\r\n description: 'Vibration sensetivity',\r\n access: 'ALL',\r\n zigbeeCommandOptions: {manufacturerCode},\r\n }),\r\n lumiZigbeeOTA(),\r\n quirkCheckinInterval('1_HOUR'),\r\n ],\r\n};\r\n\r\nmodule.exports = definition;","name":"aaa.js","transaction":"dcizo-4"}'
Error 2024-03-19 20:32:36Request 'zigbee2mqtt/bridge/request/extension/save' failed with error: 'ConstructorClass is not a constructor'
Info 2024-03-19 20:32:36MQTT publish: topic 'zigbee2mqtt/bridge/response/extension/save', payload '{"data":{},"error":"ConstructorClass is not a constructor","status":"error","transaction":"dcizo-4"}'
```

mrskycriper, 2024-03-19:
> ```
> Debug 2024-03-19 20:32:36Received MQTT message on 'zigbee2mqtt/bridge/request/extension/save' with data '{"code":"const lumi = require('zigbee-herdsman-converters/lib/lumi');\r\nconst [...]

Wait a minute, why are you saving it as an extension? They are not the same as external converters and are functionally incompatible.

mrskycriper, 2024-03-19:
> I apologize for the confusion. I assumed they were the same and took a shortcut by adding it directly on the page. I have now placed the script in the correct location and configured the external_converters. Here is the latest error log:

Doesn't seem like it's a problem with the converter this time. What happens if you remove the converter file temporarily?

eject-dream, 2024-03-19:
This will start normally without any abnormalities.

eject-dream, 2024-03-20:
Currently, the system starts up normally and displays more items, but none of the items have any values. 

Here is the log for the shaking event:
```log
Debug 2024-03-20 22:46:24Received Zigbee message from '0x04cf8cdf3c89c90a', type 'attributeReport', cluster 'ssIasZone', data '{"45":1}' from endpoint 2 with groupID 0
Debug 2024-03-20 22:46:24No converter available for 'DJT12LM' with cluster 'ssIasZone' and type 'attributeReport' and data '{"45":1}'
```

Here is the log for the triple tap event:
```log
Zigbee2MQTT:debug 2024-03-20 22:37:59: Received Zigbee message from '0x04cf8cdf3c89c90a', type 'attributeReport', cluster 'ssIasZone', data '{"45":2}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-03-20 22:37:59: No converter available for 'DJT12LM' with cluster 'ssIasZone' and type 'attributeReport' and data '{"45":2}'
```

And here is the log for adjusting the Vibration Sensitivity:
```log
Zigbee2MQTT:debug 2024-03-20 22:42:09: Received MQTT message on 'zigbee2mqtt/0x04cf8cdf3c89c90a/get' with data '{"vibration_sensetivity":""}'
Zigbee2MQTT:debug 2024-03-20 22:42:09: Publishing get 'get' 'vibration_sensetivity' to '0x04cf8cdf3c89c90a'
Zigbee2MQTT:debug 2024-03-20 22:42:16: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2024-03-20 22:42:26: Received MQTT message on 'zigbee2mqtt/0x04cf8cdf3c89c90a/set' with data '{"vibration_sensetivity":4}'
Zigbee2MQTT:debug 2024-03-20 22:42:26: Publishing 'set' 'vibration_sensetivity' to '0x04cf8cdf3c89c90a'
Zigbee2MQTT:error 2024-03-20 22:42:29: Publish 'get' 'vibration_sensetivity' to '0x04cf8cdf3c89c90a' failed: 'Error: Read 0x04cf8cdf3c89c90a/1 manuSpecificLumi([252], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 32825 - 1 - 3 - 64704 - 1 after 10000ms)'
Zigbee2MQTT:debug 2024-03-20 22:42:29: Error: Read 0x04cf8cdf3c89c90a/1 manuSpecificLumi([252], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 32825 - 1 - 3 - 64704 - 1 after 10000ms)
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
    at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:492:47)
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:466:28)
    at Object.toZigbee.convertGet (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:488:13)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:286:21)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)
Zigbee2MQTT:error 2024-03-20 22:42:49: Publish 'set' 'vibration_sensetivity' to '0x04cf8cdf3c89c90a' failed: 'Error: Write 0x04cf8cdf3c89c90a/1 manuSpecificLumi({"252":{"value":4,"type":57}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 32825 - 1 - 4 - 64704 - 4 after 10000ms)'
Zigbee2MQTT:debug 2024-03-20 22:42:49: Error: Write 0x04cf8cdf3c89c90a/1 manuSpecificLumi({"252":{"value":4,"type":57}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 32825 - 1 - 4 - 64704 - 4 after 10000ms)
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
    at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:492:47)
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
    at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:390:28)
    at Object.toZigbee.convertSet (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:484:13)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:259:36)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)
```

database.db:
```text
{"id":2,"type":"EndDevice","ieeeAddr":"0x04cf8cdf3c89c90a","nwkAddr":32825,"manufId":4447,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.vibration.agl01","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.vibration.agl01","manufacturerName":"LUMI","powerSource":3,"zclVersion":3,"appVersion":25,"stackVersion":2,"hwVersion":1,"dateCode":"20200528","swBuildId":"2019\u0000www."}},"ssIasZone":{"attributes":{"45":2}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":1026,"inClusterList":[1280],"outClusterList":[],"clusters":{"ssIasZone":{"attributes":{"45":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":25,"stackVersion":2,"hwVersion":1,"dateCode":"20200528","swBuildId":"2019\u0000www.","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1710945390313,"checkinInterval":3600}# 
```

state.json :
```text
{
    "0x04cf8cdf3c89c90a": {
        "update": {
            "state": "idle",
            "installed_version": 25,
            "latest_version": 25
        },
        "linkquality": 184
    }
```

Here is a screenshot:
[image]

mrskycriper, 2024-03-20:
Thanks, I'll update the converter. Strange that it worked differently before.

mrskycriper, 2024-03-20:
> And here is the log for adjusting the Vibration Sensitivity

Have you tried waking it up before setting the number?

eject-dream, 2024-03-20:
> Have you tried waking it up before setting the number?

new log
```log
Debug 2024-03-20 23:27:33Publishing 'set' 'vibration_sensetivity' to '0x04cf8cdf3c89c90a'
Debug 2024-03-20 23:27:33Received Zigbee message from '0x04cf8cdf3c89c90a', type 'attributeReport', cluster 'ssIasZone', data '{"45":1}' from endpoint 2 with groupID 0
Debug 2024-03-20 23:27:33No converter available for 'DJT12LM' with cluster 'ssIasZone' and type 'attributeReport' and data '{"45":1}'
Error 2024-03-20 23:27:34Publish 'set' 'vibration_sensetivity' to '0x04cf8cdf3c89c90a' failed: 'Error: Write 0x04cf8cdf3c89c90a/1 manuSpecificLumi({"252":{"value":6,"type":57}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'INVALID_DATA_TYPE')'
Debug 2024-03-20 23:27:34Error: Write 0x04cf8cdf3c89c90a/1 manuSpecificLumi({"252":{"value":6,"type":57}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'INVALID_DATA_TYPE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:319:28) at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:392:22) at Object.toZigbee.convertSet (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:484:13) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:259:36) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)
```

mrskycriper, 2024-03-25:
@eject-dream could you please test the updated converter on the dev branch of Zigbee2MQTT?

[link](https://gist.github.com/mrskycriper/f3078850b627e0441abf7cb1872b3d9c)

eject-dream, 2024-03-26:
Long time no see, I miss you a lot.

The good news is that the Action is responding. The bad news is that it's indicating an empty string, and all other values, including the battery percentage, are Null.
[image]

```log
Debug 2024-03-26 20:55:26Received Zigbee message from '0x04cf8cdf3c89c90a', type 'attributeReport', cluster 'ssIasZone', data '{"45":1}' from endpoint 2 with groupID 0
Info 2024-03-26 20:55:26MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c89c90a', payload '{"action":"shake","battery":null,"device_temperature":null,"linkquality":148,"power_outage_count":null,"update":{"installed_version":25,"latest_version":25,"state":"idle"},"update_available":null,"voltage":null}'
Info 2024-03-26 20:55:26MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c89c90a', payload '{"action":"","battery":null,"device_temperature":null,"linkquality":148,"power_outage_count":null,"update":{"installed_version":25,"latest_version":25,"state":"idle"},"update_available":null,"voltage":null}'
Info 2024-03-26 20:55:26MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c89c90a/action', payload 'shake'
```

eject-dream, 2024-03-26:
Here's a pleasant surprise - I've set up a new DJT12LM and, after changing the battery, incorporated it into the test network. Attached are the latest screenshots and the logs retrieved from Home Assistant (HA).

[image]

```log
Zigbee2MQTT:debug 2024-03-26 21:22:24: Received Zigbee message from 'new_device', type 'attributeReport', cluster 'ssIasZone', data '{"45":1}' from endpoint 2 with groupID 0
Zigbee2MQTT:info  2024-03-26 21:22:24: MQTT publish: topic 'zigbee2mqtt/new_device', payload '{"action":"shake","battery":100,"device_temperature":null,"linkquality":132,"power_outage_count":null,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":3200}'
Zigbee2MQTT:info  2024-03-26 21:22:24: MQTT publish: topic 'zigbee2mqtt/new_device', payload '{"action":"","battery":100,"device_temperature":null,"linkquality":132,"power_outage_count":null,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":3200}'
Zigbee2MQTT:info  2024-03-26 21:22:24: MQTT publish: topic 'zigbee2mqtt/new_device/action', payload 'shake'
```

```log
Zigbee2MQTT:debug 2024-03-26 21:25:39: Received Zigbee message from 'new_device', type 'attributeReport', cluster 'ssIasZone', data '{"45":2}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2024-03-26 21:25:39: MQTT publish: topic 'zigbee2mqtt/new_device', payload '{"action":"triple_strike","battery":100,"device_temperature":null,"linkquality":176,"power_outage_count":null,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":3200}'
Zigbee2MQTT:info  2024-03-26 21:25:39: MQTT publish: topic 'zigbee2mqtt/new_device', payload '{"action":"","battery":100,"device_temperature":null,"linkquality":176,"power_outage_count":null,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":3200}'
Zigbee2MQTT:info  2024-03-26 21:25:39: MQTT publish: topic 'zigbee2mqtt/new_device/action', payload 'triple_strike'
```

eject-dream, 2024-03-28:
no~~
[image]

mrskycriper, 2024-03-28:
> no~~
> 
> [image]
> 
> 

I see, oh well, I'll remove them and then my PR will be ready for merge.

Those buttons should work, but only if the device was woken up right before the request.

eject-dream, 2024-03-28:
Wake-up
```log
Debug 2024-03-29 00:03:41Received Zigbee message from 'test', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.vibration.agl01"}' from endpoint 1 with groupID 0
Debug 2024-03-29 00:03:41Received Zigbee message from 'test', type 'attributeReport', cluster 'manuSpecificLumi', data '{"247":{"data":[1,33,229,11,3,40,25,4,33,168,67,5,33,28,0,6,36,5,0,0,0,0,8,33,25,1,10,33,0,0,12,32,1],"type":"Buffer"}}' from endpoint 1 with groupID 0
```

Get battery
```log
Debug 2024-03-29 00:03:45Received MQTT message on 'zigbee2mqtt/test/get' with data '{"battery":""}'
Debug 2024-03-29 00:03:45Publishing get 'get' 'battery' to 'test'
Debug 2024-03-29 00:03:45Received Zigbee message from 'test', type 'readResponse', cluster 'genPowerCfg', data '{"batteryVoltage":30}' from endpoint 1 with groupID 0
Info 2024-03-29 00:03:45MQTT publish: topic 'zigbee2mqtt/test', payload '{"action":null,"battery":100,"device_temperature":null,"linkquality":180,"power_outage_count":null,"update":{"installed_version":25,"latest_version":25,"state":"idle"},"update_available":null,"voltage":3000}'
Debug 2024-03-29 00:03:45Reading attribute failed: Error: ZCL command 0x04cf8cdf3c89c90a/1 genPowerCfg.read(["batteryPercentageRemaining","batteryVoltage"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE'), device probably doesn't support it
```

Get voltage
```log
Debug 2024-03-29 00:06:06Received MQTT message on 'zigbee2mqtt/test/get' with data '{"voltage":""}'
Debug 2024-03-29 00:06:06Publishing get 'get' 'voltage' to 'test'
Debug 2024-03-29 00:06:07Received Zigbee message from 'test', type 'readResponse', cluster 'genPowerCfg', data '{"batteryVoltage":30}' from endpoint 1 with groupID 0
Info 2024-03-29 00:06:07MQTT publish: topic 'zigbee2mqtt/test', payload '{"action":null,"battery":100,"device_temperature":null,"linkquality":192,"power_outage_count":null,"update":{"installed_version":25,"latest_version":25,"state":"idle"},"update_available":null,"voltage":3000}'
Debug 2024-03-29 00:06:07Reading attribute failed: Error: ZCL command 0x04cf8cdf3c89c90a/1 genPowerCfg.read(["batteryPercentageRemaining","batteryVoltage"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE'), device probably doesn't support it
```

mrskycriper, 2024-03-28:
> Wake-up
>
> ```log
>
> Debug 2024-03-29 00:03:41Received Zigbee message from 'test', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.vibration.agl01"}' from endpoint 1 with groupID 0 [...]

It looks like the requests were successful. But the error message was sent, that is confusing.

mrskycriper, 2024-03-29:
UPD

This message appears because battery percentage is not reported directly, but calculated from voltage. I'll change converter a little bit to make this more clear.

closed 2024-03-31 by Koenkk

wagaman, 2024-09-16:
Last seen
N/A
Availability
[Disabled](https://www.zigbee2mqtt.io/guide/configuration/device-availability.html#availability-advanced-configuration)
Device type
EndDevice
Zigbee Model
lumi.vibration.agl01
Zigbee Manufacturer
LUMI
Description
Vibration sensor T1
Support status
Supported
IEEE Address
0x04cf8cdf3c89cbad
Network address
0x4C02 / 19458
Firmware build date
20200528
Firmware version
2019www.
Manufacturer
[Aqara](https://www.zigbee2mqtt.io/supported-devices/#v=Aqara)
Model
[DJT12LM](https://www.zigbee2mqtt.io/devices/DJT12LM.html#aqara-djt12lm)
Power
Battery 100%
Interview completed
True

```

2024-09-16 17:07:56z2m: Successfully interviewed '餐厅灯震动', device has successfully been paired
Info 2024-09-16 17:07:56z2m: Device '餐厅灯震动' is supported, identified as: Aqara Vibration sensor T1 (DJT12LM)
Info 2024-09-16 17:07:56z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Vibration sensor T1","exposes":[{"access":5,"category":"diagnostic","description":"Remaining battery in %","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":5,"category":"diagnostic","description":"Reported battery voltage in millivolts","label":"Voltage","name":"voltage","property":"voltage","type":"numeric","unit":"mV"},{"access":1,"category":"diagnostic","description":"Triggered action (e.g. a button click)","label":"Action","name":"action","property":"action","type":"enum","values":["shake","triple_strike"]},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"DJT12LM","options":[],"supports_ota":true,"vendor":"Aqara"},"friendly_name":"餐厅灯震动","ieee_address":"0x04cf8cdf3c89cbad","status":"successful","supported":true},"type":"device_interview"}'

Info 2024-09-16 17:08:17z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/餐厅灯震动', payload '{"action":null,"battery":100,"linkquality":189,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":3000}'

{"installed_version":43,"latest_version":32,"state":"idle"},"update_available":null,"voltage":229}'
Info 2024-09-16 17:08:21z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/餐厅灯震动', payload '{"action":null,"battery":100,"linkquality":185,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":3000}'
Info 2024-09-16 17:08:22z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/餐厅灯震动', payload '{"action":null,"battery":100,"linkquality":182,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":3000}'
Info 2024-09-16 17:08:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/餐厅灯震动', payload '{"action":null,"battery":100,"linkquality":174,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":3000}'
Info 2024-09-16 17:08:24z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/餐厅灯震动', payload '{"action":null,"battery":100,"linkquality":160,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":3000}'
Info 2024-09-16 17:08:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/餐厅灯震动', payload '{"action":null,"battery":100,"linkquality":171,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":3000}'
```

I got two DJT12LM with firmware 20200528. Aciton shows up with "null" no matter how much I shake the devices.

Could you please take a look? Thanks in advance.

wagaman, 2024-09-17:
Failed to configure '0x54ef4410009c30bf', attempt 3 (Error: ZCL command 0x54ef4410009c30bf/1 genPowerCfg.configReport([{"minimumReportInterval":3600,"maximumReportInterval":65000,"reportableChange":10,"attribute":"batteryPercentageRemaining"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:347:28) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:916:26) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:648:9) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:110:13) at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:155:17) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:193:21) at Configure.configure (/app/lib/extension/configure.ts:125:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

wagaman, 2024-12-02:
@mrskycriper  do you have time to take a look at this bug for me, please? thank you very much in advance.

mrskycriper, 2024-12-02:
> @mrskycriper  do you have time to take a look at this bug for me, please? thank you very much in advance. 

Sorry, unfortunately, I don't have time or motivation to continue working on Zigbee2MQTT for now.

wagaman, 2024-12-02:
> > @mrskycriper  do you have time to take a look at this bug for me, please? thank you very much in advance.
> 
> Sorry, unfortunately, I don't have time or motivation to continue working on Zigbee2MQTT for now.

No worries. Thank you for letting me know.

bonjour81, 2024-12-16:
Hello, may I ask how good is DJT12LM work with zigbee2mqtt? I was not able to figure out correctly from this discussion. Documentation mentions it publishes battery, action, link... But no vibration data?
https://www.zigbee2mqtt.io/devices/DJT12LM.html

wagaman, 2024-12-22:
> Hello, may I ask how good is DJT12LM work with zigbee2mqtt? I was not able to figure out correctly from this discussion. Documentation mentions it publishes battery, action, link... But no vibration data? https://www.zigbee2mqtt.io/devices/DJT12LM.html

it does not work properly with the version 2019www.

eject-dream, 2024-12-26:
@wagaman Apologies for the delay in response; I’ve been receiving notifications but haven’t had the time to test or log in to GitHub. I just tested the DJT12LM I have, and it is working properly. My current z2m version is 1.40.1, and the firmware version of the DJT12LM is 2019\u0000www.

My judgment is based on the logs. One very frustrating aspect of this device is that it sends two messages almost simultaneously: one for your action (e.g., triple tap/shake), and another that is empty. This causes the action displayed on the z2m web interface to always appear as empty.

For anyone reading this message in the future, my advice is: avoid purchasing the DJT12LM. Consider getting the DJT11LM instead (I don’t own the DJT11LM, so you’ll need to check with others who have it to see if it’s worth buying).

eject-dream, 2024-12-26:
Add some screenshots

[image]
---
[image]
---
[image]

wagaman, 2024-12-26:
@eject-dream

I tried to pair the sensor again and it shows this error.

Error 2024-12-26 15:55:51z2m: Failed to configure '0x04cf8cdf3c89cbad', attempt 1 (Error: ZCL command 0x04cf8cdf3c89cbad/1 genPowerCfg.configReport([{"minimumReportInterval":3600,"maximumReportInterval":65000,"reportableChange":10,"attribute":"batteryPercentageRemaining"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:350:28) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:956:26) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:688:9) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:162:17) at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:211:17) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:206:21) at Configure.configure (/app/lib/extension/configure.ts:130:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:211:17))

[image]

wagaman, 2024-12-26:
although it shows supported, the action is empty and I double checked there is no "double action"...

eject-dream, 2024-12-26:
When I first raised this issue, I was using the CC2652P. Due to the overwhelming amount of logs during debugging, I later purchased a ZBDongle-E and installed a fresh instance of HAOS specifically for testing (special thanks to @mrskycriper for their efforts). After the PR was merged and officially released, I promptly upgraded to the mainline version and added the DJT12LM to the CC2652P network. If I recall correctly, it didn’t work and threw some errors. However, since I wasn’t actively using this device and was quite busy at the time, I left it aside.

Later, one day, I got really frustrated with the CC2652P’s latency and disconnection issues (a ton of error logs). I spent several days migrating all my devices to the ZBDongle-E and rewrote parts of the automation logic. So, during my recent test, I added this device to the ZBDongle-E network for the first time.

That’s everything. I hope it’s helpful. 
I see the device you are using is zstack3x0. @wagaman 

Lastly, I’d like to reiterate: if possible, avoid purchasing the DJT12LM.

hongliangl, 2025-01-11:
DJT12LM is a piece of shit in z2m. Just buy DJT11LM and don't waste your time.

z-x-f, 2025-02-20:
Hi,@wagaman, I’ve submitted [PR #8788](https://github.com/Koenkk/zigbee-herdsman-converters/pull/8788) that might fix the issue you mentioned. Could you please test it out and let me know if it resolves the problem? Thanks a lot!

wagaman, 2025-03-10:
> Hi,[@wagaman](https://github.com/wagaman), I’ve submitted [PR #8788](https://github.com/Koenkk/zigbee-herdsman-converters/pull/8788) that might fix the issue you mentioned. Could you please test it out and let me know if it resolves the problem? Thanks a lot!

Hi @z-x-f  it is working like a charm. Sorry that i did not get back to you on time as I was away from the sensors. thank you so much again.

wagaman, 2025-03-10:
@z-x-f I tested further it looks like the "movement" status is working great. However, please take a look of the following 

{
    "linkquality": 163,
    "action": "movement",
    "battery": null,
    "report_interval": null,
    "sensitivity_adjustment": null,
    "update": {
        "installed_version": -1,
        "latest_version": -1,
        "state": null
    },
    "voltage": null
}

it looks like I cannot set sensitivity and report interval (I clicked on the buttons in the UI and also pressed once the physical button on the sensor) and the battery is not reported. Is everything working fine on your end? thanks

victorhugoam18, 2025-04-14:
Hi, I bought two DJT12LM and both presents the same issue... no action is issued (movement, shake and triple ), and battery/voltage isn't shown... I have wait more than 24 hours and I have paired sometimes with no success.

Does anyone know how solve this problem? Thanks

(https://github.com/Koenkk/zigbee2mqtt/issues/21731#top)
#21731
[image]

[image]