Koenkk/zigbee2mqtt #21737: Coordinator disconnect

Sherk737, 2024-03-09:
### What happened?

Randomly I got my coordinator ( Sonoff Zigbee 3.0 USB Dongle Plus ) disconnect from my Z2M and disconnect my devices 
this is the error msg  (failed (SRSP - AF - dataRequest after 6000ms))

### What did you expect to happen?

I looked around and see that it's a memory error and it cause due using tuya devices spamming so I disconnect them from my Z2M but it still happened 

### Zigbee2MQTT version

1.36.0 

### Adapter firmware version

20230507

### Adapter

 Sonoff Zigbee 3.0 USB Dongle Plus

### Setup

Docker Unraid

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14547873/log.txt)

cvhoang, 2024-03-10:
I'm having exactly the same issue as @Sherk737. My setup is pretty identical to that of @Sherk737. I have a few Tuya presence sensors spamming the network. However, I haven't had any issues until very recently when I upgraded to the latest version of Z2M.

chris-1243, 2024-03-10:
Is the usb autosuspend feature disabled ?

`cat /sys/module/usbcore/parameters/autosuspend `

If the return is `1` or `2`, it means it is on.

cvhoang, 2024-03-10:
Thanks @chris-1243. The command indeed returns `2`. I've made the changes and will monitor to see if they fixes the issue.  However, Z2M was running great before the last few updates.

chris-1243, 2024-03-11:
It may be related to an other issue. At least, you remove one possible problem

Sherk737, 2024-03-11:
I change coordinator from sonoff dongle to poe dongle (Tubeszb) still got the same issue

Sherk737, 2024-03-11:
Thanks @chris-1243. The command  returns 2 but I change the coordinator to poe now with same issue

Sherk737, 2024-03-11:
@cvhoang  any update in your system ?

Koenkk, 2024-03-11:
The `20230507` fw might be unstable, try the [20221226 firmware](https://github.com/Koenkk/Z-Stack-firmware/tree/Z-Stack_3.x.0_coordinator_20221226/coordinator/Z-Stack_3.x.0/bin)

cvhoang, 2024-03-11:
@Sherk737 it's still unstable and has just crashed :(

@Koenkk I'll revert the firmware as there is some correlation between crashes and firmware upgrade.

Sherk737, 2024-03-12:
@Koenkk  Thanks for reply and I revert the firmware back to 26122022, I will monitor and check stability. will update you here

Sherk737, 2024-03-12:
after I revert to older firmware I got same issue again

cvhoang, 2024-03-12:
@Koenkk @Sherk737 Reverting the firmware seems to have stabilise the issues in the last 24 hours.

Sherk737, 2024-03-13:
same with me now its stabilize after reverting to old firmware - in beginning the issue occurred once then I reboot the dongle nd now its stable

danielsmith89, 2024-03-13:
I'm still on the "20220219" firmware which worked perfectly for over a year, but in the last few days I've been having the issues above exactly as described.

Sherk737, 2024-03-13:
@Koenkk @cvhoang Now z2m is stable and working without any issues after downgrade the dongle firmware

@danielsmith89 I prefer to update you dongle to [20221226 firmware](https://github.com/Koenkk/Z-Stack-firmware/tree/Z-Stack_3.x.0_coordinator_20221226/coordinator/Z-Stack_3.x.0/bin) and recheck your system if it still have an issue open new issue with your system log

closed 2024-03-13 by Sherk737