Koenkk/zigbee2mqtt #21738: [New device support]: TS0601 _TZE204_7gclukjs

ayasystems, 2024-03-09:
### Link

https://es.aliexpress.com/item/1005004704966668.html?spm=a2g0o.order_list.order_list_main.39.2d6d194dDYF1wq&gatewayAdapt=glo2esp

### Database entry

{"id":12,"type":"Router","ieeeAddr":"0xa4c1385c2518322c","nwkAddr":57339,"manufId":4417,"manufName":"_TZE204_7gclukjs","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE204_7gclukjs","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1710022644573}

### Comments

Version Zigbee 24G
manufacturerName _TZE204_7gclukjs

I don't have tuya brigde to get the correct datapoints

2024-03-09 23:29:10Datapoint 104 not defined for '_TZE204_7gclukjs' with value 1
2024-03-09 23:29:10Datapoint 104 not defined for '_TZE204_7gclukjs' with value 1
2024-03-09 23:29:10Datapoint 9 not defined for '_TZE204_7gclukjs' with value 13
2024-03-09 23:29:10Datapoint 103 not defined for '_TZE204_7gclukjs' with value 5
2024-03-09 23:29:10Datapoint 2 not defined for '_TZE204_7gclukjs' with value 0
2024-03-09 23:29:10Datapoint 102 not defined for '_TZE204_7gclukjs' with value 0
2024-03-09 23:29:11Datapoint 3 not defined for '_TZE204_7gclukjs' with value 75
2024-03-09 23:29:11Datapoint 4 not defined for '_TZE204_7gclukjs' with value 750
2024-03-09 23:29:11Datapoint 105 not defined for '_TZE204_7gclukjs' with value 5
2024-03-09 23:29:11Datapoint 9 not defined for '_TZE204_7gclukjs' with value 12
2024-03-09 23:29:11Datapoint 9 not defined for '_TZE204_7gclukjs' with value 12
2024-03-09 23:29:11Datapoint 9 not defined for '_TZE204_7gclukjs' with value 12
2024-03-09 23:29:11Datapoint 9 not defined for '_TZE204_7gclukjs' with value 12
2024-03-09 23:29:11Datapoint 9 not defined for '_TZE204_7gclukjs' with value 12
2024-03-09 23:29:12Datapoint 9 not defined for '_TZE204_7gclukjs' with value 12
2024-03-09 23:29:13Datapoint 9 not defined for '_TZE204_7gclukjs' with value 12
2024-03-09 23:29:14Datapoint 255 not defined for '_TZE204_7gclukjs' with value false
2024-03-09 23:29:14Datapoint 255 not defined for '_TZE204_7gclukjs' with value false
2024-03-09 23:29:14Datapoint 1 not defined for '_TZE204_7gclukjs' with value 1
2024-03-09 23:29:14Datapoint 104 not defined for '_TZE204_7gclukjs' with value 1
2024-03-09 23:29:14Datapoint 9 not defined for '_TZE204_7gclukjs' with value 12
2024-03-09 23:29:14Datapoint 103 not defined for '_TZE204_7gclukjs' with value 5
2024-03-09 23:29:14Datapoint 2 not defined for '_TZE204_7gclukjs' with value 0
2024-03-09 23:29:14Datapoint 2 not defined for '_TZE204_7gclukjs' with value 0
2024-03-09 23:29:14Datapoint 102 not defined for '_TZE204_7gclukjs' with value 0
2024-03-09 23:29:14Datapoint 3 not defined for '_TZE204_7gclukjs' with value 75

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_7gclukjs',
        },
    ],
    model: 'TS0601',
    vendor: 'TuYa',
    description: 'RadarTuya',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        exposes: [
            e.presence(), e.illuminance_lux(), 
        ],
        meta: {
            tuyaDatapoints: [
     
            ],
    },
};

module.exports = definition;
```

ayasystems, 2024-03-09:
Atacched an excel with all datapoints registered in debug mode

[image]

[datapoints_debug.xlsx](https://github.com/Koenkk/zigbee2mqtt/files/14548054/datapoints_debug.xlsx)

ayasystems, 2024-03-10:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_7gclukjs',
        },
    ],
    model: 'TS0601',
    vendor: 'TuYa',
    description: 'RadarTuya',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        exposes: [
				exposes.enum('approach_distance', ea.STATE, ['none', 'small_movement', 'large_movement'])
					.withDescription('traffic of movement from the point of view of the radar'),
				e.presence(), 
				 exposes.numeric('current_distance', ea.STATE).withDescription('Distance to target').withUnit('cm'),
				 exposes.numeric('current_distance2', ea.STATE).withDescription('Distance to target 2').withUnit('cm'),
        ],
        meta: {
            tuyaDatapoints: [
				[104, 'presence', tuya.valueConverter.trueFalse1],
				[1, 'approach_distance', tuya.valueConverterBasic.lookup({'none': 0, 'small_movement': 1, 'large_movement': 2})],
				[103, 'current_distance', tuya.valueConverter.raw],
				[9, 'current_distance2', tuya.valueConverter.raw],
            ],
    },
};

module.exports = definition;`

I've tested this settings and here is the result:

[image]

ayasystems, 2024-03-15:
I have bought a zigbee bridge to extract all datapoints. I will back soon

yarik-m, 2024-03-18:
hi, DP103 its illuminance

Vallefor, 2024-03-19:
Updated it a little. Can't determine delay datapoint, but in other parameters I am sure (have tuya gateway).

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_7gclukjs',
        },
    ],
    model: 'TS0601',
    vendor: 'TuYa',
    description: 'RadarTuya',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        exposes: [
				exposes.enum('state', ea.STATE, ['none', 'presence', 'move']),
				e.presence(), 
				 exposes.numeric('current_distance', ea.STATE).withDescription('Distance to target').withUnit('cm'),
				 exposes.numeric('illuminance_lux', ea.STATE).withDescription('Illuminance').withUnit('lux'),
				 exposes.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(9).withValueStep(1).withDescription('sensitivity of the radar'),
				 exposes.numeric('minimum_range', ea.STATE_SET).withValueMin(0).withValueMax(825).withValueStep(75).withDescription('Minimum range').withUnit('cm'),
				 exposes.numeric('maximum_range', ea.STATE_SET).withValueMin(75).withValueMax(900).withValueStep(75).withDescription('Maximum range').withUnit('cm'),
        ],
        meta: {
            tuyaDatapoints: [
				[104, 'presence', tuya.valueConverter.trueFalse1],
				[1, 'state', tuya.valueConverterBasic.lookup({'none': 0, 'presence': 1, 'move': 2})],
				[103, 'illuminance_lux', tuya.valueConverter.raw],
				[9, 'current_distance', tuya.valueConverter.raw],
                		[105, 'radar_sensitivity', tuya.valueConverter.raw],
                		[3, 'minimum_range', tuya.valueConverter.raw],
                		[4, 'maximum_range', tuya.valueConverter.raw],
                
            ],
    },
};

module.exports = definition;
```

ayasystems, 2024-03-22:
I can see these values with tuya bridge on iot.tuya cloud

{"1":"Induction state"
,"2":"Motion Sensitivity"
,"3":"Near Detection"
,"4":"Far Detection"
,"9":"Target distance"
,"102":"Presence Sensitivity"
,"103":"Illuminance Value"
,"104":"有无人状态"
,"105":"Delay Time"}

and these options to set values

[image]

[image]

So i think this is correct

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_7gclukjs',
        },
    ],
    model: 'TS0601',
    vendor: 'TuYa',
    description: 'RadarTuya',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        exposes: [
				e.enum('approach_distance', ea.STATE, ['none', 'small_movement', 'large_movement']).withDescription('traffic of movement from the point of view of the radar'),
				e.presence(), 
				exposes.numeric('illuminance_lux', ea.STATE).withDescription('Illuminance').withUnit('lux'),
				exposes.numeric('current_distance', ea.STATE).withDescription('Distance to target').withUnit('cm'),
				exposes.numeric('delay_time', ea.STATE_SET).withValueMin(0).withValueMax(600).withValueStep(5).withDescription('Delay time').withUnit('s'),
				exposes.numeric('near_detection', ea.STATE_SET).withValueMin(0).withValueMax(825).withValueStep(75).withDescription('Near Detection').withUnit('cm'),		
				exposes.numeric('far_detection', ea.STATE_SET).withValueMin(75).withValueMax(900).withValueStep(75).withDescription('Far Detection').withUnit('cm'),
				exposes.numeric('motion_sensitivity', ea.STATE_SET).withDescription('Motion Sensitivity').withValueMin(0).withValueMax(10).withValueStep(1).withDescription('Motion Sensitivity'),		
        ],
        meta: {
            tuyaDatapoints: [
				[104, 'presence', tuya.valueConverter.trueFalse1],
				[1, 'approach_distance', tuya.valueConverterBasic.lookup({'none': 0, 'small_movement': 1, 'large_movement': 2})],
				[103, 'illuminance_lux', tuya.valueConverter.raw],
				[9, 'current_distance', tuya.valueConverter.raw],
				[105, 'delay_time',   tuya.valueConverter.raw],
                [3, 'near_detection', tuya.valueConverter.raw],
                [4, 'far_detection', tuya.valueConverter.raw],
                [2, 'motion_sensitivity', tuya.valueConverter.raw],
				
            ],
    },
};

module.exports = definition;
```

Koenkk, 2024-03-23:
looks good, could you make a pull request?

justicepng, 2024-03-23:
> I can see these values with tuya bridge on iot.tuya cloud
> 
> {"1":"Induction state" ,"2":"Motion Sensitivity" ,"3":"Near Detection" ,"4":"Far Detection" ,"9":"Target distance" ,"102":"Presence Sensitivity" ,"103":"Illuminance Value" ,"104":"有无人状态" ,"105":"Delay Time"}
> 
> and these options to set values
> 
> ![image](https://private-user-images.githubusercontent.com/7864168/315985368-82b615ee-18d0-4bf0-96d0-b565cae8e3e0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTExODc4MDcsIm5iZiI6MTcxMTE4NzUwNywicGF0aCI6Ii83ODY0MTY4LzMxNTk4NTM2OC04MmI2MTVlZS0xOGQwLTRiZjAtOTZkMC1iNTY1Y2FlOGUzZTAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDMyMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAzMjNUMDk1MTQ3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MjY1NzY1MjY0NjIxYTI0OTNkYTZjYjc0YjBlM2NiMjAwNDczMjA0ZDBmNTE3ZTEwN2FhNTNjMjA3OTBjNzU0NyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.W2dpPm8iGd3akafFcrUTnaWWvLbx5bHjJDLhvAaraRQ)
> 
> ![image](https://private-user-images.githubusercontent.com/7864168/315985691-d5b5952a-df2f-41e0-80b7-98417e4e6ea8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTExODc4MDcsIm5iZiI6MTcxMTE4NzUwNywicGF0aCI6Ii83ODY0MTY4LzMxNTk4NTY5MS1kNWI1OTUyYS1kZjJmLTQxZTAtODBiNy05ODQxN2U0ZTZlYTgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDMyMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAzMjNUMDk1MTQ3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OTMyNjY1YzYyOTIwODkzOWUyOTFiMzMxMjc2YmJjM2RiZDMwZjk3ODc1MTZhZDY3MzUzZDZjYWU1ZTIzMjhlYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.4JCevWjlx1GyaoPtWqeCvAxKq84lKDvrIPWHPtTqBbw)
> 
> So i think this is correct.
> 
> **Note that state has beed changed to none/presence/Move and presence and seems works fine**
> 
> ```
> 
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const extend = require('zigbee-herdsman-converters/lib/extend');
> const e = exposes.presets;
> const ea = exposes.access;
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> 
> const definition = {
>     // Since a lot of TuYa devices use the same modelID, but use different datapoints
>     // it's necessary to provide a fingerprint instead of a zigbeeModel
>     fingerprint: [
>         {
>             // The model ID from: Device with modelID 'TS0601' is not supported
>             // You may need to add \u0000 at the end of the name in some cases
>             modelID: 'TS0601',
>             // The manufacturer name from: Device with modelID 'TS0601' is not supported.
>             manufacturerName: '_TZE204_7gclukjs',
>         },
>     ],
>     model: 'TS0601',
>     vendor: 'TuYa',
>     description: 'RadarTuya',
>     fromZigbee: [tuya.fz.datapoints],
>     toZigbee: [tuya.tz.datapoints],
>     onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
>     configure: tuya.configureMagicPacket,
>         fromZigbee: [tuya.fz.datapoints],
>         toZigbee: [tuya.tz.datapoints],
>         exposes: [
> 				e.enum('state', ea.STATE, ['none', 'Presence', 'Move and presence']).withDescription('state of taget'),
> 				e.presence(), 
> 				exposes.numeric('illuminance_lux', ea.STATE).withDescription('Illuminance').withUnit('lux'),
> 				exposes.numeric('current_distance', ea.STATE).withDescription('Distance to target').withUnit('cm'),
> 				exposes.numeric('delay_time', ea.STATE_SET).withValueMin(0).withValueMax(600).withValueStep(5).withDescription('Delay time').withUnit('s'),
> 				exposes.numeric('minimum_range', ea.STATE_SET).withValueMin(0).withValueMax(825).withValueStep(75).withDescription('Near Detection').withUnit('cm'),		
> 				exposes.numeric('maximum_range', ea.STATE_SET).withValueMin(75).withValueMax(900).withValueStep(75).withDescription('Far Detection').withUnit('cm'),
> 				exposes.numeric('motion_sensitivity', ea.STATE_SET).withDescription('Motion Sensitivity').withValueMin(0).withValueMax(10).withValueStep(1).withDescription('Motion Sensitivity'),		
>         ],
>         meta: {
>             tuyaDatapoints: [
> 				[104, 'presence', tuya.valueConverter.trueFalse1],
> 				[1, 'state', tuya.valueConverterBasic.lookup({'none': 0, 'Presence': 1, 'Move and presence': 2})],
> 				[103, 'illuminance_lux', tuya.valueConverter.raw],
> 				[9, 'current_distance', tuya.valueConverter.raw],
> 				[105, 'delay_time',   tuya.valueConverter.raw],
>                 [3, 'minimum_range', tuya.valueConverter.raw],
>                 [4, 'maximum_range', tuya.valueConverter.raw],
>                 [2, 'motion_sensitivity', tuya.valueConverter.raw],
> 				
>             ],
>     },
> };
> 
> module.exports = definition;
> ```

it works well. thanks

justicepng, 2024-03-23:
> I can see these values with tuya bridge on iot.tuya cloud
>
> {"1":"Induction state" ,"2":"Motion Sensitivity" ,"3":"Near Detection" ,"4":"Far Detection" ,"9":"Target distance" ,"102":"Presence [...]

I can't change any value like sensitivity.....

BartCardan, 2024-04-05:
I am trying to integrate exact this device. Is it already possible to add it with an external convertor? Or should I wait a bit? Could you please give me instructions how to?

smithbill17, 2024-04-05:
I've also just bought this device & trying to understand where I can find the latest appropriate external converter to try?

Vallefor, 2024-04-05:
This converter works fine for me. I used all from @ayasystems comment and changed `delay_time` to `delay`. Do not know why but `delay_time` everytime throw an error. May be it some kind of reserved property...

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_7gclukjs',
        },
    ],
    model: 'TS0601',
    vendor: 'TuYa',
    description: 'RadarTuya',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        exposes: [
				e.enum('state', ea.STATE, ['none', 'Presence', 'Move and presence']).withDescription('state of taget'),
				e.presence(), 
				exposes.numeric('illuminance_lux', ea.STATE).withDescription('Illuminance').withUnit('lux'),
				exposes.numeric('current_distance', ea.STATE).withDescription('Distance to target').withUnit('cm'),
				exposes.numeric('delay', ea.STATE_SET).withValueMin(0).withValueMax(600).withValueStep(1).withDescription('Delay time').withUnit('s'),
				exposes.numeric('minimum_range', ea.STATE_SET).withValueMin(0).withValueMax(825).withValueStep(75).withDescription('Near Detection').withUnit('cm'),		
				exposes.numeric('maximum_range', ea.STATE_SET).withValueMin(75).withValueMax(900).withValueStep(75).withDescription('Far Detection').withUnit('cm'),
				exposes.numeric('motion_sensitivity', ea.STATE_SET).withDescription('Motion Sensitivity').withValueMin(0).withValueMax(10).withValueStep(1).withDescription('Motion Sensitivity'),		
        ],
        meta: {
            tuyaDatapoints: [
				[104, 'presence', tuya.valueConverter.trueFalse1],
				[1, 'state', tuya.valueConverterBasic.lookup({'none': 0, 'Presence': 1, 'Move and presence': 2})],
				[103, 'illuminance_lux', tuya.valueConverter.raw],
				[9, 'current_distance', tuya.valueConverter.raw],
				[105, 'delay', tuya.valueConverter.raw],
                [3, 'minimum_range', tuya.valueConverter.raw],
                [4, 'maximum_range', tuya.valueConverter.raw],
                [2, 'motion_sensitivity', tuya.valueConverter.raw],
				
            ],
    },
};

module.exports = definition;
```

anushian, 2024-04-06:
> I just bought one of these devices on AliExpress & discovered that Zigbee2MQTT didn't recognise it. Google brought me here & using the details that the clever guys above posted, I've got it working now - thank you!
> 
> Just to help others that Google brings here, I simply copied the code ayasystems posted above pasted it into a file I named _TZE204_7gclukjs.js (I don't know if I needed to, but I removed the 'const extend' line mentioned in issue #22026 ), then saved the file in my zigbee2mqtt folder within HomeAssistant.
> 
> Then I went to the Zigbee2MQTT frontend & within the menu there I selected Settings > External converters and then added the _TZE204_7gclukjs.js file I saved earlier.
> 
> Restarted HomeAssistant & now my Presence sensor seems to be working fine 😁
> 
> I'm not sure about all the various available parameters as essentially I'm just using presence yes/no, but it works fine, so I'm happy.
> 
> Thank you guys 👍

Thanks for taking the time to write these helpful step by step instructions, my sensor appears to be working fine now 👍. Just to note. I did need to remove the 'const extend' line mentioned in issue #22026 for it to work (didnt work if I left it in). 

And thanks to all the other posters above for their work on getting this supported!

shalak, 2024-04-11:
This one appears to work correctly:

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_7gclukjs',
        },
    ],
    model: 'TS0601',
    vendor: 'TuYa',
    description: 'RadarTuya',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        exposes: [
				e.enum('state', ea.STATE, ['none', 'Presence', 'Move and presence']).withDescription('state of taget'),
				e.presence(),
				exposes.numeric('illuminance_lux', ea.STATE).withDescription('Illuminance').withUnit('lux'),
				exposes.numeric('current_distance', ea.STATE).withDescription('Distance to target').withUnit('cm'),
				exposes.numeric('delay', ea.STATE_SET).withValueMin(0).withValueMax(600).withValueStep(5).withDescription('Delay time').withUnit('s'),
				exposes.numeric('minimum_range', ea.STATE_SET).withValueMin(0).withValueMax(825).withValueStep(75).withDescription('Near Detection').withUnit('cm'),
				exposes.numeric('maximum_range', ea.STATE_SET).withValueMin(75).withValueMax(900).withValueStep(75).withDescription('Far Detection').withUnit('cm'),
				exposes.numeric('motion_sensitivity', ea.STATE_SET).withDescription('Motion Sensitivity').withValueMin(0).withValueMax(10).withValueStep(1).withDescription('Motion Sensitivity'),
        ],
        meta: {
            tuyaDatapoints: [
				[104, 'presence', tuya.valueConverter.trueFalse1],
				[1, 'state', tuya.valueConverterBasic.lookup({'none': 0, 'Presence': 1, 'Move and presence': 2})],
				[103, 'illuminance_lux', tuya.valueConverter.raw],
				[9, 'current_distance', tuya.valueConverter.raw],
				[105, 'delay',   tuya.valueConverter.raw],
                [3, 'minimum_range', tuya.valueConverter.raw],
                [4, 'maximum_range', tuya.valueConverter.raw],
                [2, 'motion_sensitivity', tuya.valueConverter.raw],

            ],
    },
};

module.exports = definition;
``` 
(i.e. without the `const extend = ...` and `delay_time` -> `delay`)

However, the problem is that it's triggering the presence sensor when nobody's around. I tried tuning the `motion_sensitivity` - both with low and high values, and it's still detecting motion. I have another sensor, but of [5GHz type](https://www.zigbee2mqtt.io/devices/ZY-M100-S_2.html#tuya-zy-m100-s_2) and that one works just fine.

May this be a case of a faulty unit, or is there something I can try to fix this?

ayasystems, 2024-04-11:
> I have same device. Upon first plugin and registering I got a whole lot of error messages in z2m. With initial instructions from [#21738 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/21738#issuecomment-2039818205) and file from [#21738 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/21738#issuecomment-2049511623) (I had to place the file within docker_data_dir/data) errors are gone now. But I only get linkquality messages and no motion. Also when I check the exposes tab in the device I only see "Empty exposes definition". Is this to be expected?

Did you set external_converter into yaml ?

[image]

heiko1988, 2024-04-11:
Thank you, this has been very helpful! 

I've been able to integrate z2m successfully and it's working quite well within iobroker. 
I'm running z2m separately without any issues. However, I'm encountering a single problem: 
the current_distance value consistently displays between 10-50cm, while my actual distance is around 300-400cm.

[image]

yarik-m, 2024-04-12:
Hi

yarik-m, 2024-04-12:
Does everyone's external converter fall off?

thombec, 2024-04-12:
Hello, I have the issue, that the vendor: _TZE204_7gclukjs is unknown at Homeassistant/zigbee2mqtt for zigbeemodel TS0601
I found your code here at this page to let the radar sensor run - how I have to implement it into zigbe2mqtt in homeassistant?
thank you in advance for help :-)

smithbill17, 2024-04-12:
> Hello, I have the issue, that the vendor: _TZE204_7gclukjs is unknown at Homeassistant/zigbee2mqtt for zigbeemodel TS0601 I found your code here at this page to let the radar sensor run - how I have to implement it into zigbe2mqtt in homeassistant? thank you in advance for help :-)

See my [comment](https://github.com/Koenkk/zigbee2mqtt/issues/21738#issuecomment-2039818205) above for basic instructions.

shalak, 2024-04-12:
> Hello, I have the issue, that the vendor: _TZE204_7gclukjs is unknown at Homeassistant/zigbee2mqtt for zigbeemodel TS0601 I found your code here at this page to let the radar sensor run - how I have to implement it into zigbe2mqtt in homeassistant? thank you in advance for help :-)

You need to:
1. Create a file with e.g. `_TZE204_7gclukjs.js` in `/share/zigbee2mqtt` folder in Home Assistant and paste the content there
2. Add the `_TZE204_7gclukjs.js` via the zigbee2mqtt Settings -> Settings -> External Converters -> "+"
3. Restart zigbee2mqtt via the Add-On panel

ayasystems, 2024-04-12:
> looks good, could you make a pull request?

I dont know the steps to do it. Could you do it?

Regards

casa3c, 2024-04-13:
Can I use this converter code for ZHA by adding the code as a new quirk file?

thombec, 2024-04-13:
After implementing the code above via "external converters" my device now shows the right parameters. Perfect - thank you! 

But a problem was discovered after adding the device. Battery-powered sensors nearby have lost connection to the server after start running the device. A large number of errors from this devices suddenly appear in the log from Zigbee2MQTT. 

one example of the errors:
[image]

After deleting of the device this problem was gone. It seems that the repeater functionality does not work properly. Did you also notice this behaviour?

opravdin, 2024-04-25:
For people struggling to make this thing work stable: I found this setup working good.
* Sensitivity 8/10
* Detection range 375 to 675 cm
* Delay 1s
It looks like value decoding is some way wrong, so these values don't make sense, but actually it works perfectly fine in 3*4 meters kitchen (both motion and presence detection). Distance detection is a mess (showing 20ish cm while I'm 2.5 meters away).

matte-a, 2024-04-28:
> For people struggling to make this thing work stable: I found this setup working good.
> 
> * Sensitivity 8/10
> * Detection range 375 to 675 cm
> * Delay 1s
> 
> It looks like value decoding is some way wrong, so these values don't make sense, but actually it works perfectly fine in 3*4 meters kitchen (both motion and presence detection). Distance detection is a mess (showing 20ish cm while I'm 2.5 meters away). UPD: I have 24Ghz version of this sensor

With your advices the sensor is extremely fast to react. I couldn't believe it. Thank you so much!

Aksenov3000, 2024-05-07:
> looks good, could you make a pull request?

Htllo! Where I need place file with this converter to make pull request ?

gabriel403, 2024-05-08:
I'm trying to get this working on my haos on my rpi5
I've got the file in the right place with the content from one of the [above comments](https://github.com/Koenkk/zigbee2mqtt/issues/21738#issuecomment-2039849157) , but it still says unsupported device
[image]

```
Warning 2024-05-08 15:04:54Device '0xa4c138522d992e20' with Zigbee model 'TS0601' and manufacturer name '_TZE204_7gclukjs' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
```

gabriel403, 2024-05-08:
https://github.com/Koenkk/zigbee2mqtt/issues/22083#issuecomment-2037342969
This fixed it for me, looks like an update to ha needed some tweaks

gabriel403, 2024-05-08:
I keep getting these errors now though

```
Error 2024-05-08 21:13:49Publish 'set' 'motion_sensitivity' to '0xa4c138522d992e20' failed: 'Error: ZCL command 0xa4c138522d992e20/1 manuSpecificTuya.dataRequest({"seq":1,"dpValues":[{"dp":2,"datatype":3,"data":[]}]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'
```
For sensitivity, delay, minimum distance everything etc

exurb1aTR, 2024-05-21:
> This one appears to work correctly:
> 
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const e = exposes.presets;
> const ea = exposes.access;
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> 
> const definition = {
>     // Since a lot of TuYa devices use the same modelID, but use different datapoints
>     // it's necessary to provide a fingerprint instead of a zigbeeModel
>     fingerprint: [
>         {
>             // The model ID from: Device with modelID 'TS0601' is not supported
>             // You may need to add \u0000 at the end of the name in some cases
>             modelID: 'TS0601',
>             // The manufacturer name from: Device with modelID 'TS0601' is not supported.
>             manufacturerName: '_TZE204_7gclukjs',
>         },
>     ],
>     model: 'TS0601',
>     vendor: 'TuYa',
>     description: 'RadarTuya',
>     fromZigbee: [tuya.fz.datapoints],
>     toZigbee: [tuya.tz.datapoints],
>     onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
>     configure: tuya.configureMagicPacket,
>         fromZigbee: [tuya.fz.datapoints],
>         toZigbee: [tuya.tz.datapoints],
>         exposes: [
> 				e.enum('state', ea.STATE, ['none', 'Presence', 'Move and presence']).withDescription('state of taget'),
> 				e.presence(),
> 				exposes.numeric('illuminance_lux', ea.STATE).withDescription('Illuminance').withUnit('lux'),
> 				exposes.numeric('current_distance', ea.STATE).withDescription('Distance to target').withUnit('cm'),
> 				exposes.numeric('delay', ea.STATE_SET).withValueMin(0).withValueMax(600).withValueStep(5).withDescription('Delay time').withUnit('s'),
> 				exposes.numeric('minimum_range', ea.STATE_SET).withValueMin(0).withValueMax(825).withValueStep(75).withDescription('Near Detection').withUnit('cm'),
> 				exposes.numeric('maximum_range', ea.STATE_SET).withValueMin(75).withValueMax(900).withValueStep(75).withDescription('Far Detection').withUnit('cm'),
> 				exposes.numeric('motion_sensitivity', ea.STATE_SET).withDescription('Motion Sensitivity').withValueMin(0).withValueMax(10).withValueStep(1).withDescription('Motion Sensitivity'),
>         ],
>         meta: {
>             tuyaDatapoints: [
> 				[104, 'presence', tuya.valueConverter.trueFalse1],
> 				[1, 'state', tuya.valueConverterBasic.lookup({'none': 0, 'Presence': 1, 'Move and presence': 2})],
> 				[103, 'illuminance_lux', tuya.valueConverter.raw],
> 				[9, 'current_distance', tuya.valueConverter.raw],
> 				[105, 'delay',   tuya.valueConverter.raw],
>                 [3, 'minimum_range', tuya.valueConverter.raw],
>                 [4, 'maximum_range', tuya.valueConverter.raw],
>                 [2, 'motion_sensitivity', tuya.valueConverter.raw],
> 
>             ],
>     },
> };
> 
> module.exports = definition;
> ```
> 
> (i.e. without the `const extend = ...` and `delay_time` -> `delay`)
> 
> However, the problem is that it's triggering the presence sensor when nobody's around. I tried tuning the `motion_sensitivity` - both with low and high values, and it's still detecting motion. I have another sensor, but of [5GHz type](https://www.zigbee2mqtt.io/devices/ZY-M100-S_2.html#tuya-zy-m100-s_2) and that one works just fine.
> 
> May this be a case of a faulty unit, or is there something I can try to fix this?

I just used the converter and the device is up and running. However here is the problem, as you say motion_sensitivity does not affect the way sensor works. When i check the state it seem there are way more options to tweak:

[image]

We need more data points!!!

pimpmytechno, 2024-06-03:
> This one appears to work correctly:
>
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes [...]

I encountered an issue with the accuracy of the "Current Distance" measurement. When standing 2 meters away from the sensor, it indicates a distance of 20 cm. Can this be fixed somehow? Otherwise, the converter works without any complaints.

TeHashX, 2024-06-14:
> I encountered an issue with the accuracy of the "Current Distance" measurement. When standing 2 meters away from the sensor, it indicates a distance of 20 cm. Can this be fixed somehow? Otherwise, [...]

Same here, usually stays at -3 cm and sometimes 20 cm but the biggest issue is with the log flooding every second...
This converter works only with Z2M Edge, on master branch will not detect motion.

Proncio, 2024-08-31:
Quick note: there are 2 versions of the sensor, in and on wall. Code above is related to In Wall, to be used for On Wall, manufacturer name should be set to _TZE204_ya4ft0w4 in place of _TZE204_7gclukjs - otherwise it's not recognized

garanda21, 2024-09-07:
> Quick note: there are 2 versions of the sensor, in and on wall. Code above is related to In Wall, to be used for On Wall, manufacturer name should be set to _TZE204_ya4ft0w4 in place of _TZE204_7gclukjs - otherwise it's not recognized

Just bought it the 24ghz On wall version but z2mqtt recognizes it as `TZE204_7gclukjs` and shows wrong image. On my case the device works it’s just recognizing it wrong

drupry, 2024-09-09:
I have the _TZE204_7gclukjs ceiling version, so for me, the image is correct. However, I also have an issue with the `Distance `value. The default is -0.3, and when someone is present, the values are quite random, ranging from 0.1 to 0.5 approx. I don’t know what this `Distance `value represents because it has no unit. Should it be meters?

Another problem is with the `Move sensitivity` and `Presence sensitivity` sliders. The values are integers from 0 to 10, but no matter what number I set, after a couple of seconds, they revert to 0. I don’t know if they work at all.

opravdin, 2024-09-10:
> Another problem is with the `Move sensitivity` and `Presence sensitivity` sliders. The values are integers from 0 to 10, but no matter what number I set, after a couple of seconds, they revert to 0. I don’t know if they work at all.

I had the same behavior. The only thing helped me is to power cycle the device. Change some parameters, reboot, fix wrong values and reboot again. Repeat until each value will be saved in a way you want it to. Also, I would highly recommend you a combination of values from my comment above. Got this thing working perfectly since then.

drupry, 2024-09-11:
>  Change some parameters, reboot, fix wrong values and reboot again. 

reboot how ? because every time I cut it from power, it sets random parameter to `detection distance min` and `detection distance max` and of course `presence` and `motion sensitivity` to 0

drupry, 2024-09-22:
I bought exactly the same sensor (ceiling 24 GHz version 80-220V from the same vendor from Ali) [_TZE204_7gclukjs](https://www.zigbee2mqtt.io/devices/ZY-M100-24GV2.html) and it pair itself as wall sensor [_TZE204_ya4ft0w4](https://www.zigbee2mqtt.io/devices/ZY-M100-24GV3.html). I tried to re-pair it several times but it always report itself as _ya4ft0w4. 
Now the fun part: it works flawless! `Distance` still has no units (meters) but it reports correct values and I have none of the above mentioned issues!!! 
Can someone fix this one (ceiling version) to work just perfect like the wall version [_TZE204_ya4ft0w4](https://www.zigbee2mqtt.io/devices/ZY-M100-24GV3.html) ?

Testingfor101, 2024-09-29:
With this configuration motion and presence sensitivity remains the same without going back to 0 or 1.
The original converter was using the incorrect value converter type.

```typescript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_7gclukjs',
        },
    ],
    model: 'ZY-M100-24GV2',
    vendor: 'Tuya',
    description: '24G MmWave radar human presence motion sensorEDITED',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.enum('state', ea.STATE, ['none', 'presence', 'move']).withDescription('Presence state sensor'),
        e.presence().withDescription('Occupancy'),
        e.numeric('distance', ea.STATE).withDescription('Target distance'),
        e.illuminance_lux().withDescription('Illuminance sensor'),
        e.numeric('move_sensitivity', ea.STATE_SET)
            .withValueMin(1).withValueMax(10).withValueStep(1)
            .withDescription('Motion Sensitivity'),
        e.numeric('presence_sensitivity', ea.STATE_SET)
            .withValueMin(1).withValueMax(10).withValueStep(1)
            .withDescription('Presence Sensitivity'),
        e.numeric('detection_distance_min', ea.STATE_SET)
            .withValueMin(0).withValueMax(8.25).withValueStep(0.75)
            .withUnit('m').withDescription('Minimum range'),
        e.numeric('detection_distance_max', ea.STATE_SET)
            .withValueMin(0.75).withValueMax(9.0).withValueStep(0.75)
            .withUnit('m').withDescription('Maximum range'),
        e.numeric('presence_timeout', ea.STATE_SET)
            .withValueMin(1).withValueMax(1500).withValueStep(1)
            .withUnit('s').withDescription('Fade time'),
    ],
    meta: {
        tuyaDatapoints: [
            [104, 'presence', tuya.valueConverter.trueFalse1],
            [2, 'move_sensitivity', tuya.valueConverter.raw],
            [102, 'presence_sensitivity', tuya.valueConverter.raw],
            [3, 'detection_distance_min', tuya.valueConverter.divideBy100],
            [4, 'detection_distance_max', tuya.valueConverter.divideBy100],
            [9, 'distance', tuya.valueConverter.divideBy100],
            [105, 'presence_timeout', tuya.valueConverter.raw],
            [103, 'illuminance_lux', tuya.valueConverter.raw],
            [1, 'state', tuya.valueConverterBasic.lookup({none: 0, presence: 1, move: 2})],
        ],
    },
};

module.exports = definition;

```

drupry, 2024-09-29:
> With this configuration motion and presence sensitivity remains the same without going back to 0 or 1.
The original converter was using the incorrect value converter type.

Oh yes, this is working! Thank you @Testingfor101!  I edited `distance` parametr to show unit (meters) and set divide parametr to 10 - now it's showing right values BUT!! the default value is still negative, in my case -0,3.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_7gclukjs',
        },
    ],
    model: 'ZY-M100-24GV2',
    vendor: 'Tuya',
    description: '24G MmWave radar human presence motion sensorEDITED',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.enum('state', ea.STATE, ['none', 'presence', 'move']).withDescription('Presence state sensor'),
        e.presence().withDescription('Occupancy'),
        e.numeric('distance', ea.STATE).withUnit('m').withDescription('Target distance'), 
        e.illuminance_lux().withDescription('Illuminance sensor'),
        e.numeric('move_sensitivity', ea.STATE_SET)
            .withValueMin(1).withValueMax(10).withValueStep(1)
            .withDescription('Motion Sensitivity'),
        e.numeric('presence_sensitivity', ea.STATE_SET)
            .withValueMin(1).withValueMax(10).withValueStep(1)
            .withDescription('Presence Sensitivity'),
        e.numeric('detection_distance_min', ea.STATE_SET)
            .withValueMin(0).withValueMax(8.25).withValueStep(0.75)
            .withUnit('m').withDescription('Minimum range'),
        e.numeric('detection_distance_max', ea.STATE_SET)
            .withValueMin(0.75).withValueMax(9.0).withValueStep(0.75)
            .withUnit('m').withDescription('Maximum range'),
        e.numeric('presence_timeout', ea.STATE_SET)
            .withValueMin(1).withValueMax(1500).withValueStep(1)
            .withUnit('s').withDescription('Fade time'),
    ],
    meta: {
        tuyaDatapoints: [
            [104, 'presence', tuya.valueConverter.trueFalse1],
            [2, 'move_sensitivity', tuya.valueConverter.raw],
            [102, 'presence_sensitivity', tuya.valueConverter.raw],
            [3, 'detection_distance_min', tuya.valueConverter.divideBy100],
            [4, 'detection_distance_max', tuya.valueConverter.divideBy100],
            [9, 'distance', tuya.valueConverter.divideBy10],
            [105, 'presence_timeout', tuya.valueConverter.raw],
            [103, 'illuminance_lux', tuya.valueConverter.raw],
            [1, 'state', tuya.valueConverterBasic.lookup({none: 0, presence: 1, move: 2})],
        ],
    },
};

module.exports = definition;

```

Bjk8kds, 2024-09-30:
Thanks alot of you ALL... Now my sensor working properly!

Koenkk, 2024-10-01:
Integrated the converter, will be available in todays release.

drupry, 2024-10-14:
After a few days, the `presence sensitivity` and `move sensitivity` settings reset themselves to 0 again. I’m not sure if these settings are actually working because I haven’t noticed any difference in behavior. So, I’m okay with it.

jkaberg, 2024-10-15:
> After a few days, the `presence sensitivity` and `move sensitivity` settings reset themselves to 0 again. I’m not sure if these settings are actually working because I haven’t noticed any difference in behavior. So, I’m okay with it.

I'm seeing this aswell, and this in turn results in the following log spam in/from Home Assistant

```
2024-10-15 07:34:08.140 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.inngang_bevegelsesensor_presence_sensitivity: 0 (range 1.0 - 10.0)
2024-10-15 07:34:08.140 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.inngang_bevegelsesensor_move_sensitivity: 0 (range 1.0 - 10.0)
2024-10-15 07:34:08.190 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.inngang_bevegelsesensor_presence_sensitivity: 0 (range 1.0 - 10.0)
2024-10-15 07:34:08.191 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.inngang_bevegelsesensor_move_sensitivity: 0 (range 1.0 - 10.0)
2024-10-15 07:34:08.236 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.inngang_bevegelsesensor_presence_sensitivity: 0 (range 1.0 - 10.0)
2024-10-15 07:34:08.236 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.inngang_bevegelsesensor_move_sensitivity: 0 (range 1.0 - 10.0)
```

Is this the intended behavior @Koenkk ?

Koenkk, 2024-10-16:
There are 2 errors here:
- The value resetting to `0`, this seems to be an issue of the device (can likely not be solved from z2m)
- The `Invalid value` error in HA -> this will be fixed in the next z2m release (1 November), at least for the `_TZE204_7gclukjs` (not sure if you have a different one)

semsem54, 2024-11-12:
guys im getting lost. can anyone please help me. I have bought this motion sensor from aliexpress and wanted to integrate to Z2MQTT but its not working. I have used the code from @Testingfor101 but could identify my device. Everything else is working but this device i cannot add. Im sure its just a minor dumb error from myside but I am a noob and need help :/

I have created the .js file and configured z2mqtt to start with an external converter in the settings. how can I get Z2MQTT to support my motion sensor?

[image]

> With this configuration motion and presence sensitivity remains the same without going back to 0 or 1. The original converter was using the incorrect value converter type.
> 
> ```ts
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const e = exposes.presets;
> const ea = exposes.access;
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> 
> const definition = {
>     fingerprint: [
>         {
>             // The model ID from: Device with modelID 'TS0601' is not supported
>             // You may need to add \u0000 at the end of the name in some cases
>             modelID: 'TS0601',
>             // The manufacturer name from: Device with modelID 'TS0601' is not supported.
>             manufacturerName: '_TZE204_7gclukjs',
>         },
>     ],
>     model: 'ZY-M100-24GV2',
>     vendor: 'Tuya',
>     description: '24G MmWave radar human presence motion sensorEDITED',
>     fromZigbee: [tuya.fz.datapoints],
>     toZigbee: [tuya.tz.datapoints],
>     configure: tuya.configureMagicPacket,
>     exposes: [
>         e.enum('state', ea.STATE, ['none', 'presence', 'move']).withDescription('Presence state sensor'),
>         e.presence().withDescription('Occupancy'),
>         e.numeric('distance', ea.STATE).withDescription('Target distance'),
>         e.illuminance_lux().withDescription('Illuminance sensor'),
>         e.numeric('move_sensitivity', ea.STATE_SET)
>             .withValueMin(1).withValueMax(10).withValueStep(1)
>             .withDescription('Motion Sensitivity'),
>         e.numeric('presence_sensitivity', ea.STATE_SET)
>             .withValueMin(1).withValueMax(10).withValueStep(1)
>             .withDescription('Presence Sensitivity'),
>         e.numeric('detection_distance_min', ea.STATE_SET)
>             .withValueMin(0).withValueMax(8.25).withValueStep(0.75)
>             .withUnit('m').withDescription('Minimum range'),
>         e.numeric('detection_distance_max', ea.STATE_SET)
>             .withValueMin(0.75).withValueMax(9.0).withValueStep(0.75)
>             .withUnit('m').withDescription('Maximum range'),
>         e.numeric('presence_timeout', ea.STATE_SET)
>             .withValueMin(1).withValueMax(1500).withValueStep(1)
>             .withUnit('s').withDescription('Fade time'),
>     ],
>     meta: {
>         tuyaDatapoints: [
>             [104, 'presence', tuya.valueConverter.trueFalse1],
>             [2, 'move_sensitivity', tuya.valueConverter.raw],
>             [102, 'presence_sensitivity', tuya.valueConverter.raw],
>             [3, 'detection_distance_min', tuya.valueConverter.divideBy100],
>             [4, 'detection_distance_max', tuya.valueConverter.divideBy100],
>             [9, 'distance', tuya.valueConverter.divideBy100],
>             [105, 'presence_timeout', tuya.valueConverter.raw],
>             [103, 'illuminance_lux', tuya.valueConverter.raw],
>             [1, 'state', tuya.valueConverterBasic.lookup({none: 0, presence: 1, move: 2})],
>         ],
>     },
> };
> 
> module.exports = definition;
> ```

semsem54, 2024-11-12:
after hours of trying i finally managed to add my device and I see some values. I used the "ZY-M100-24GV3" from the converter of koenkk and this works best so far for my device. but the value for "target distance" is not korrekt (output is something around 30m). Presence sensitivity seems like an on/off switch and not a value...
TS0601_TZE200_clrdrnya 24G Motion Sensor
maybe someone as an idea how to configure properly?

closed 2025-01-19 by github-actions[bot]