Koenkk/zigbee2mqtt #21741: [New device support]:  Gas sensor (methane) by EKF (TS0601_TZE200_nus5kk3n)

onmobs, 2024-03-10:
### Link

https://ekfgroup.com/ru/catalog/products/umnyj-datchik-gaza-zigbee

### Database entry

{"id":11,"type":"Router","ieeeAddr":"0x142d41fffe50a2a9","nwkAddr":32885,"manufId":4098,"manufName":"_TZE200_nus5kk3n","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"NG-\u0012OG-\u0012qG-\u0012rG-\u0012\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0007\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0012","65506":31,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_nus5kk3n","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359}

### Comments

Tuya Gas detector analog, works with tuya gas sensor config that I copied to external converter file. Maybe I did something wrong, it is my first time creating custom converters, but it works - expose "gas" triggers "true" when sensing gas leak.

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_nus5kk3n',
        },
    ],
        model: 'TS0601_TZE200_nus5kk3n',
        vendor: 'TuYa',
        description: 'Gas sensor',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        configure: tuya.configureMagicPacket,
        exposes: [e.gas(), tuya.exposes.selfTest(), tuya.exposes.selfTestResult(), tuya.exposes.faultAlarm(), tuya.exposes.silence()],
        meta: {
            tuyaDatapoints: [
                [1, 'gas', tuya.valueConverter.trueFalse0],
                [9, 'self_test_result', tuya.valueConverter.selfTestResult],
                [11, 'fault_alarm', tuya.valueConverter.trueFalse1],
        ],
    },
};

module.exports = definition;
```

Koenkk, 2024-03-10:
Could you try with the following [external converter](https://gist.github.com/Koenkk/8cccc01489086697e7cc70b5e4419c83)?

onmobs, 2024-03-10:
I have already tried this converter before, copied it from tuya.ts, only 'gas' and 'self_test_result' (physical button press) exposes are working, maybe 'fault_alarm' too (not sure), other datapoints returns nothing.

Koenkk, 2024-03-11:
No problem, I've added your converter.

closed 2024-03-11 by Koenkk

onmobs, 2024-04-09:
I had to reinstall z2m and rejoin all devices due to strange crash when pressing "permit join" button and I noticed some errors when pairing gas sensors (but they are working as they should):

/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_3c24cd1445b6ed118ccae00ea8669f5d-if00-port0
Zigbee2MQTT:error 2024-04-09 13:07:36: Failed to configure 'smoke sensor2', attempt 1 (Error: ZCL command 0x0c4314fffe727586/1 genPowerCfg.configReport([{"minimumReportInterval":3600,"maximumReportInterval":65000,"reportableChange":10,"attribute":"batteryPercentageRemaining"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 390 - 1 - 184 - 1 - 7 after 10000ms)
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:499:47)
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:760:28)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:561:9)
    at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:69:13)
    at result.configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:287:17)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:162:21)
    at Configure.configure (/app/lib/extension/configure.ts:122:13))

Failed to configure 'smoke sensor3', attempt 2 (Error: ZCL command 0x0c4314fffe727586/1 genPowerCfg.configReport([{"minimumReportInterval":3600,"maximumReportInterval":65000,"reportableChange":10,"attribute":"batteryPercentageRemaining"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 390 - 1 - 57 - 1 - 7 after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:499:47) at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:760:28) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:561:9) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:69:13) at result.configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:287:17) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:162:21) at Configure.configure (/app/lib/extension/configure.ts:122:13))

onmobs, 2024-04-09:
I updated z2m to 1.36.1 but gas detector still unsupported when I remove external converter and trying to rejoin it.

Koenkk, 2024-04-10:
Found the issue, fixed it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

onmobs, 2024-04-11:
Thank you, but I`ll wait for stable branch)