Koenkk/zigbee2mqtt #21755: [New device support]: TRADFRI bulb E27 CWS globe 806lm 205.475.32 LED2109G6

millionsofjeffries, 2024-03-10:
### Link

https://www.ikea.com/gb/en/search/?q=20547532

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0x3c2ef5fffeeee338","nwkAddr":21346,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E27 CWS globe 806lm","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,64636],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI bulb E27 CWS globe 806lm","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":8,"appVersion":1,"stackVersion":113,"hwVersion":1,"dateCode":"20230913","swBuildId":"1.0.38"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":250,"colorTempPhysicalMax":454}},"genOnOff":{"attributes":{"startUpOnOff":255,"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":113,"hwVersion":1,"dateCode":"20230913","swBuildId":"1.0.38","zclVersion":8,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1710093360976}

### Comments

Purchased yesterday in the UK - Ikea article number 205.475.32.  It's not on their website though.
Everything appears to be working fine in z2m but bulb says it's unsupported.
Many thanks!!
[image]

### External definition

```shell
const {light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TRADFRI bulb E27 CWS globe 806lm'],
    model: 'TRADFRI bulb E27 CWS globe 806lm',
    vendor: 'IKEA of Sweden',
    description: 'Automatically generated definition',
    extend: [light({"colorTemp":{"range":[250,454]},"color":{"modes":["xy","hs"],"enhancedHue":true}})],
    meta: {},
};

module.exports = definition;
```

kremik, 2024-03-10:
Yes, please. Yesterday purchased in CZ and is reported as unsupported.

Koenkk, 2024-03-11:
added, thanks!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-03-11 by Koenkk

marazmarci, 2024-03-15:
Did you notice any differences compared to the old version (LED1924G9: [IKEA](https://www.ikea.com/gb/en/p/tradfri-led-bulb-e27-806-lumen-smart-wireless-dimmable-colour-and-white-spectrum-globe-50517637/), [Z2M](https://www.zigbee2mqtt.io/devices/LED1924G9.html))? So far I only know that this new one has an EU energy rating of `E` compared to the old one having just an `F`. All light-related parameters are the same: kelvins, lumens, CRI, and rated lifetime.

@millionsofjeffries, could you please check if the following issue is still present on this new bulb? [[Problem]: Can't trigger effect on IKEA LED1924G9 #10572](https://github.com/Koenkk/zigbee2mqtt/issues/10572)

And could you also please check the firmware version?

I'm asking these because I don't (yet) own one.

millionsofjeffries, 2024-03-15:
Can confirm
Firmware build date
20230913
Firmware version
1.0.38
Only the energy rating seems to have changed. 
Bulb does the effects e.g. Blink,  breathe
Thanks!!

GioMez, 2024-04-23:
> Did you notice any differences compared to the old version (LED1924G9: [...]

I bought one of the LED2109G6 and I can confirm that the colors are far better than the previous model! Now the entire color spectrum is completely saturated (especially greens and indigos, which were more similar to lime and cold white, respectively).

ireun, 2024-05-23:
@Koenkk I don't want to create another issue but I'm receiving this warning: 

`Images currently unavailable for device 'TRADFRI bulb E27 CWS globe 806lm', hardwareVersion '1', manufacturerName IKEA of Sweden, {"fieldControl":1,"manufacturerCode":4476,"imageType":10245,"fileVersion":16777272}'`

What can be done to solve it? I see that there is an image in the repository https://github.com/Koenkk/zigbee2mqtt.io/blob/master/public/images/devices/LED2109G6.png but it's not correct. See at https://www.ikea.com/pl/pl/p/tradfri-zarowka-led-e27-806-lumenow-bezprzewodowy-przyciemniany-barwne-i-biale-spektrum-kula-opalowa-biel-30547471/

[image]

I can create an image without background, but I don't know is this is needed.

marazmarci, 2024-05-23:
@ireun, I think here in this context, _images_ mean _OTA firmware images_ 😄

ireun, 2024-05-24:
Ohhhh! Now I feel stupid :D Thanks! I've also seen that the 'picture-image' has been changed in #2761 :)

nonyhaha, 2024-05-25:
Sorry to bring this thread back up but i am not sure where the issue is originating. I just bought 2 of these bulbs, but after pairing, there is no way ro control them through an entity light.bulb*address. There are no entity controls for the bulb. I can control it from z2m bulb page but the light.* Entity does not appear in  home assistant. 

Any ideas where i should look into this?

YannickvanVugt, 2024-06-19:
Based on the manual of the bulb, it should have a way broader spectrum of white tones. 

In the generated definition, the range is 250 to 454 mireds:
> ```shell
> extend: [light({"colorTemp":{"range":[250,454]},"color":{"modes":["xy","hs"],"enhancedHue":true}})]
> ```

This is what the manual states:
[image]

Based on my calculations the range should be from 166/167 to 561/562.
Cold sky: 1.000.000 / 6000 = 166,66666666666666666666666666667
Candle light: 1.000.000 / 1780 = 561,79775280898876404494382022472

@Koenkk Is that something that is not detected properly by Zigbee2MQTT when the external definition is generated? If you change these values, will that work?