Koenkk/zigbee2mqtt #21756: Service won't start

Momro, 2024-03-10:
### What happened?

I'm trying to start z2m via a service, but it doesn't work.

It works via "normal" start:

```
# /usr/bin/node index.js
Zigbee2MQTT:info  2024-03-10 21:48:10: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2024-03-10.21-48-10' filename: log.txt
Zigbee2MQTT:info  2024-03-10 21:48:11: Starting Zigbee2MQTT version 1.35.3 (commit #8b694432)
Zigbee2MQTT:info  2024-03-10 21:48:11: Starting zigbee-herdsman (0.33.8)
Zigbee2MQTT:info  2024-03-10 21:48:17: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2024-03-10 21:48:17: Coordinator firmware version: '{"meta":{"maintrel":"3 ","majorrel":"6","minorrel":"10","product":8,"revision":"6.10.3.0 build 297"},"type":"EZSP v8"}'
Zigbee2MQTT:info  2024-03-10 21:48:17: Currently 0 devices are joined:
Zigbee2MQTT:warn  2024-03-10 21:48:17: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2024-03-10 21:48:17: Allowing new devices to join.
Zigbee2MQTT:warn  2024-03-10 21:48:17: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2024-03-10 21:48:17: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2024-03-10 21:48:17: Connecting to MQTT server at mqtt://192.168.1.100
Zigbee2MQTT:info  2024-03-10 21:48:17: Connected to MQTT server
Zigbee2MQTT:info  2024-03-10 21:48:17: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
Zigbee2MQTT:info  2024-03-10 21:48:18: Started frontend on port 80
```

### What did you expect to happen?

As posted here, https://www.zigbee2mqtt.io/guide/installation/01_linux.html#optional-running-as-a-daemon-with-systemctl, the daemon should start :-(

### How to reproduce it (minimal and precise)

I start the service like in the description:

```
# sudo systemctl start zigbee2mqtt
Job for zigbee2mqtt.service failed because a timeout was exceeded.
See "systemctl status zigbee2mqtt.service" and "journalctl -xe" for details.
```

Config/services file looks like this:
```
# cat /etc/systemd/system/zigbee2mqtt.service
[Unit]
Description=zigbee2mqtt
After=network.target

[Service]
Environment=NODE_ENV=production
Type=notify
ExecStart=/usr/bin/node index.js
WorkingDirectory=/opt/zigbee2mqtt
StandardOutput=null
# Or use StandardOutput=null if you don't want Zigbee2MQTT messages filling syslog, for more options see systemd.exec(5)
StandardError=inherit
WatchdogSec=10s
Restart=always
RestartSec=10s
User=z2m # my user is called z2m, not "pi"
AmbientCapabilities = CAP_NET_BIND_SERVICE # need this to run on port 80

[Install]
WantedBy=multi-user.target
```

### Zigbee2MQTT version

1.35.3

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF Zigbee Gateway, ZBDongle-E 3.0, CC2652(?)

### Setup

plain on a RPi3

Momro, 2024-03-10:
I tried my very best to answer the "default" questions, but I'm quite new to this :-( I hope I made no mistakes :-(

N3rdix, 2024-03-11:
What's the output of `systemctl status zigbee2mqtt.service`?

Momro, 2024-03-12:
```
# systemctl status zigbee2mqtt.service
● zigbee2mqtt.service - zigbee2mqtt
     Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; disabled; vendor preset: enabled)
     Active: inactive (dead)
```

Momro, 2024-03-12:
hm, I did `sudo systemctl enable zigbee2mqtt.service` once more and now the service is running ... odd ...

Sorry for the inconvenience!

closed 2024-03-12 by Momro