Koenkk/zigbee2mqtt #21758: YSRSAI YSR-MINI-01_rgbcct being identified as Lonsonho ZB-RGBCW

timzinfandel, 2024-03-11:
### What happened?

I'm pairing four strip light controllers from Ysrsai and they are being mistakenly identified as Lonsonho lightbulbs. On/off and color change commands aren't working.

### What did you expect to happen?

I expected them to pair as YSRSAI YSR-MINI-01_rgbcct being identified as Lonsonho ZB-RGBCW (a supported device)

### Zigbee2MQTT version

1.36.0 commit: unknown

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8

### Setup

VIrtualbox on a Macbook Pro

dotWee, 2024-05-11:
See my workaround fix here: https://github.com/Koenkk/zigbee2mqtt/issues/21361#issuecomment-2105394183

thinkbig1979, 2024-06-26:
Is there some way to force a re-interrogation to get it to be recognized as the correct device again without resorting to custom scripts? 
I've tried the reconfigure and the interview commands, and I've tried removing and joining it again, but it keeps getting recognized incorrectly.

dotWee, 2024-07-04:
Using a custom script basically overrides the recognition signature, therefore it's possible the wrong signature to-be-overridden is used.

I can try to put some code together, if my linked workaround won't work.

Is your device still being recognised as Lonsonho ZB-RGBCW?

thinkbig1979, 2024-07-05:
Hi @dotWee, my issue us the `YSR-MINI-01_rgbcct` being recognised incorrectly as a `YSR-MINI-01_wwcw`.

thinkbig1979, 2024-11-13:
Morning all, curious to know if the cause of this mis-identification is known. I've just run the latest set of updates, then removed and rejoined the device, and the devive again gets recognised as a YSR-MINI-01_wwcw, while before it was correctly recognised as  the rgbcct variant, and was working as such.

FWIW, below is the log data of the interview process for this device:

```bash

0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/bridge/request/device/interview' with data '{"id":"0x00124b001d401397","transaction":"gvyjl-6"}'
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:controller:device: Interview - start device '0x00124b001d401397'
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:controller:device: Interview - skip node descriptor request for '0x00124b001d401397', already got it
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:controller:device: Interview - Detected potential TuYa end device, reading modelID and manufacturerName...
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:controller:endpoint: ZCL command 0x00124b001d401397/1 genBasic.read(["modelId","manufacturerName"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false,"sendPolicy":"immediate"})
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x00124b001d401397:16794/1 (0,0,1)
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":16794,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":176,"options":0,"radius":30,"len":7,"data":{"type":"Buffer","data":[16,102,0,5,0,4,0]}}
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:zstack:unpi:writer: --> frame [254,17,36,1,154,65,1,1,0,0,176,0,30,7,16,102,0,5,0,4,0,49]
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:zstack:unpi:parser: --- parseNext []
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,176,118]
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,176,118]
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,176] - 118
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":176}
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:zstack:unpi:parser: --- parseNext []
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:zstack:unpi:parser: <-- [254,3,69,196,154,65,0,89]
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,69,196,154,65,0,89]
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 5 - 196 - [154,65,0] - 89
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:zstack:znp: AREQ: <-- ZDO - srcRtgInd - {"dstaddr":16794,"relaycount":0,"relaylist":[]}
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:zstack:unpi:parser: --- parseNext []
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:zstack:unpi:parser: <-- [254,46,68,129,0,0,0,0,154,65,1,1,0,63,0,238,151,155,0,0,26,24,102,1,5,0,0,66,7,90,66,45,67,84,48,49,4,0,0,66,6,89,83,82,83,65,73,154,65,29,110]
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:zstack:unpi:parser: --- parseNext [254,46,68,129,0,0,0,0,154,65,1,1,0,63,0,238,151,155,0,0,26,24,102,1,5,0,0,66,7,90,66,45,67,84,48,49,4,0,0,66,6,89,83,82,83,65,73,154,65,29,110]
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:zstack:unpi:parser: --> parsed 46 - 2 - 4 - 129 - [0,0,0,0,154,65,1,1,0,63,0,238,151,155,0,0,26,24,102,1,5,0,0,66,7,90,66,45,67,84,48,49,4,0,0,66,6,89,83,82,83,65,73,154,65,29] - 110
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":16794,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":63,"securityuse":0,"timestamp":10196974,"transseqnumber":0,"len":26,"data":{"type":"Buffer","data":[24,102,1,5,0,0,66,7,90,66,45,67,84,48,49,4,0,0,66,6,89,83,82,83,65,73]}}
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:controller: Received payload: clusterID=0, address=16794, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=63, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":102,"commandIdentifier":1},"payload":[{"attrId":5,"status":0,"dataType":66,"attrData":"ZB-CT01"},{"attrId":4,"status":0,"dataType":66,"attrData":"YSRSAI"}],"command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}}
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:zstack:unpi:parser: --- parseNext []
[... cut ...]
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:controller:device: Interview - got simple descriptor for endpoint '13' device '0x00124b001d401397'
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:controller:device: Interview - completed for device '0x00124b001d401397'
0|zigbee2mqtt  | [2024-11-13 10:14:39] debug: 	zh:controller:database: Writing database to '/opt/zigbee2mqtt/data/database.db'
0|zigbee2mqtt  | [2024-11-13 10:14:40] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/interview', payload '{"data":{"id":"0x00124b001d401397"},"status":"ok","transaction":"gvyjl-6"}'
```

thinkbig1979, 2025-01-13:
Just upgraded tp 2.0 and re-joined the device....unfortunately it's still being recognized incorrectly.

closed 2025-03-22 by github-actions[bot]