Koenkk/zigbee2mqtt #21761: TS0505B is not properly working

CvH, 2024-03-11:
### What happened?

After joining a TS0505B_1 device (FUT037Z+) the device is not controllable. Also tried the recent dev branch without any luck.
I was able, somehow, to change a colour and the brightness directly after I reset and rejoining the device x times (2-3 seconds window). I am unsure how I managed that and can't reproduce it. Device works at ZHA reasoning well (with basically no features).

Similar sounding errors (with different errors) are for example https://github.com/Koenkk/zigbee2mqtt/issues/17956  https://github.com/Koenkk/zigbee2mqtt/issues/19655

### What did you expect to happen?

It is controllable like every other device.

### How to reproduce it (minimal and precise)

Join a TS0505B_1 device and try to control anything.

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

7.1.4.0 build 389

### Adapter

SkyConnect v1.0

### Setup

HASS OS at X86_64

### Debug log

startup 
```
Zigbee2MQTT:debug 2024-03-11 11:53:26: Received MQTT message on 'homeassistant/sensor/0xa4c13833576b2b4d/color_power_on_behavior/config' with data ''
Zigbee2MQTT:debug 2024-03-11 11:53:26: Received MQTT message on 'homeassistant/sensor/0xa4c13833576b2b4d/linkquality/config' with data ''
Zigbee2MQTT:debug 2024-03-11 11:55:44: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"61fxe-2","value":true}'
Zigbee2MQTT:info  2024-03-11 11:55:44: Zigbee: allowing new devices to join.
Zigbee2MQTT:debug 2024-03-11 11:55:44: Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"data":[25,133,2,11,254,0],"type":"Buffer"}' from endpoint 242 with groupID null, ignoring since it is from coordinator
Zigbee2MQTT:info  2024-03-11 11:55:45: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"61fxe-2"}'
Zigbee2MQTT:warn  2024-03-11 11:56:03: Device '0xa4c13833576b2b4d' left the network
Zigbee2MQTT:info  2024-03-11 11:56:03: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13833576b2b4d","ieee_address":"0xa4c13833576b2b4d"},"type":"device_leave"}'
Zigbee2MQTT:debug 2024-03-11 11:56:57: Saving state to file /config/zigbee2mqtt/state.json
```

start to join
```
Zigbee2MQTT:info  2024-03-11 11:57:36: Device '0xa4c13833576b2b4d' joined
Zigbee2MQTT:info  2024-03-11 11:57:36: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13833576b2b4d","ieee_address":"0xa4c13833576b2b4d"},"type":"device_joined"}'
Zigbee2MQTT:info  2024-03-11 11:57:36: Starting interview of '0xa4c13833576b2b4d'
Zigbee2MQTT:info  2024-03-11 11:57:36: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13833576b2b4d","ieee_address":"0xa4c13833576b2b4d","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:debug 2024-03-11 11:57:36: Received Zigbee message from '0xa4c13833576b2b4d', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-03-11 11:57:36: Skipping message, still interviewing
Zigbee2MQTT:debug 2024-03-11 11:57:36: Received Zigbee message from '0xa4c13833576b2b4d', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-03-11 11:57:36: Skipping message, still interviewing
Zigbee2MQTT:debug 2024-03-11 11:57:36: Received Zigbee message from '0xa4c13833576b2b4d', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-03-11 11:57:36: Skipping message, still interviewing
Zigbee2MQTT:debug 2024-03-11 11:57:36: Received Zigbee message from '0xa4c13833576b2b4d', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-03-11 11:57:36: Skipping message, still interviewing
Zigbee2MQTT:debug 2024-03-11 11:57:36: Received Zigbee message from '0xa4c13833576b2b4d', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-03-11 11:57:36: Skipping message, still interviewing
Zigbee2MQTT:debug 2024-03-11 11:57:37: Received Zigbee message from '0xa4c13833576b2b4d', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-03-11 11:57:37: Skipping message, still interviewing
Zigbee2MQTT:info  2024-03-11 11:57:37: Successfully interviewed '0xa4c13833576b2b4d', device has successfully been paired
Zigbee2MQTT:info  2024-03-11 11:57:37: Device '0xa4c13833576b2b4d' is supported, identified as: TuYa Zigbee RGB+CCT light (TS0505B_1)
Zigbee2MQTT:info  2024-03-11 11:57:37: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Zigbee RGB+CCT light","exposes":[{"features":[{"access":7,"description":"On/off state of this light","label":"State","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"Brightness of this light","label":"Brightness","name":"brightness","property":"brightness","type":"numeric","value_max":254,"value_min":0},{"access":7,"description":"Color temperature of this light","label":"Color temp","name":"color_temp","presets":[{"description":"Coolest temperature supported","name":"coolest","value":153},{"description":"Cool temperature (250 mireds / 4000 Kelvin)","name":"cool","value":250},{"description":"Neutral temperature (370 mireds / 2700 Kelvin)","name":"neutral","value":370},{"description":"Warm temperature (454 mireds / 2200 Kelvin)","name":"warm","value":454},{"description":"Warmest temperature supported","name":"warmest","value":500}],"property":"color_temp","type":"numeric","unit":"mired","value_max":500,"value_min":153},{"access":7,"description":"Color of this light in the CIE 1931 color space (x/y)","features":[{"access":7,"label":"X","name":"x","property":"x","type":"numeric"},{"access":7,"label":"Y","name":"y","property":"y","type":"numeric"}],"label":"Color (X/Y)","name":"color_xy","property":"color","type":"composite"}],"type":"light"},{"access":2,"description":"Triggers an effect on the light (e.g. make light blink for a few seconds)","label":"Effect","name":"effect","property":"effect","type":"enum","values":["blink","breathe","okay","channel_change","finish_effect","stop_effect"]},{"access":3,"description":"Do not disturb mode, when enabled this function will keep the light OFF after a power outage","label":"Do not disturb","name":"do_not_disturb","property":"do_not_disturb","type":"binary","value_off":false,"value_on":true},{"access":3,"description":"Power on behavior state","label":"Color power on behavior","name":"color_power_on_behavior","property":"color_power_on_behavior","type":"enum","values":["initial","previous","cutomized"]},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0505B_1","options":[{"access":2,"description":"Controls the transition time (in seconds) of on/off, brightness, color temperature (if applicable) and color (if applicable) changes. Defaults to `0` (no transition).","label":"Transition","name":"transition","property":"transition","type":"numeric","value_min":0},{"access":2,"description":"When enabled colors will be synced, e.g. if the light supports both color x/y and color temperature a conversion from color x/y to color temperature will be done when setting the x/y color (default true).","label":"Color sync","name":"color_sync","property":"color_sync","type":"binary","value_off":false,"value_on":true},{"access":2,"description":"State actions will also be published as 'action' when true (default false).","label":"State action","name":"state_action","property":"state_action","type":"binary","value_off":false,"value_on":true}],"supports_ota":false,"vendor":"TuYa"},"friendly_name":"0xa4c13833576b2b4d","ieee_address":"0xa4c13833576b2b4d","status":"successful","supported":true},"type":"device_interview"}'
Zigbee2MQTT:info  2024-03-11 11:57:37: Configuring '0xa4c13833576b2b4d'
Zigbee2MQTT:info  2024-03-11 11:57:37: MQTT publish: topic 'homeassistant/light/0xa4c13833576b2b4d/light/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}" [... cut ...]
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:1207:32)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:259:36)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)
Zigbee2MQTT:debug 2024-03-11 11:59:59: Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"data":[25,139,2,10,0,0],"type":"Buffer"}' from endpoint 242 with groupID null, ignoring since it is from coordinator
```

radaoggy, 2024-04-30:
The same is happening to me. After a few seconds it disconnects and reconnects. It does not respond to most requests. And it identifies as a different device: #21659.

I can't find the model without "+" in stock almost anywhere here. It looks like some kind of problem with this specific "+" model. Does anyone know how to fix it if is it possible?

xiduis15, 2024-11-26:
I have the same problem, I just paired my fut037z+ led driver, but z2m recognizes it as a tuya TS0505B device.
The miboxer device seems to be supported: https://www.zigbee2mqtt.io/devices/FUT037Z_.html
I tried to install the latest edge version of z2m, but the device is still recognized as a tuya.

dawiinci, 2024-12-25:
I do have issues with that device as well. If I set the device brightness to 100 all the colors are set to 100. 

I expect it to keep the colors. It does keep colors if the light is already on.

dawiinci, 2024-12-27:
> I have the same problem, I just paired my fut037z+ led driver, but z2m recognizes it as a tuya TS0505B device.
>
> The miboxer device seems to be supported: [...]

Interestingly in Home Assistant the device is recognized correctly over MQTT. But it still fails to maintain the color after switched off. The effects aren't working either.

closed 2025-04-08 by github-actions[bot]