Koenkk/zigbee2mqtt #21763: After 2 min, unable to send commands to zigbee devices

fernando155, 2024-03-11:
### What happened?

So from this morning (as far as I can know), I not able to communicate with zigbee devices on the network. Its been working fine, but everything is gone now. When you restart the docker inside homeassistant core, everthing works ok, but 2 min later, it just stops. I attached logs where you can see is working at the start and then just sends errors. Thank you for any help

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

20210708

### Adapter

zStack3x0 (sonoff)

### Setup

addon on homeassistant core

### Debug log

2024-03-11T14:40:30.905Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x70b3d52b6001be28:31631/1 (0,0,3)
2024-03-11T14:40:30.920Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2024-03-11T14:40:30.920Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2024-03-11T14:40:30.920Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2024-03-11T14:40:30.920Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2024-03-11T14:40:30.920Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2024-03-11T14:40:30.920Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":4629,"destendpoint":1,"srcendpoint":1,"clusterid":2820,"transid":19,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[16,13,0,5,5,8,5,11,5]}}
2024-03-11T14:40:30.920Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,19,36,1,21,18,1,1,4,11,19,0,30,9,16,13,0,5,5,8,5,11,5,36]
2024-03-11T14:40:31.159Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100,254,3,69,196,215,160,0,245,254,3,69,196,252,155,0,229]
2024-03-11T14:40:31.159Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100,254,3,69,196,215,160,0,245,254,3,69,196,252,155,0,229]
2024-03-11T14:40:31.159Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2024-03-11T14:40:31.159Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2024-03-11T14:40:31.159Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,215,160,0,245,254,3,69,196,252,155,0,229]
2024-03-11T14:40:31.159Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [215,160,0] - 245
2024-03-11T14:40:31.159Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":41175,"relaycount":0,"relaylist":[]}
2024-03-11T14:40:31.159Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,252,155,0,229]
2024-03-11T14:40:31.159Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [252,155,0] - 229
2024-03-11T14:40:31.159Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":39932,"relaycount":0,"relaylist":[]}
2024-03-11T14:40:31.159Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2024-03-11T14:40:31.159Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":31631,"destendpoint":1,"srcendpoint":1,"clusterid":2820,"transid":20,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[16,14,0,5,5,8,5,11,5]}}
2024-03-11T14:40:31.159Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,19,36,1,143,123,1,1,4,11,20,0,30,9,16,14,0,5,5,8,5,11,5,211]
2024-03-11T14:40:31.162Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,19,213]
2024-03-11T14:40:31.162Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,19,213]
2024-03-11T14:40:31.162Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,19] - 213
2024-03-11T14:40:31.162Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":19}
2024-03-11T14:40:31.162Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2024-03-11T14:40:31.177Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2024-03-11T14:40:31.177Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2024-03-11T14:40:31.177Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2024-03-11T14:40:31.177Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2024-03-11T14:40:31.177Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2024-03-11T14:40:31.199Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,21,18,1,153,128,155,254,3,68,128,0,1,20,210]
2024-03-11T14:40:31.199Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,21,18,1,153,128,155,254,3,68,128,0,1,20,210]
2024-03-11T14:40:31.199Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [21,18,1,153,128] - 155
2024-03-11T14:40:31.199Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":4629,"relaycount":1,"relaylist":[32921]}
2024-03-11T14:40:31.199Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,20,210]
2024-03-11T14:40:31.199Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,20] - 210
2024-03-11T14:40:31.199Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":20}
2024-03-11T14:40:31.199Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2024-03-11T14:40:31.260Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,41,68,129,0,0,4,11,21,18,1,1,0,75,0,40,189,66,0,0,21,24,13,1,5,5,0,33,239,0,8,5,0,33,16,0,11,5,0,33,0,0,153,128,28,161]
2024-03-11T14:40:31.260Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,41,68,129,0,0,4,11,21,18,1,1,0,75,0,40,189,66,0,0,21,24,13,1,5,5,0,33,239,0,8,5,0,33,16,0,11,5,0,33,0,0,153,128,28,161]
2024-03-11T14:40:31.260Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 41 - 2 - 4 - 129 - [0,0,4,11,21,18,1,1,0,75,0,40,189,66,0,0,21,24,13,1,5,5,0,33,239,0,8,5,0,33,16,0,11,5,0,33,0,0,153,128,28] - 161
2024-03-11T14:40:31.260Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":2820,"srcaddr":4629,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":75,"securityuse":0,"timestamp":4373800,"transseqnumber":0,"len":21,"data":{"type":"Buffer","data":[24,13,1,5,5,0,33,239,0,8,5,0,33,16,0,11,5,0,33,0,0]}}
2024-03-11T14:40:31.262Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":13,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":1285,"status":0,"dataType":33,"attrData":239},{"attrId":1288,"status":0,"dataType":33,"attrData":16},{"attrId":1291,"status":0,"dataType":33,"attrData":0}],"Command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}},"address":4629,"endpoint":1,"linkquality":75,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2024-03-11T14:40:31.263Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:debug 2024-03-11 14:40:31: Received Zigbee message from 'AirFryer', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":16,"rmsVoltage":239}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2024-03-11 14:40:31: MQTT publish: topic 'zigbee2mqtt/AirFryer', payload '{"child_lock":null,"current":0.02,"energy":40.97,"indicator_mode":null,"linkquality":75,"power":0,"power_outage_memory":null,"state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":239}'
2024-03-11T14:40:31.269Z zigbee-herdsman:controller:endpoint Read 0x70b3d52b6001bf24/1 seMetering(["currentSummDelivered"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
[... 27 lines cut ...]
Zigbee2MQTT:debug 2024-03-11 14:41:34: Error: Command 0x54ef4410004afeb1/2 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)
[... cut ...]
2024-03-11T14:41:49.285Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,19,36,1,143,123,1,1,4,11,26,0,30,9,16,20,0,5,5,8,5,11,5,199]
2024-03-11T14:41:49.285Z zigbee-herdsman:controller:endpoint Read 0x70b3d52b6001be36/1 haElectricalMeasurement(["rmsVoltage","rmsCurrent","activePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)

danielsmith89, 2024-03-13:
I'm getting something similar today. Been working flawlessly for over a year and now it stops responding after about 2 minutes after start-up (although sometimes it lasts a bit longer). I've got a new stick on order just in case it's that, but the fact someone else has had something similar recently makes me wonder if it's a software thing.

chrizzlibaer, 2024-03-13:
I got something similar to report. After some online time, some of my zigbee actors stop responding. Switches do not signal or lights cannot be turned on or off via web gui of Z2M.

When i switch back to 1.35 everything is flawless again.

I can't find any recent logs when this happens, the devices just don't respond anymore.

fernando155, 2024-03-13:
I basically change the channel and repaired everything... and now is been working for 2 days. I dont know, is quite weird

andrzejf1994, 2024-03-14:
> I basically change the channel and repaired everything... and now is been working for 2 days. I dont know, is quite weird

@fernando155 mind sharing which channels you tweaked?

fernando155, 2024-03-14:
> > I basically change the channel and repaired everything... and now is been working for 2 days. I dont know, is quite weird
> 
> @fernando155 mind sharing which channels you tweaked?

The problem is on my house I use two zigbee network+ WiFi. I check it and channel 11 was quite busy, so I change it to Channel 15, I was using that one for zha before and was quite stable.

closed 2024-10-11 by github-actions[bot]