Koenkk/zigbee2mqtt #21765: Unable to start  Sonoff Dongle .

mikey6283, 2024-03-11:
### Is your feature request related to a problem? Please describe

Install new Zigbee 3.0 Dongle plus  to replace  Conbee due to issue . Install as a new install  created new config  new pan id and nework_key.  Loads ok but have  error Error while starting zigbee-herdsman

### Describe the solution you'd like

To use the Sonoff usb Zigbee Dongle Plus instead of the Conbee11 please see the log error message.

Zigbee2MQTT:info  2024-03-11 17:21:41: Starting Zigbee2MQTT version 1.36.0 (commit #86ed71c)
Zigbee2MQTT:info  2024-03-11 17:21:41: Starting zigbee-herdsman (0.35.1)
Zigbee2MQTT:error 2024-03-11 17:21:51: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2024-03-11 17:21:51: Failed to start zigbee
Zigbee2MQTT:error 2024-03-11 17:21:51: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-03-11 17:21:51: Exiting...
Zigbee2MQTT:error 2024-03-11 17:21:51: undefined

### Describe alternatives you've considered

I followed the Didblur video and the fault finding  section

### Additional context

Here is a copy of my config:
homeassistant: true
frontend:
  port: 9442
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.xx.xx:1883
  user: mqtt6283
  password: mikey
  version: 4
  keepalive: 60
  reject_unauthorized: true
serial:
  port: >-
    /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_7098e1ad86d8ed11bad86b>
  adapter: 
advanced:
  log_level: info
  pan_id: 15439
  network_key:

 The config `loads` docker in Unraid``

chris-1243, 2024-03-11:
Could you sent a full debug log ? Just change `log_level: info` by `log_level: debug` in your `configuration.yaml`.

chris-1243, 2024-03-13:
If I understand correctly, you are trying to connect the Dongle-P based on TI CC2652P.

In `configuration.yaml`

```
serial:
  port: path of your adapter without special character @
  adapter: zstack # or you may completely remove this line without issue as on `zstack`
```

mikey6283, 2024-03-16:
> If I understand correctly, you are trying to connect the Dongle-P based on TI CC2652P.
> 
> In `configuration.yaml`
> 
> ```
> serial:
>   port: path of your adapter without special character @
>   adapter: zstack # or you may completely remove this line without issue as on `zstack`
> 
> or
> 
>   adapter: null
> ```
> 
> https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#configuration-of-the-zigbee-adapter
> 
> If Dongle-E:
> 
> ```
> serial:
>   adapter: ezsp
> ```
I finally installed the Sonoff Dongle by deleting  my Z2M database and creating a new config and devices . Seems to be working fine e only a few aall sonoff switch which are not connecting, however i cn live without these for the moment.

 Thanks again for point in the right direction.

closed 2024-10-13 by github-actions[bot]

jmcaviglia, 2026-01-09:
> > Si entiendo correctamente, estás intentando conectar el Dongle-P basado en TI CC2652P.
> > En`configuration.yaml`
> > ```
> > serial:
> >   port: path of your adapter without special character @
> >   adapter: zstack # or you may completely remove this line without issue as on `zstack`
> > 
> > or
> > 
> >   adapter: null
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > [https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#configuracion-del-adaptador-zigbee](https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#configuration-of-the-zigbee-adapter)
> > Si Dongle-E:
> > ```
> > serial:
> >   adapter: ezsp
> > ```
> 
> Finalmente instalé el Sonoff Dongle borrando mi base de datos Z2M y creando una nueva configuración y dispositivos. Parece funcionar bien, solo algunos interruptores Sonoff no se conectan; sin embargo, por ahora puedo vivir sin ellos.
> 
> Gracias de nuevo por señalarnos en la dirección correcta.

Hi. Could you help me? I'm having the same problem with the Sonoff coordinator. Could you explain step by step how you solved it? Thanks!