Koenkk/zigbee2mqtt #21773: [New device support]: Aqara MCCGQ13LM - Door and window sensor P1

wachtell, 2024-03-12:
### Link

https://www.zigbee2mqtt.io/devices/MCCGQ13LM.html

### Database entry

*

### Comments

I got stuck at the sensor for door "opening," and the tamper is permanently set to "unknown" or "open".  Tried to restart the Zigbee2MQTT bridge (Sonoff Zigbee 3.0 USB Dongle Plus) flashed with  zStack3x0 20230507 (Firmware 1.36.0) several times to no avail. Tried to delete the devices and re-connect without any change. Do you have any suggestions?
Thank you for all you do!
Best regards

### External definition

```shell
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2024.2.3",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.12.1",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Amsterdam",
    "os_name": "Linux",
    "os_version": "3.10.108",
    "run_as_root": true
  },
  "custom_components": {
    "watchman": {
      "version": "0.5.1",
      "requirements": [
        "prettytable==3.0.0"
      ]
    },
    "alarmo": {
      "version": "v1.9.15",
      "requirements": []
    },
    "mypyllant": {
      "version": "v0.7.3",
      "requirements": [
        "myPyllant==0.7.19"
      ]
    },
    "auto_backup": {
      "version": "1.4.1",
      "requirements": []
    },
    "hacs": {
      "version": "1.34.0",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    }
  },
  "integration_manifest": {
    "domain": "mqtt",
    "name": "MQTT",
    "codeowners": [
      "@emontnemery",
      "@jbouwh"
    ],
    "config_flow": true,
    "dependencies": [
      "file_upload",
[... 622 lines cut ...]
              "object_id": "0x54ef44100040d86e_detection_distance",
              "options": [
                "10mm",
                "20mm",
                "30mm"
              ],
              "origin": {
                "name": "Zigbee2MQTT",
                "sw_version": "1.36.0",
                "support_url": "https://www.zigbee2mqtt.io"
              },
              "state_topic": "zigbee2mqtt/0x54ef44100040d86e",
              "unique_id": "0x54ef44100040d86e_detection_distance_zigbee2mqtt",
              "value_template": "{{ value_json.detection_distance }}",
              "platform": "mqtt"
            }
          },
          "transmitted": []
        }
      ],
      "triggers": []
    }
  }
}
```

gvillo, 2024-03-13:
omg! I am about to buy like 10 Aqara P1 sensors because I thought it was working ok and it has a massive battery.

A couple of questions to see if helps:
* is your MQTT using version 5? and using version 5 in your z2m config?
* do you have availability (simple) check in z2m?
* Zigbee channel free? 15/20/25?

I am running z2m with that config and it works ok, I am not getting any unknown at all on my current devices.

I'll wait to see if it works ok for you before buying! :/

wachtell, 2024-03-18:
I got them working by changing to version 5 but importantly by changing the batteries even though the sensors came right out of the box - go figure!

closed 2024-10-15 by github-actions[bot]