Koenkk/zigbee2mqtt #21775: Crash in dev branch (modernExtend's customTimeResponse?)

srett, 2024-03-12:
### What happened?

I just pulled and tested the latest dev version. I get crashes every couple minutes:

```
node[5096]: Error: ReadResponse 0x38398ffffe495a64/1 genTime({"time":1710264252,"localTime":1710267852}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
node[5096]:     at ZStackAdapter.sendZclFrameToEndpointInternal (/opt/zigbee2mqtt-dev/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23)
node[5096]:     at Queue.execute (/opt/zigbee2mqtt-dev/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
node[5096]:     at Request.send (/opt/zigbee2mqtt-dev/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
node[5096]:     at Endpoint.readResponse (/opt/zigbee2mqtt-dev/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:509:13)
systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
```

It seems the exception [ends up here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/35937ce2fbcbdfde3780c405008c3f5785c5983d/src/lib/modernExtend.ts#L687) and isn't handled any further, but I'm just guessing since async stuff makes my head spin. :)

### What did you expect to happen?

Not crashing

### How to reproduce it (minimal and precise)

Dunno lol

### Zigbee2MQTT version

1.36.0-dev commit: 6daa7aa6

### Adapter firmware version

20230507

### Adapter

CC2652p

### Setup

systemd-nspawn

srett, 2024-03-12:
:man_facepalming: 

After checking everything like ten times I just realized I was working in a directory with a detached HEAD.

srett, 2024-03-12:
OK so someone tried to explain async programming and Promises and stuff to me to no avail but the gist of it is that

```js
data.endpoint.readResponse('genTime', data.meta.zclTransactionSequenceNumber, payload); 
```

should have an `await` in front of it :-)

Koenkk, 2024-03-12:
Yes it should have an `await`, do you want to make a PR?

srett, 2024-03-13:
Nah it's just one line and I didn't even come up with the fix, you can do it directly. :)

Koenkk, 2024-03-13:
committed, thanks!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-03-13 by Koenkk